BiancoRoyal/node-red-contrib-iiot-opcua #1: Call OPC UA Methods with Array Input parameters

eoursel, 2017-12-30:
Hi, 

I am trying to call a method on the Global Discovery Server just published by OPC Foundation. The trouble is that the Method-Caller node does not allow to enter parameters which are an array of strings. Most OPC UA methods of the Global Discovery Server use a parameter ServerCapabilities which is a StringList. 

I understood it could inject a message instead of using the wizard but i am not sure how to define the syntax of the message. 

Thanks

biancode, 2017-12-31:
Hi Eric,

please try the new v1.2.0 with more options to send parameters via msg object
and I'll test more with such methods.

example for bark method is now:

`[{"id":"7b860fbc.baba08","type":"OPCUA-IIoT-Inject","z":"9f1dab9d.97fac","injectType":"inject","payload":"","payloadType":"date","topic":"","repeat":"","crontab":"","once":false,"startDelay":"","name":"","addressSpaceItems":[],"x":140,"y":140,"wires":[["1dacbf1b.0b3f01"]]}]`

cya
Klaus

biancode, 2018-01-06:
reopen if it does not fit your needs

closed 2018-01-06 by biancode