BiancoRoyal/node-red-contrib-iiot-opcua #2: Error: Cannot find module '../_generated_/_auto_generated_LocalizedText'

giminni, 2018-01-06:
Today I tried the 1.2.0 version, inside an alpine docker container. 

Looking at the docker logs I found "Error: Cannot find module '../_generated_/_auto_generated_LocalizedText'" for all nodes.

I guess this is the lower lib from Etienne, I am not shure. BTW I want you know about this.

> 6 Jan 16:16:17 - [warn] [OPCUA-IIoT-Connector] Error: Cannot find module '../_generated_/_auto_generated_LocalizedText'
6 Jan 16:16:17 - [warn] [OPCUA-IIoT-Inject] Error: Cannot find module '../_generated_/_auto_generated_LocalizedText'
6 Jan 16:16:17 - [warn] [OPCUA-IIoT-Node] Error: Cannot find module '../_generated_/_auto_generated_LocalizedText'
6 Jan 16:16:17 - [warn] [OPCUA-IIoT-Event] Error: Cannot find module '../_generated_/_auto_generated_LocalizedText'
6 Jan 16:16:17 - [warn] [OPCUA-IIoT-Listener] Error: Cannot find module '../_generated_/_auto_generated_LocalizedText'
6 Jan 16:16:17 - [warn] [OPCUA-IIoT-Read] Error: Cannot find module '../_generated_/_auto_generated_LocalizedText'
6 Jan 16:16:17 - [warn] [OPCUA-IIoT-Write] Error: Cannot find module '../_generated_/_auto_generated_LocalizedText'
6 Jan 16:16:17 - [warn] [OPCUA-IIoT-Response] Error: Cannot find module '../_generated_/_auto_generated_LocalizedText'
6 Jan 16:16:17 - [warn] [OPCUA-IIoT-Browser] Error: Cannot find module '../_generated_/_auto_generated_LocalizedText'
6 Jan 16:16:17 - [warn] [OPCUA-IIoT-Server] Error: Cannot find module '../_generated_/_auto_generated_LocalizedText'
6 Jan 16:16:17 - [warn] [OPCUA-IIoT-Server-ASO] Error: Cannot find module '../_generated_/_auto_generated_LocalizedText'
6 Jan 16:16:17 - [warn] [OPCUA-IIoT-Server-Command] Error: Cannot find module '../_generated_/_auto_generated_LocalizedText'
6 Jan 16:16:17 - [warn] [OPCUA-IIoT-Server-ISA95] Error: Cannot find module '../_generated_/_auto_generated_LocalizedText'
6 Jan 16:16:17 - [warn] [OPCUA-IIoT-Result-Filter] Error: Cannot find module '../_generated_/_auto_generated_LocalizedText'
6 Jan 16:16:17 - [warn] [OPCUA-IIoT-Method-Caller] Error: Cannot find module '../_generated_/_auto_generated_LocalizedText'
6 Jan 16:16:17 - [warn] [OPCUA-IIoT-Flex-Server] Error: Cannot find module '../_generated_/_auto_generated_LocalizedText'
6 Jan 16:16:17 - [warn] ------------------------------------------------------

Runtime:
======
Alpine Linux: 3.7.0
npm: 5.6.0
node: 8.9.4
node-red: 0.17.5
node-red-contrib-iiot-opcua: 1.2.0

biancode, 2018-01-06:
Yes, I had that problem too, fixed it and now I have trouble in Travis with node8.

biancode, 2018-01-06:
node6 is working well

giminni, 2018-01-06:
Thx Klaus, should I revert to node6?

biancode, 2018-01-06:
You could give it a try - v6 works great for me.

giminni, 2018-01-06:
Let me test it, I'll report if it was successful.

biancode, 2018-01-06:
I released the v1.3.0 with variables in flex server, that should also work with node6.

giminni, 2018-01-06:
Is v1.3.0 public available?

biancode, 2018-01-06:
Yes, [it is](https://www.npmjs.com/package/node-red-contrib-iiot-opcua).

giminni, 2018-01-06:
Klaus, you save my day 👍 
I confirm what you said.

If I use the following runtime:
Alpine Linux: 3.6.x
node: 6.10.3
npm: 3.10.10
node-red-contrib-iiot-opcua: 1.3.0

I had no warnings and the nodes appears in node-RED again.
Please put a note somewhere for the other users, maybe they fall into the same problem.
Thx a lot for your outstanding support.

biancode, 2018-01-06:
Travis did it after using lts/* with node: 8.9.4

giminni, 2018-01-06:
Great, that means I can close this issue.

closed 2018-01-06 by giminni