BiancoRoyal/node-red-contrib-iiot-opcua #3: [error] [OPCUA-IIoT-Connector:OPCServer] TypeError: Cannot read property 'join' of null

ManniMo, 2018-01-22:
Hello,
i get this error when i try to connect to a specific OPC UA server. So theoreticaly it could be a fault from the server but I have no specific idea where the error is from after hours of searching. I hope someone here can help me.

Security Policy and Mode are both None. Credentials are used.

My setup is:
OS: Windows 7
Node.js: v8.9.2
Node-RED: v0.17.5
node-red-contrib-iiot-opcua: v1.3.3

> 22 Jan 14:02:10 - [info] Started flows
>  WARNING SERVER responseHeader.requestHandle is invalid: expecting 0x1  but got 0x0  Class = OpenSecureChannelResponse
>  ERROR !!! , please check here !!!! callback may be called twice !!  TypeError: Cannot read property 'join' of null
>     at C:\Users\myUser\.node-red\node_modules\node-red-contrib-iiot-opcua\opcuaIIoT\maps\core\core\opcua-iiot-core-connector.js:101:47
>     at Array.forEach (<anonymous>)
>     at C:\Users\myUser\.node-red\node_modules\node-red-contrib-iiot-opcua\opcuaIIoT\maps\core\core\opcua-iiot-core-connector.js:93:15
>     at C:\Users\myUser\.node-red\node_modules\node-opcua-client\src\client_base.js:684:9
>     at modified_callback (C:\Users\myUser\.node-red\node_modules\node-opcua-secure-channel\src\client\client_secure_channel_layer.js:986:28)
>     at process_request_callback (C:\Users\myUser\.node-red\node_modules\node-opcua-secure-channel\src\client\client_secure_channel_layer.js:76:5)
>     at ClientSecureChannelLayer._on_message_received (C:\Users\myUser\.node-red\node_modules\node-opcua-secure-channel\src\client\client_secure_channel_layer.js:122:5)
>     at emitMany (events.js:147:13)
>     at MessageBuilder.emit (events.js:224:7)
>     at MessageBuilder._decode_message_body (C:\Users\myUser\.node-red\node_modules\node-opcua-secure-channel\src\message_builder.js:442:22)
> 22 Jan 14:02:12 - [error] [OPCUA-IIoT-Connector:OPCServer] TypeError: Cannot read property 'join' of null
>     at C:\Users\myUser\.node-red\node_modules\node-red-contrib-iiot-opcua\opcuaIIoT\maps\core\core\opcua-iiot-core-connector.js:101:47
>     at Array.forEach (<anonymous>)
>     at C:\Users\myUser\.node-red\node_modules\node-red-contrib-iiot-opcua\opcuaIIoT\maps\core\core\opcua-iiot-core-connector.js:93:15
>     at C:\Users\myUser\.node-red\node_modules\node-opcua-client\src\client_base.js:684:9
>     at modified_callback (C:\Users\myUser\.node-red\node_modules\node-opcua-secure-channel\src\client\client_secure_channel_layer.js:986:28)
>     at process_request_callback (C:\Users\myUser\.node-red\node_modules\node-opcua-secure-channel\src\client\client_secure_channel_layer.js:76:5)
>     at ClientSecureChannelLayer._on_message_received (C:\Users\myUser\.node-red\node_modules\node-opcua-secure-channel\src\client\client_secure_channel_layer.js:122:5)
>     at emitMany (events.js:147:13)
>     at MessageBuilder.emit (events.js:224:7)
>     at MessageBuilder._decode_message_body (C:\Users\myUser\.node-red\node_modules\node-opcua-secure-channel\src\message_builder.js:442:22)

"WARNING SERVER ..."  is posted every few seconds with, for me, random "expecting 0x**" 

Thanks in advance and if this error ist not from the client side forgive me 😄.
ManniMo

biancode, 2018-01-22:
Hi ManniMo,

hm, the code lines from the stack trace are different. Are you sure to have 1.3.3 running?

[image]

It looks like a problem with that endpoints or with the `endpoint.server.discoveryUrls`.
Please, could you [record some logs](https://github.com/biancode/node-red-contrib-iiot-opcua/wiki/DEBUG) for the connector?

With the switch for logging, you should see what you get from the server.

Happy logging!
Klaus

biancode, 2018-01-22:
[image]

Please, watch on "Manage pallet" in Node-RED, which version is running.

biancode, 2018-01-22:
You could try the [opcua-commander](https://github.com/node-opcua/opcua-commander) too. we have the same base ;)

ManniMo, 2018-01-23:
Thanks for your quick answer.
I've duble checked the version number, it is 1.3.3. In adition, the installation is from yesterday with the node-RED palette manager, so i think it should be the newest stable build. I have checked the node-opcua version too, it is 0.1.1-0 in the package of this node. My only idea if this errors are from different versions is that i have an older version of node-opcua installed somewhere else but there nothing tries to connect to a server.
When i look at the debug log in node-red  it shows me:

> "Error: connect ETIMEDOUT ip"
>"Error stopping node: TimeoutError: timed out after 15000ms"
>"TypeError: Cannot read property 'join' of null"

I will try the debugger now, and will edit this post after that.

biancode, 2018-01-30:
Hi, what did you get from the [opcua-commander](https://github.com/node-opcua/opcua-commander)?

ManniMo, 2018-01-31:
Hi,
if I try the opcua-commander I get "WARNING SERVER responseHeader.requestHandle is invalid: expecting 0x*** but got 0x0 Class = OpenSecureChannelResponse" where * is a hexadecimal from 0 to f.
But I can connect to the server this way.

ManniMo, 2018-01-31:
Could it be possible that iiot-opcua can't handle the Application URI of urn:unconfigured:Application? Since other clients (opcua-commander and i have tried UaExpert too today) respond with errors that can be ignored I think the server isn't configured correctly but it would be nice if that could be handled here too. 
From the UaExpert I get 
[image]

biancode, 2018-01-31:
looks like it is fixed in node-opcua 0.2.0, but there are some new issues, why it could not be updated for now [node-opcua issue](https://github.com/node-opcua/node-opcua/pull/394)

ManniMo, 2018-02-02:
That would be nice. Then I will wait until there is an update. Unfortunately I dont know how to edit Nodes to try 0.2.1 myself and publish it here

ManniMo, 2018-02-12:
I tried the new Version of this node now and it seems the error ist fixed. But it needs a huge amount of Time to connect to the server (about 1 minute). I'll add the error log later.

biancode, 2018-02-12:
Wow! +17s to close ... but there is not a lot I can do for that problem - please check the open lib close behavior with the node-opcua lib and help to fix that by node-opcua

closed 2018-02-12 by biancode