BiancoRoyal/node-red-contrib-iiot-opcua #4: No description item found using the browse node

giminni, 2018-02-06:
I can browse a top level OPC/UA directory successfully, looking into the array I am missing the description field on each item.  I am using version 1.3.3. 
Can you add this to the code?

biancode, 2018-02-06:
Hi Giovanni, 

with node-opcua you have to browse 
and send the browse results to a read node to get that attributes you wanna get.

see and test the example flow in Node-RED:

[image]

biancode, 2018-02-06:
API changed in node-opcua for readAllAttributes

biancode, 2018-02-06:
if the [node-opcua issue v0.2.x](https://github.com/node-opcua/node-opcua/pull/394) is fixed it will work again:

[image]

biancode, 2018-02-06:
but browser result stay:

[image]

biancode, 2018-02-12:
Please, test v2.x+

closed 2018-02-12 by biancode

biancode, 2018-02-16:
please check v2.0.4 or above

giminni, 2018-02-19:
Thank a lot Klaus, I‘ll try it