BiancoRoyal/node-red-contrib-iiot-opcua #6: Node S-ASO fails to add boolean variable to Server

SilviuToc, 2018-02-08:
Hello, 

The node S-ASO fails to add boolean variable when inject with value "false" ( works with value "true" ).

[image]
Sorry if I made any mistake, I have just started learning about OPC-UA.
My code:

[{"id":"e9e4aee6.b1a5b","type":"OPCUA-IIoT-Server","z":"8258c812.8b1668","port":"55388","endpoint":"","acceptExternalCommands":true,"maxAllowedSessionNumber":"","maxConnectionsPerEndpoint":"","maxAllowedSubscriptionNumber":"","alternateHostname":"","name":"","showStatusActivities":true,"showErrors":true,"allowAnonymous":false,"isAuditing":false,"users":[{"name":"admin","password":"admin001"}],"x":557.072868347168,"y":88.29517269134521,"wires":[[]]},{"id":"cfe49f8a.79ef","type":"OPCUA-IIoT-Inject","z":"8258c812.8b1668","injectType":"inject","payload":"","payloadType":"date","topic":"","repeat":"","crontab":"","once":true,"startDelay":"10","name":"","addressSpaceItems":[],"x":145.07638549804688,"y":76.12153816223145,"wires":[["f939eb43.d420c8"]]},{"id":"f939eb43.d420c8","type":"OPCUA-IIoT-Server-ASO","z":"8258c812.8b1668","nodeId":"ns=0;s=Folder","browsename":"Folder","displayname":"Folder","objecttype":"FolderType","datatype":"","value":"","referenceNodeId":"ns=0;i=85","referencetype":"Organizes","name":"","x":293.0799102783203,"y":76.33334255218506,"wires":[["e9e4aee6.b1a5b","8fb4d458.0cccd8"]]},{"id":"8c4a900d.0e689","type":"OPCUA-IIoT-Inject","z":"8258c812.8b1668","injectType":"inject","payload":"false","payloadType":"bool","topic":"","repeat":"","crontab":"","once":true,"startDelay":"15","name":"","addressSpaceItems":[],"x":123.01736068725586,"y":116.01043701171875,"wires":[["2b48b9a3.1db426","ea37dd39.62793"]]},{"id":"2b48b9a3.1db426","type":"OPCUA-IIoT-Server-ASO","z":"8258c812.8b1668","nodeId":"ns=0;s=Var","browsename":"Var","displayname":"Var","objecttype":"BaseDataVariableType","datatype":"Boolean","value":"false","referenceNodeId":"ns=0;s=Folder","referencetype":"HasComponent","name":"","x":292.02090072631836,"y":122.22222232818604,"wires":[["e9e4aee6.b1a5b","8fb4d458.0cccd8"]]},{"id":"8fb4d458.0cccd8","type":"debug","z":"8258c812.8b1668","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":557.0833053588867,"y":144.1874942779541,"wires":[]},{"id":"ea37dd39.62793","type":"debug","z":"8258c812.8b1668","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":315.0173797607422,"y":167.01041221618652,"wires":[]}]

Thank you !

biancode, 2018-02-12:
Hi,

Yes, you are right. The value check of the boolean accepting only type of boolean values in the new version. I have fixed it and will release a fix within a few days.

Thank you for that issue!
Klaus

biancode, 2018-02-12:
[image]

biancode, 2018-02-12:
`[{"id":"25e800c9.07097","type":"tab","label":"Flow 1","disabled":false,"info":""},{"id":"c59c3609.9d6bf8","type":"OPCUA-IIoT-Inject","z":"25e800c9.07097","injectType":"inject","payload":"","payloadType":"date","topic":"","repeat":"","crontab":"","once":true,"startDelay":"10","name":"","addressSpaceItems":[],"x":248.0035171508789,"y":120.82636547088623,"wires":[["a71f890c.4b45f8"]]},{"id":"a71f890c.4b45f8","type":"OPCUA-IIoT-Server-ASO","z":"25e800c9.07097","nodeId":"ns=1;s=Folder","browsename":"Folder","displayname":"Folder","objecttype":"FolderType","datatype":"","value":"","referenceNodeId":"ns=0;i=85","referencetype":"Organizes","name":"","x":396.00704193115234,"y":121.03816986083984,"wires":[["c9479df9.848bf"]]},{"id":"df89825e.a0e44","type":"OPCUA-IIoT-Inject","z":"25e800c9.07097","injectType":"inject","payload":"false","payloadType":"bool","topic":"","repeat":"","crontab":"","once":true,"startDelay":"15","name":"","addressSpaceItems":[],"x":234.94448852539062,"y":183.71527099609375,"wires":[["b18bf3af.a5fef","7759f84b.b77e88"]]},{"id":"b18bf3af.a5fef","type":"OPCUA-IIoT-Server-ASO","z":"25e800c9.07097","nodeId":"ns=1;s=VarTrue","browsename":"VarTrue","displayname":"Var True","objecttype":"BaseDataVariableType","datatype":"Boolean","value":"","referenceNodeId":"ns=1;s=Folder","referencetype":"Organizes","name":"","x":400.947998046875,"y":202.92706298828125,"wires":[["c9479df9.848bf"]]},{"id":"c9479df9.848bf","type":"OPCUA-IIoT-Server","z":"25e800c9.07097","port":"55388","endpoint":"","acceptExternalCommands":true,"maxAllowedSessionNumber":"","maxConnectionsPerEndpoint":"","maxAllowedSubscriptionNumber":"","alternateHostname":"","name":"","showStatusActivities":false,"showErrors":false,"allowAnonymous":true,"isAuditing":false,"users":[],"publicCertificateFile":"","privateCertificateFile":"","x":605.5,"y":186.5,"wires":[["8802ef34.6b5408"]]},{"id":"7759f84b.b77e88","type":"OPCUA-IIoT-Server-ASO","z":"25e800c9.07097","nodeId":"ns=1;s=FloatTest","browsename":"FloatTest","displayname":"Float Test","objecttype":"BaseDataVariableType","datatype":"Float","value":"1000.0","referenceNodeId":"ns=1;s=Folder","referencetype":"Organizes","name":"","x":399,"y":159,"wires":[["c9479df9.848bf"]]},{"id":"8802ef34.6b5408","type":"debug","z":"25e800c9.07097","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","x":758.5,"y":185.5,"wires":[]},{"id":"b0f180ee.ce6858","type":"OPCUA-IIoT-Server-ASO","z":"25e800c9.07097","nodeId":"ns=1;s=VarFalse","browsename":"VarFalse","displayname":"Var False","objecttype":"BaseDataVariableType","datatype":"Boolean","value":"","referenceNodeId":"ns=1;s=Folder","referencetype":"Organizes","name":"","x":402,"y":249,"wires":[["c9479df9.848bf"]]},{"id":"4bb38682.312bb8","type":"OPCUA-IIoT-Inject","z":"25e800c9.07097","injectType":"inject","payload":"true","payloadType":"bool","topic":"","repeat":"","crontab":"","once":true,"startDelay":"15","name":"","addressSpaceItems":[],"x":241,"y":246,"wires":[["b0f180ee.ce6858"]]}]`

biancode, 2018-02-12:
fixed since v2.0.2+

closed 2018-02-12 by biancode