BiancoRoyal/node-red-contrib-iiot-opcua #7: How to control client subscription publishing interval?

cristipogacean, 2018-02-12:
Hi, 

It seams that Client's subscription publishing interval is defaulted to 10 seconds.  Is there a possibility to be changed to a desired value? Monitored item's sampling rate respective queue size are configurable and  seam to function properly.

Thank you and kind regards,
Cristian

biancode, 2018-02-12:
Hi,

Yes, sure!

With an inject ...

[image]

cristipogacean, 2018-02-13:
Hi again,  

I was aware about the inject you mentioned, nevertheless I think, from the behavior I see, the "interval" parameter refers there to the monitored item "sampling rate" and not to the subscription's publishing interval. 

The behavior that I see is that with the above mentioned settings: I get every 10 seconds a package of 10 data-changes (sampled every 500ms). when I change the queue size to 1, then I get a data-change every 10 seconds.

How would you recommend to configure the inject in order to receive data-changes packed in publish responses with a frequency different than 10 seconds? 

Thank you and kind regards,
Cristian

biancode, 2018-02-25:
[image]

biancode, 2018-02-25:
[image]

[image]

The msg.payload could change the requestedPublishingInterval parameter.

biancode, 2018-02-25:
read the new info of the node and try the new msg.payload.options to test, please

closed 2018-02-27 by biancode