BiancoRoyal/node-red-contrib-iiot-opcua #8: OPC UA IIoT Read - String in Payload instead of Object Array

HaegMar, 2018-02-14:
Since V 2.0.x the Read Node writes a String in Payload instead of an Object Array as it was before. My current Workaround is to convert it with a json-node to an JavaScript-Object.
Next problem is that in this new Object is no "[0].convertedValue" like it was before.

> "[{"value":{"dataType":"Float","arrayType":"Scalar","value":0},"statusCode":{"value":0,"description":"No Error","name":"Good"},"sourcePicoseconds":0,"serverPicoseconds":0},{"value":{"dataType":"Float","arrayType":"Scalar","value":0},"statusCode":{"value":0,"description":"No Error","name":"Good"},"sourcePicoseconds":0,"serverPicoseconds":0},{"value":{"dataType":"Float","arrayType":"Scalar","value":0},"statusCode":{"value":0,"description":"No Error","name":"Good"},"sourcePicoseconds":0,"serverPicoseconds":0},{"value":{"dataType":"Float","arrayType":"Scalar","value":0},"statusCode":{"value":0,"description":"No Error","name":"Good"},"sourcePicoseconds":0,"serverPicoseconds":0},{"value":{"dataType":"Float","arrayType":"Scalar","value":3276.699951171875},"statusCode":{"value":0,"description":"No Error","name":"Good"},"sourcePicoseconds":0,"serverPicoseconds":0}]"

biancode, 2018-02-14:
Yes, the API of node-opcua has changed and if you get a string, then it is not to convert and send without problems. I am working on that.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

HaegMar, 2018-02-14:
Okay, thank you.

It would be nice if it would work like before, because i have a huge self written function behind the read-node to sort a lot of Datapoints in a SQL DB. For this Time i went back to the old version.

biancode, 2018-02-16:
You could try v2.0.4 if it works for you. I am sorry if not, but it is a significant upgrade in v0.2.x of node-opcua.

closed 2018-02-16 by biancode