BiancoRoyal/node-red-contrib-iiot-opcua #9: Node ASO fails to add DateTime variable to Server

pcogermany, 2018-02-20:
This error is like #6 but with a DateTime instead of Boolean.
In the debug section I get:
```AssertionError [ERR_ASSERTION]: value === null || value instanceof Date```
I tried version from 1.0.15 to 2.0.4

pcogermany, 2018-02-20:
additional information: I used a OPCUA Inject Node with a timestamp as payload. That didn't work as described above, but I also tried to put a function block with `msg.payload = new Date();` between the inject and ASO and that didn't work either.

biancode, 2018-02-20:
server has to parse write value

biancode, 2018-02-20:
fixed in v2.0.5

biancode, 2018-02-20:
[image]

pcogermany, 2018-02-21:
That was fast, thank you!
However I'm not able to write a DateTime value by defining msg.addressSpaceItems and then put the Date into msg.valuesToWrite.
All I get is another assertion error, as the code "thinks", the datatype is a string instead of a DateTime:
`"AssertionError [ERR_ASSERTION]: typeof value === "string" || value === null"`
Do you have an idea about the problem?

pcogermany, 2018-02-21:
Looks like the DateTime datatype is missing in this switch/case:
https://github.com/biancode/node-red-contrib-iiot-opcua/blob/9093171e89b0db5e9e15befd70344138e8598ccd/src/core/opcua-iiot-core.js#L198
Would you mind aading something like this:
```javascript
    case 'DateTime':
    case opcua.DataType.DateTime:
      variantValue = {
        dataType: opcua.DataType.DateTime,
        value: new Date(value)
      }
    break
```

biancode, 2018-02-22:
please test v2.0.7 or above

closed 2018-02-22 by biancode