BiancoRoyal/node-red-contrib-iiot-opcua #10: No datapoint info inside a subscribed event

giminni, 2018-02-22:
Hi Klaus,

I am testing an injector with two data points connected to a listener that itself is connected to an OPC/UA server. After the injector is started I see two subscribed data points. So far so good.

If I trigger a change of value I get the following data structure:

`{"value":{"dataType":"Boolean","arrayType":"Scalar","value":false},"statusCode":{"value":0,"description":"No Error","name":"Good"},"sourceTimestamp":"2018-02-22T14:58:11.260Z","sourcePicoseconds":0,"serverTimestamp":"2018-02-22T14:58:11.721Z","serverPicoseconds":0}`

What I need is the corresponding data point, description that triggered this event.

How can I get this info bound to the event? 

... and what is meta inject, multi node option means for the listener node?

biancode, 2018-02-22:
Hi Giovanni,

did you check the msg object?

[image]

biancode, 2018-02-22:
and you get a message per monitored object / nodeId

biancode, 2018-02-22:
and the result filter could pick up what you are searching for ...

[image]

biancode, 2018-02-22:
The Meta Inject was an idea to take care about sending some meta data to react (read, write ...) on events, but I found no time to finish that. It is inactive for now.

giminni, 2018-02-22:
I cannot see the payload mentioned before

[image]

biancode, 2018-02-22:
Switch the debug node to msg and not just msg.payload ;)

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

closed 2018-02-24 by biancode