BiancoRoyal/node-red-contrib-iiot-opcua #11: OPC-UA connector not working anymore

giminni, 2018-02-24:
Hi Klaus,

tried 2.0.5 and 2.0.8 cannot connect to an OPC-UA server anymore. It works before, log shows
[ 'opc.tcp://10.1.27.101:4840 NONE http://opcfoundation.org/UA/SecurityPolicy#None' ]

biancode, 2018-02-24:
Hi Giovanni,

could you give more logs?

biancode, 2018-02-24:
strange ... just if a clean install had that problem in my development environment ... should be fixed with v2.0.9

giminni, 2018-02-24:
Tested and deployed 2.0.9 same error
[ 'opc.tcp://10.1.27.101:4840 NONE http://opcfoundation.org/UA/SecurityPolicy#None' ]

The listener node cannot connect to an external OPC/UA server the status message is always waiting...

giminni, 2018-02-24:
I am not sure, I guess the problem is coming from node-opcua CreateSession ?

biancode, 2018-02-24:
v2.0.9 is on node-opcua 0.2.1 and it works for me

[image]

biancode, 2018-02-24:
You'll get the endpoint in debug log - try the given endpoint, please

biancode, 2018-02-25:
Please test the v2.0.10, what you get for the endpoints

coreConnector.internalDebugLog('Get Endpoints Request Error ' + err.message)

giminni, 2018-02-25:
Nothing changed since 2.0.5:
 
- In the log I get always: [ 'opc.tcp://10.1.27.101:4840 NONE http://opcfoundation.org/UA/SecurityPolicy#None' ]
- My OPC-UA server has definitively another IP address and port. 
- The status of IIOTConnector is waiting...
- From the command line I can ping the OPC-UA server. 
- If I use the ProSys OPC Client I can successfully connect to the OPC-UA server.
- I am using node 6.10.10 and npm 3.10.3
- I am running Alpine Linux 3.7 on Docker 17.09.1-ce

Should I enable DEBUG on IIOTConnector to track down the problem?

biancode, 2018-02-25:
> Questions:
> What caused this breaking change?
API changed in major style for node-opcua 0.2.x and I am working in iterative steps to fix all until 2.1.x
You could also use v1.x of my package if 2.x is not stable for you

> Should I enable DEBUG on IIOTConnector to track down the problem?
Yes, but better to see is a DEBUG=opcuaIIoT* log

biancode, 2018-02-25:
opc.tcp://pcname.mshome.net:53530/

biancode, 2018-02-25:
[image]

biancode, 2018-02-25:
[image]

biancode, 2018-02-25:
I have tested with the opcua-commander and it looks like an issue in node-opcua. Please, open an issue for that at node-opcua!

closed 2018-02-25 by biancode

biancode, 2018-02-25:
tested with opcua-commander v0.2.1 and v0.2.2 node-opcua

giminni, 2018-02-27:
Hi @biancode 

Here is the debug output from the connector, please bear in mind my OPC-UA server has two network adapter LAN and WAN. the 10.0.1.123 is my LAN address and 10.1.27.101 is my WAN address. I want to connect from the LAN side using 10.0.1.123

>2018-02-26T19:46:51.222Z opcuaIIoT:connector Connecting On opc.tcp://10.0.1.123:4840
2018-02-26T19:46:51.551Z opcuaIIoT:connector Connected On opc.tcp://10.0.1.123:4840 Options {"securityPolicy":"None","securityMode":"NONE","defaultSecureTokenLifetime":"60000","keepSessionAlive":true,"certificateFile":"/data/red/node_modules/node-opcua-client/certificates/client_selfsigned_cert_1024.pem","privateKeyFile":"/data/red/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem"}
2018-02-26T19:46:51.564Z opcuaIIoT:connector endpoint: opc.tcp://10.1.27.101:4840
2018-02-26T19:46:51.565Z opcuaIIoT:connector urn:10.1.27.101
2018-02-26T19:46:51.569Z opcuaIIoT:connector Setup Certified Options For opc.tcp://10.0.1.123:4840
2018-02-26T19:46:51.585Z opcuaIIoT:connector !!!!!!!!!!!!!!!!!!!!!!!!  CONNECTION CLOSED !!!!!!!!!!!!!!!!!!!
2018-02-26T19:46:51.592Z opcuaIIoT:connector Disconnected From opc.tcp://10.0.1.123:4840
2018-02-26T19:46:51.638Z opcuaIIoT:connector Secured Connected On opc.tcp://10.0.1.123:4840
2018-02-26T19:46:51.640Z opcuaIIoT:listener starting OPC UA session with delay of 1000
2018-02-26T19:46:52.643Z opcuaIIoT:listener Listener Start OPC UA Session
2018-02-26T19:46:52.644Z opcuaIIoT:connector Request For New Session From Listener Node
2018-02-26T19:46:52.669Z opcuaIIoT:connector Session Start Error Error:  End point must exist opc.tcp://10.0.1.123:4840
2018-02-26T19:46:52.670Z opcuaIIoT:listener Handle Session Error Error:  End point must exist opc.tcp://10.0.1.123:4840
[ 'opc.tcp://10.1.27.101:4840 NONE http://opcfoundation.org/UA/SecurityPolicy#None' ]

biancode, 2018-02-27:
@erossignon - please - some idea about that problem? - I tested with ProSys server too and it does not work - same problem with the opcua-commander

erossignon, 2018-02-28:
BTW, I just retested installing opcua-commander@0.2.2 this way and it worked as expected
```
npm install opcua-commander -g
opcua-commander opc.tcp://MYLAPTOP:53530/OPCUA/SimulationServer
```

biancode, 2018-02-28:
OK, I'll add the endpoint_must_exist option to control from node settings.

biancode, 2018-02-28:
@erossignon thank you - step forward but Internal Error in ProSys server.

[image]

[image]

biancode, 2018-02-28:
[image]

biancode, 2018-02-28:
[image]

deothermo, 2018-03-05:
Hi,

I have a similar / the same issue: I can't connect to an opc server using the current version.

node: v8.9.4
node-red: v0.17.5
node-opcua: v0.2.2
node-red-contrib-iiot-opcua: v2.0.12

opc ua server: open62541

I can connect without problems using UAExpert. Using node-red / OPCUA-IIoT, I get the following errors on the server:

> [03/05/2018 19:56:23.715] info/network  Connection 5 | New connection over TCP from 192.168.178.36:40705
[03/05/2018 19:56:23.814] info/channel  Connection 5 | SecureChannel 2 | OpenSecureChannel: Opened SecureChannel
[03/05/2018 19:56:23.828] info/session  Try to use Session with token 00000000-0000-0000-0000-000000000000 but is not found
[03/05/2018 19:56:25.919] info/session  Try to use Session with token 00000000-0000-0000-0000-000000000000 but is not found
[03/05/2018 19:56:25.927] info/channel  Connection 5 | SecureChannel 2 | CloseSecureChannel
[03/05/2018 19:57:25.948] info/network  Connection 6 | New connection over TCP from 192.168.178.36:40791
[03/05/2018 19:57:25.967] info/channel  Connection 6 | SecureChannel 3 | OpenSecureChannel: Opened SecureChannel
[03/05/2018 19:57:25.984] info/session  Try to use Session with token 00000000-0000-0000-0000-000000000000 but is not found
[03/05/2018 19:57:27.036] info/session  Connection 0 | SecureChannel 0 | Session dd73bc5f-2f84-a3ea-e0ce-5ca505509509 | ActivateSession: SecureChannel 3 wants to activate, but the UserIdentify token is invalid

In node-red, I receive:
> "Error: ServiceResult is BadIdentityTokenInvalid (0x80200000)"

biancode, 2018-03-12:
[older issues on node-opcua for that](https://github.com/node-opcua/node-opcua/issues/158)