BiancoRoyal/node-red-contrib-iiot-opcua #12: How To send a dynamic value to a node connector?

giminni, 2018-02-25:
I want to set the value in the node connector dynamically, which output structure should I fill to populate the node data value from an input msg.payload.d.value field?

giminni, 2018-02-27:
To clarify more I want extract a previous computed or received msg item value (e.g. 80) and inject it into any  IIOT nodes that can handling data values (read, write, listener, ...).
If you provide hooks like msg.value or something similar, the user can pass dynamic data to a placeholder without taking care where to fill the data in your node.

Can you implement this for all values that can be changed programmatically?

biancode, 2018-02-27:
Please, use the structure as it is. From my point it is all there to send dynamic data. All work of the inject could be done by the event message. It is not clearly to see, what you are missing or searching for.

giminni, 2018-02-27:
Should I use nodesTowrite.value.value.value for injecting the value?

biancode, 2018-02-27:
nodesToWrite is to fill as an Array of values and objects ... why do you think you have to use nodesTowrite.value.value.value?

giminni, 2018-02-27:
Looking into the debug node I could see a value object that is nested into another value object and at least the value item with the data in it.

biancode, 2018-02-27:
I am sorry, but maybe you are at the wrong place to debug?!

[image]

biancode, 2018-03-01:
please reopen if you need more ...

closed 2018-03-01 by biancode