BiancoRoyal/node-red-contrib-iiot-opcua #13: Make Listener work with Browse Result

araisch, 2018-02-28:
First of all thanks for your great work with this plugin. I love it.

Problem: If you want to transfer huge amounts of variables to a sink with limited afford, it's working great with Browse & Read. The drawback is, that you spam the sink with lot of near-static values (with changes less then once a day) with the given polling rate. 
Solution: Make the Listener node work with Browse results, like the Read node does already today.

Right, I could code a function transforming a Browse result to a msg-style the Listener is able to read by myself. But subscribe is such a great function for minimizing data from opc ua that this might be a nice enhancement for other users as well.

biancode, 2018-03-01:
Thank you!

It is not clearly, what are you want to do. In normal case you send one subscription to all variables to listen for. Then the Listener works as your settings set the options (publish, sample interval ...) - means you do not need a next inject to the listener from here if you just want to listen to the same variables all the time.

Please, describe more in detail, what is expected!

cya
Klaus

araisch, 2018-03-01:
The flow to poll many variables at once looks like that:
[image]
The flow for subscribe many variables at once should look like that (in my opinion):
[image]

But that does not work. Therefore i have to enter all the UA-nodeIDs manually in the inject node and do it like this:
[image]

To generate the variable-list for Listening automatically and keep it on date (when add or remove UA-Nodes from the server) the approach with Inject -> Browse -> Listen would be great. In other words: The Listener should be able to deal with the output message of the Browser, like the Read-Node does.

biancode, 2018-03-01:
like that?

[image]

biancode, 2018-03-01:
please test v2.0.12 or above

biancode, 2018-03-02:
Take a look to the whole msg object. There you will find more information.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

araisch, 2018-03-02:
It is the whole msg object. There is just the _msgid outside of the payload.
Like I said: In the upper output message there is an array "addressSpaceItem" with the field "nodeID". But this message contains also huge arrays input.addressSpaceItem and input.nodesToRead. 

In my opinion the "Send just values" property in the "Listener"-Node makes no sense. For me it should be like:

- "Send just values" is ticked: The Listener just sends the light-weight message I screenshoted above (with added field containing the node-ID).
- "Send just values" is not ticked: The Listener sends the big message with all the information.
Reason: I'm not sure if there is an influence on performance if you spam this huge message hundred times a second. On a big machine there will be no problem. But assume that you run this on a small "on-the-edge"-Device for connecting an OPC-UA-Device to IoT with transfering hundreds of variables a second. I have the fear to book out the device with that hundreds of huge messages.

Therefore my suggest: One option for big machines with that huge all-containing message (you provide already at the upper output) and one option for light-weight on-the-edge devices with a minimum footprint message (you provide already at the lower output+nodeID). And the user can choose with the "Send just values" property.
[image]

biancode, 2018-03-02:
[image]

araisch, 2018-03-02:
This is the message of the upper output.
[image]
If you subscribe 1000 variables it generates at least on start-up 1000 messages with the content:
[image]
And all the 1000 messages will have two arrays with again 1000 objects each (yellow highlighted in screenshot). This means **2 Millions objects are generated at once**. For small devices this could be a problem. 
So my point is:
- Property to switch on/off this upper output. But not just the output itself, the creation of those objects internal as well. 
- Give the lower output a new field named node-ID, because the lower output is really fine. Enough for 99% of all users and still very small. But there is no way to identify the variable, yet.

biancode, 2018-03-02:
Yes, you are right and I know that. I am working on the structures step by step. The API of node-opcua becomes more easy so it will become less spam in the future in communication, but I need time to do all the work ;)

biancode, 2018-03-02:
next version event msg

[image]

biancode, 2018-03-02:
next version monitor msg 1 & 2

[image]

biancode, 2018-03-05:
please test with v2.1.0

closed 2018-03-05 by biancode