BiancoRoyal/node-red-contrib-iiot-opcua #14: Session Not Valid and AssertionError in

sixdiamants, 2018-03-02:
Dear Klaus,

your package is quite appealing and I would love to get it running on my AWS Ubuntu 16.04 instance. The version of node-red-contrib-iiot-opcua is 2.0.9. Node-red version is 0.18.3

The example flow IIoT OPC UA Browser produces below error. The Session Not Valid error also occurs permanently and I suspect that a setting is wrong. This is surprising given the fact that I run the flow out-of-the-box without any modification.

Here's the content of the log-file `(DEBUG=opcuaIIoT* node-red -v 1>nodeREDIIoTOPCUA.log 2>&1)`

`2 Mar 12:56:39 - [info] 

Welcome to Node-RED
===================

2 Mar 12:56:39 - [info] Node-RED version: v0.18.3
2 Mar 12:56:39 - [info] Node.js  version: v8.9.4
2 Mar 12:56:39 - [info] Linux 4.4.0-1052-aws x64 LE
2 Mar 12:56:39 - [info] Loading palette nodes
 Setting debug for  factories_schema_helpers        to  false
 Setting debug for  tools_event_filter              to  false
 Setting debug for  client_base                     to  false
 Setting debug for  tcp_transport                   to  false
 Setting debug for  client_tcp_transport            to  false
 Setting debug for  server_tcp_transport            to  false
 Setting debug for  client_secure_channel_layer     to  false
 Setting debug for  message_builder                 to  false
 Setting debug for  client_session                  to  false
 Setting debug for  opcua_client                    to  false
 Setting debug for  client_subscription             to  false
 Setting debug for  node_crawler                    to  false
 Setting debug for  base_server                     to  false
 Setting debug for  ua_variable                     to  false
 Setting debug for  ua_variable_type                to  false
 Setting debug for  condition                       to  false
 Setting debug for  argument_list                   to  false
 Setting debug for  opcua_server                    to  false
 Setting debug for  subscription                    to  false
 Setting debug for  monitored_item                  to  false
 Setting debug for  server_publish_engine           to  false
 Setting debug for  server_session                  to  false
 Setting debug for  server_engine                   to  false
 Setting debug for  server_end_point                to  false
 Setting debug for  server_secure_channel_layer     to  false
2018-03-02T12:56:40.245Z opcuaIIoT:core LE
2018-03-02T12:56:40.245Z opcuaIIoT:core ip-172-26-5-24
2018-03-02T12:56:40.245Z opcuaIIoT:core linux
2018-03-02T12:56:40.245Z opcuaIIoT:core Linux
2018-03-02T12:56:40.245Z opcuaIIoT:core x64
2 Mar 12:56:40 - [info] Dashboard version 2.8.1 started at /ui
2 Mar 12:56:40 - [warn] ------------------------------------------------------
2 Mar 12:56:40 - [warn] [node-red/rpi-gpio] Info : Ignoring Raspberry Pi specific node
2 Mar 12:56:40 - [warn] ------------------------------------------------------
2 Mar 12:56:40 - [info] Settings file  : /home/bitnami/.node-red/settings.js
2 Mar 12:56:40 - [info] User directory : /home/bitnami/.node-red
2 Mar 12:56:40 - [warn] Projects disabled : set editorTheme.projects.enabled=true to enable
2 Mar 12:56:40 - [info] Flows file     : /home/bitnami/.node-red/flows_ip-172-26-5-24.json
2 Mar 12:56:40 - [info] Server now running at http://127.0.0.1:1880/
2 Mar 12:56:40 - [info] Starting flows
2018-03-02T12:56:40.668Z opcuaIIoT:core /home/bitnami/.node-red/node_modules/node-opcua-client
2018-03-02T12:56:40.668Z opcuaIIoT:connector:details config: undefined
2018-03-02T12:56:40.668Z opcuaIIoT:connector:details config: undefined
2018-03-02T12:56:40.671Z opcuaIIoT:inject injecting once at start delay timeout 10000 msec.
2018-03-02T12:56:40.677Z opcuaIIoT:core /home/bitnami/.node-red/node_modules/node-opcua-server
2018-03-02T12:56:40.677Z opcuaIIoT:server:details config: undefined
2018-03-02T12:56:40.677Z opcuaIIoT:server:details config: undefined
2018-03-02T12:56:40.678Z opcuaIIoT:server node set:/home/bitnami/.node-red/node_modules/node-opcua-nodesets/nodesets/Opc.Ua.NodeSet2.xml,/home/bitnami/.node-red/node_modules/node-red-contrib-iiot-opcua/opcuaIIoT/public/vendor/opc-foundation/xml/Opc.Ua.Di.NodeSet2.xml
2018-03-02T12:56:40.790Z opcuaIIoT:server:details serverOptions:{"port":"55390","nodeset_filename":["/home/bitnami/.node-red/node_modules/node-opcua-nodesets/nodesets/Opc.Ua.NodeSet2.xml","/home/bitnami/.node-red/node_modules/node-red-contrib-iiot-opcua/opcuaIIoT/public/vendor/opc-foundation/xml/Opc.Ua.Di.NodeSet2.xml"],"resourcePath":"UA/NodeREDFlexIIoTServer","buildInfo":{"productName":"Node-RED Flex IIoT Server","buildNumber":"24122017","buildDate":"2018-01-24T00:00:00.000Z"},"serverCapabilities":{"operationLimits":{"maxNodesPerRead":1000,"maxNodesPerBrowse":2000}},"serverInfo":{"applicationUri":"urn:ip-172-26-5-24:NodeRED-Flex-IIoT-Server","productUri":"NodeRED-Flex-IIoT-Server","applicationName":{"text":"NodeRED","locale":"en"},"gatewayServerUri":null,"discoveryProfileUri":null,"discoveryUrls":[]},"maxAllowedSessionNumber":10,"maxConnectionsPerEndpoint":10,"allowAnonymous":true,"alternateHostname":"","userManager":{},"isAuditing":false}
2 Mar 12:56:40 - [info] Started flows
2018-03-02T12:56:41.707Z opcuaIIoT:server:flex Construct Address Space From Script
2018-03-02T12:56:41.709Z opcuaIIoT:server init dynamic address space
2 Mar 12:56:41 - [warn] [OPCUA-IIoT-Flex-Server:8238e194.49a66] construct new address space for OPC UA
xxxx VALUE =  { [Number: 11] key: 'Double', value: 11 }
2 Mar 12:56:41 - [red] Uncaught Exception:
2 Mar 12:56:41 - AssertionError [ERR_ASSERTION]: this.identifierType
    at new NodeId (/home/bitnami/.node-red/node_modules/node-opcua-nodeid/src/nodeid.js:70:5)
    at coerceNodeId (/home/bitnami/.node-red/node_modules/node-opcua-nodeid/src/nodeid.js:280:16)
    at resolveNodeId (/home/bitnami/.node-red/node_modules/node-opcua-nodeid/src/nodeid.js:384:18)
    at AddressSpace._coerce_Type (/home/bitnami/.node-red/node_modules/node-opcua-address-space/src/address_space.js:751:18)
    at AddressSpace._coerce_DataType (/home/bitnami/.node-red/node_modules/node-opcua-address-space/src/address_space.js:776:17)
    at AddressSpace._addVariable (/home/bitnami/.node-red/node_modules/node-opcua-address-space/src/address_space.js:845:29)
    at AddressSpace.addVariable (/home/bitnami/.node-red/node_modules/node-opcua-address-space/src/address_space.js:902:17)
    at Object.apply (/home/bitnami/.node-red/node_modules/vm2/lib/contextify.js:288:34)
    at constructAlarmAddressSpace (vm.js:122:43)
    at Object.apply (/home/bitnami/.node-red/node_modules/vm2/lib/contextify.js:87:36)
`

biancode, 2018-03-02:
Please, update to 2.0.12 first!

sixdiamants, 2018-03-04:
Hello, it seems that the update did the trick. It seems however that the palette doesn't reliably reflect the installed versions because I had actually installed the update from the palette (rather then by running npm manually). Only after doing a restart of node-red did things start working.

Another thing that got me, was the server configuration. I'm running a Ubuntu image on AWS and it _seemed_ that it started working after connecting to the endpoint opc.tcp://xxx:55388/UA/NodeREDIIoTServer

biancode, 2018-03-05:
please test with v2.1.0

closed 2018-03-05 by biancode