BiancoRoyal/node-red-contrib-iiot-opcua #16: Cannot delete original objects from server

tsangiotis, 2018-03-03:
Hello,

I cannot figure out how to delete the original example objects from the server.

- Bianco Royal
- Device Set
- Server*
- Tanks

* I assume _Server_ is needed for OPC UA to work.

biancode, 2018-03-03:
Yes, Device Set and Server are part of the OPC UA server and not to delete.

delete the Bianco Royal and Tanks folder via Command :

[image]

tsangiotis, 2018-03-03:
It seems like I cannot get this to work.

Here are screenshots of my setup:

[image]
[image]
[image]

biancode, 2018-03-03:
Please, send the flow or an example.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

closed 2018-03-05 by biancode