BiancoRoyal/node-red-contrib-iiot-opcua #17: BadIdentityTokenInvalid (0x80200000) using securityPolicy:NONE and securityMode:None

giminni, 2018-03-06:
Hi Klaus,

I am testing your latest lib 2.1.0 on node-red 0.18.4 with node 6.10.3 and npm 3.10.10.

Now I solved my opcua discovery problem (It was located on the opcua server side)
I enabled Basic256 with Sign and I could not create a listener and connect the listener
successfully.

Please find attached the log file, look at the end.

`
2018-03-06T14:07:39.913Z opcuaIIoT:core /data/red/node_modules/node-opcua-client
2018-03-06T14:07:39.914Z opcuaIIoT:connector:details config:
2018-03-06T14:07:39.914Z opcuaIIoT:connector:details default key: /data/red/node_modules/node-opcua-client/certificates/client_selfsigned_cert_1024.pem
2018-03-06T14:07:39.914Z opcuaIIoT:connector:details config:
2018-03-06T14:07:39.915Z opcuaIIoT:connector:details default key: /data/red/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem
2018-03-06T14:07:39.923Z opcuaIIoT:inject injecting once at start delay timeout 5000 msec.
2018-03-06T14:07:39.927Z opcuaIIoT:inject injecting once at start delay timeout 5000 msec.
6 Mar 14:07:39 - [info] Started flows
2018-03-06T14:07:41.917Z opcuaIIoT:connector Connecting On opc.tcp://10.0.1.127:4840/
2018-03-06T14:07:41.918Z opcuaIIoT:connector:details Options {"securityPolicy":"None","securityMode":"NONE","defaultSecureTokenLifetime":"0","keepSessionAlive":true,"certificateFile":"/data/red/node_modules/node-opcua-client/certificates/client_selfsigned_cert_1024.pem","privateKeyFile":"/data/red/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem","endpoint_must_exist":false}
2018-03-06T14:07:41.947Z opcuaIIoT:connector Connected On opc.tcp://10.0.1.127:4840/ Options {"securityPolicy":"None","securityMode":"NONE","defaultSecureTokenLifetime":"0","keepSessionAlive":true,"certificateFile":"/data/red/node_modules/node-opcua-client/certificates/client_selfsigned_cert_1024.pem","privateKeyFile":"/data/red/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem","endpoint_must_exist":false}
2018-03-06T14:07:41.954Z opcuaIIoT:connector endpoint: opc.tcp://10.0.1.127:4840
2018-03-06T14:07:41.954Z opcuaIIoT:connector urn:10.0.1.127
2018-03-06T14:07:41.954Z opcuaIIoT:connector:details NONE
2018-03-06T14:07:41.955Z opcuaIIoT:connector:details securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#None
2018-03-06T14:07:41.955Z opcuaIIoT:connector:details endpoint: opc.tcp://10.0.1.127:4840
2018-03-06T14:07:41.955Z opcuaIIoT:connector:details Application URI: urn:10.0.1.127
2018-03-06T14:07:41.955Z opcuaIIoT:connector:details Product URI: Loytec OPC UA Server
2018-03-06T14:07:41.955Z opcuaIIoT:connector:details Application Name: Loytec OPC UA Server
2018-03-06T14:07:41.956Z opcuaIIoT:connector:details Type: SERVER
2018-03-06T14:07:41.956Z opcuaIIoT:connector:details NONE
2018-03-06T14:07:41.963Z opcuaIIoT:connector:details securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#None
2018-03-06T14:07:41.963Z opcuaIIoT:connector:details discoveryUrls: opc.tcp://10.0.1.127:4840
2018-03-06T14:07:41.964Z opcuaIIoT:connector:details serverCertificate: None
2018-03-06T14:07:41.964Z opcuaIIoT:connector:details Identify Token for : Security Mode=NONE Policy=http://opcfoundation.org/UA/SecurityPolicy#None
2018-03-06T14:07:41.965Z opcuaIIoT:connector:details policyId: NoIdentityTokens
2018-03-06T14:07:41.965Z opcuaIIoT:connector:details tokenType: ANONYMOUS
2018-03-06T14:07:41.965Z opcuaIIoT:connector:details issuedTokenType: null
2018-03-06T14:07:41.965Z opcuaIIoT:connector:details issuerEndpointUrl: null
2018-03-06T14:07:41.965Z opcuaIIoT:connector:details securityPolicyUri: null
2018-03-06T14:07:41.966Z opcuaIIoT:connector Setup Certified Options For opc.tcp://10.0.1.127:4840/
2018-03-06T14:07:41.967Z opcuaIIoT:connector:details Options {"securityPolicy":"None","securityMode":"NONE","defaultSecureTokenLifetime":"0","keepSessionAlive":true,"certificateFile":"/data/red/node_modules/node-opcua-client/certificates/client_selfsigned_cert_1024.pem","privateKeyFile":"/data/red/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem","endpoint_must_exist":false}
2018-03-06T14:07:41.975Z opcuaIIoT:connector !!!!!!!!!!!!!!!!!!!!!!!!  CONNECTION CLOSED !!!!!!!!!!!!!!!!!!!
2018-03-06T14:07:41.976Z opcuaIIoT:connector Disconnected From opc.tcp://10.0.1.127:4840/
2018-03-06T14:07:41.976Z opcuaIIoT:connector:details Options {"securityPolicy":"None","securityMode":"NONE","defaultSecureTokenLifetime":"0","certificateFile":"/data/red/node_modules/node-opcua-client/certificates/client_selfsigned_cert_1024.pem","privateKeyFile":"/data/red/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem","endpoint_must_exist":false}
2018-03-06T14:07:42.008Z opcuaIIoT:connector Secured Connected On opc.tcp://10.0.1.127:4840/
2018-03-06T14:07:42.009Z opcuaIIoT:connector:details Options {"securityPolicy":"None","securityMode":"NONE","defaultSecureTokenLifetime":"0","certificateFile":"/data/red/node_modules/node-opcua-client/certificates/client_selfsigned_cert_1024.pem","privateKeyFile":"/data/red/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem","endpoint_must_exist":false}
2018-03-06T14:07:42.010Z opcuaIIoT:listener starting OPC UA session with delay of 1000
2018-03-06T14:07:42.010Z opcuaIIoT:listener starting OPC UA session with delay of 1000
2018-03-06T14:07:43.010Z opcuaIIoT:listener Listener Start OPC UA Session
2018-03-06T14:07:43.011Z opcuaIIoT:connector Request For New Session From Listener Node
2018-03-06T14:07:43.028Z opcuaIIoT:listener Listener Start OPC UA Session
2018-03-06T14:07:43.029Z opcuaIIoT:connector Request For New Session From Listener Node
2018-03-06T14:07:43.052Z opcuaIIoT:connector Session Start Error Error:  ServiceResult is BadIdentityTokenInvalid (0x80200000)
2018-03-06T14:07:43.053Z opcuaIIoT:listener Handle Session Error Error:  ServiceResult is BadIdentityTokenInvalid (0x80200000)
2018-03-06T14:07:43.066Z opcuaIIoT:connector Session Start Error Error:  ServiceResult is BadIdentityTokenInvalid (0x80200000)
2018-03-06T14:07:43.067Z opcuaIIoT:listener Handle Session Error Error:  ServiceResult is BadIdentityTokenInvalid (0x80200000)
2018-03-06T14:07:44.925Z opcuaIIoT:inject injecting once at start
2018-03-06T14:07:44.926Z opcuaIIoT:listener:details subscribe listener input {"topic":"","nodetype":"inject","injectType":"listen","addressSpaceItems":[{"name":"State R09L1","nodeId":"ns=1;i=476976","datatypeName":"Boolean"},{"name":"State R09L2","nodeId":"ns=1;i=477392","datatypeName":"Boolean"},{"name":"State R02L1","nodeId":"ns=1;i=517088","datatypeName":"Boolean"}],"payload":{"interval":500,"queueSize":1000},"_msgid":"aa6f9780.082788"}
2018-03-06T14:07:44.927Z opcuaIIoT:listener create monitoring subscription
2018-03-06T14:07:44.927Z opcuaIIoT:listener:subscribe Subscription Session Not Valid`

biancode, 2018-03-06:
Please use node 8 or better 9

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

giminni, 2018-03-06:
I must use node 6 because I have other libs that are not ready to be migrated now.

closed 2018-03-06 by biancode