BiancoRoyal/node-red-contrib-iiot-opcua #18: BadIdentityTokenInvalid (0x80200000) using securityPolicy:Basic256 and securityMode:Sign

giminni, 2018-03-06:
Klaus,

another strange thing. If I test the iiot connector with securityPolicy:Basic256 and securityMode:SIGN I get the same error as using securityPolicy:None and securityMode:NONE.

> Now my questions:
- Why will the first connection always close?
- Why are the certificate and private key fields displayed on Policy:NONE Security:None?
- Why is the default tokenId set to USERNAME if the username checkbox is not checked? 

Please find attached the detailed log output:
`2018-03-06T15:51:09.576Z opcuaIIoT:core /data/red/node_modules/node-opcua-client
2018-03-06T15:51:09.579Z opcuaIIoT:connector:details config:
2018-03-06T15:51:09.579Z opcuaIIoT:connector:details default key: /data/red/node_modules/node-opcua-client/certificates/client_selfsigned_cert_1024.pem
2018-03-06T15:51:09.580Z opcuaIIoT:connector:details config:
2018-03-06T15:51:09.580Z opcuaIIoT:connector:details default key: /data/red/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem
2018-03-06T15:51:09.612Z opcuaIIoT:inject injecting once at start delay timeout 5000 msec.
2018-03-06T15:51:09.621Z opcuaIIoT:inject injecting once at start delay timeout 5000 msec.
6 Mar 15:51:09 - [info] Started flows
2018-03-06T15:51:11.593Z opcuaIIoT:connector Connecting On opc.tcp://10.0.1.127:4840/
2018-03-06T15:51:11.594Z opcuaIIoT:connector:details Options {"securityPolicy":"Basic256","securityMode":"SIGN","defaultSecureTokenLifetime":60000,"keepSessionAlive":true,"certificateFile":"/data/red/node_modules/node-opcua-client/certificates/client_selfsigned_cert_1024.pem","privateKeyFile":"/data/red/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem","endpoint_must_exist":false}
2018-03-06T15:51:11.914Z opcuaIIoT:connector Connected On opc.tcp://10.0.1.127:4840/ Options {"securityPolicy":"Basic256","securityMode":"SIGN","defaultSecureTokenLifetime":60000,"keepSessionAlive":true,"certificateFile":"/data/red/node_modules/node-opcua-client/certificates/client_selfsigned_cert_1024.pem","privateKeyFile":"/data/red/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem","endpoint_must_exist":false}
2018-03-06T15:51:11.927Z opcuaIIoT:connector endpoint: opc.tcp://10.0.1.127:4840
2018-03-06T15:51:11.927Z opcuaIIoT:connector urn:10.0.1.127
2018-03-06T15:51:11.928Z opcuaIIoT:connector:details SIGNANDENCRYPT
2018-03-06T15:51:11.928Z opcuaIIoT:connector:details securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#Basic256
2018-03-06T15:51:11.935Z opcuaIIoT:connector:details endpoint: opc.tcp://10.0.1.127:4840
2018-03-06T15:51:11.935Z opcuaIIoT:connector:details Application URI: urn:10.0.1.127
2018-03-06T15:51:11.935Z opcuaIIoT:connector:details Product URI: Loytec OPC UA Server
2018-03-06T15:51:11.936Z opcuaIIoT:connector:details Application Name: Loytec OPC UA Server
2018-03-06T15:51:11.936Z opcuaIIoT:connector:details Type: SERVER
2018-03-06T15:51:11.937Z opcuaIIoT:connector:details SIGNANDENCRYPT
2018-03-06T15:51:11.937Z opcuaIIoT:connector:details securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#Basic256
2018-03-06T15:51:11.938Z opcuaIIoT:connector:details discoveryUrls: opc.tcp://10.0.1.127:4840
2018-03-06T15:51:11.957Z opcuaIIoT:connector:details serverCertificate: <obfuscated>
2018-03-06T15:51:11.957Z opcuaIIoT:core /data/red/node_modules/node-opcua-client
2018-03-06T15:51:11.958Z opcuaIIoT:connector:details /data/red/node_modules/node-opcua-client/certificates/PKI/server_certificate0.pem
2018-03-06T15:51:11.963Z opcuaIIoT:connector:details endpoint: opc.tcp://10.0.1.127:4840
2018-03-06T15:51:11.963Z opcuaIIoT:connector:details Application URI: urn:10.0.1.127
2018-03-06T15:51:11.964Z opcuaIIoT:connector:details Product URI: Loytec OPC UA Server
2018-03-06T15:51:11.964Z opcuaIIoT:connector:details Application Name: Loytec OPC UA Server
2018-03-06T15:51:11.965Z opcuaIIoT:connector:details Type: SERVER
2018-03-06T15:51:11.965Z opcuaIIoT:connector:details SIGN
2018-03-06T15:51:11.966Z opcuaIIoT:connector:details securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#Basic256
2018-03-06T15:51:11.966Z opcuaIIoT:connector:details discoveryUrls: opc.tcp://10.0.1.127:4840
2018-03-06T15:51:11.974Z opcuaIIoT:connector:details serverCertificate: <obfuscated>
2018-03-06T15:51:11.975Z opcuaIIoT:core /data/red/node_modules/node-opcua-client
2018-03-06T15:51:11.975Z opcuaIIoT:connector:details /data/red/node_modules/node-opcua-client/certificates/PKI/server_certificate1.pem
2018-03-06T15:51:11.978Z opcuaIIoT:connector:details Identify Token for : Security Mode=SIGNANDENCRYPT Policy=http://opcfoundation.org/UA/SecurityPolicy#Basic256
2018-03-06T15:51:11.979Z opcuaIIoT:connector:details policyId: Username/Password
2018-03-06T15:51:11.981Z opcuaIIoT:connector:details tokenType: USERNAME
2018-03-06T15:51:11.981Z opcuaIIoT:connector:details issuedTokenType: null
2018-03-06T15:51:11.982Z opcuaIIoT:connector:details issuerEndpointUrl: null
2018-03-06T15:51:11.983Z opcuaIIoT:connector:details securityPolicyUri: null
2018-03-06T15:51:11.984Z opcuaIIoT:connector:details Identify Token for : Security Mode=SIGN Policy=http://opcfoundation.org/UA/SecurityPolicy#Basic256
2018-03-06T15:51:11.985Z opcuaIIoT:connector:details policyId: Username/Password
2018-03-06T15:51:11.985Z opcuaIIoT:connector:details tokenType: USERNAME
2018-03-06T15:51:11.986Z opcuaIIoT:connector:details issuedTokenType: null
2018-03-06T15:51:11.986Z opcuaIIoT:connector:details issuerEndpointUrl: null
2018-03-06T15:51:11.987Z opcuaIIoT:connector:details securityPolicyUri: null
2018-03-06T15:51:11.990Z opcuaIIoT:connector Setup Certified Options For opc.tcp://10.0.1.127:4840/
2018-03-06T15:51:11.993Z opcuaIIoT:connector:details Options {"securityPolicy":"Basic256","securityMode":"SIGN","defaultSecureTokenLifetime":60000,"keepSessionAlive":true,"certificateFile":"/data/red/node_modules/node-opcua-client/certificates/client_selfsigned_cert_1024.pem","privateKeyFile":"/data/red/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem","endpoint_must_exist":false,"serverCertificate":{"type":"Buffer","data":<obfuscated>
2018-03-06T15:51:12.013Z opcuaIIoT:connector !!!!!!!!!!!!!!!!!!!!!!!!  CONNECTION CLOSED !!!!!!!!!!!!!!!!!!!
2018-03-06T15:51:12.019Z opcuaIIoT:connector Disconnected From opc.tcp://10.0.1.127:4840/
2018-03-06T15:51:12.021Z opcuaIIoT:connector:details Options {"securityPolicy":"Basic256","securityMode":"SIGN","defaultSecureTokenLifetime":60000,"certificateFile":"/data/red/node_modules/node-opcua-client/certificates/client_selfsigned_cert_1024.pem","privateKeyFile":"/data/red/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem","endpoint_must_exist":false,"serverCertificate":{"type":"Buffer","data":<obfuscated>
2018-03-06T15:51:12.109Z opcuaIIoT:connector:details The certificate file /data/red/node_modules/node-opcua-client/certificates/PKI/server_certificate0.pem has been saved!
2018-03-06T15:51:12.118Z opcuaIIoT:connector:details The certificate file /data/red/node_modules/node-opcua-client/certificates/PKI/server_certificate1.pem has been saved!
2018-03-06T15:51:12.859Z opcuaIIoT:connector Secured Connected On opc.tcp://10.0.1.127:4840/
2018-03-06T15:51:12.862Z opcuaIIoT:connector:details Options {"securityPolicy":"Basic256","securityMode":"SIGN","defaultSecureTokenLifetime":60000,"certificateFile":"/data/red/node_modules/node-opcua-client/certificates/client_selfsigned_cert_1024.pem","privateKeyFile":"/data/red/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem","endpoint_must_exist":false,"serverCertificate":{"type":"Buffer","data":<obfuscated>
2018-03-06T15:51:12.864Z opcuaIIoT:listener starting OPC UA session with delay of 1000
2018-03-06T15:51:12.865Z opcuaIIoT:listener starting OPC UA session with delay of 1000
2018-03-06T15:51:13.868Z opcuaIIoT:listener Listener Start OPC UA Session
2018-03-06T15:51:13.871Z opcuaIIoT:connector Request For New Session From Listener Node
2018-03-06T15:51:13.934Z opcuaIIoT:listener Listener Start OPC UA Session
2018-03-06T15:51:13.936Z opcuaIIoT:connector Request For New Session From Listener Node
2018-03-06T15:51:14.340Z opcuaIIoT:connector Session Start Error Error:  ServiceResult is BadIdentityTokenInvalid (0x80200000)
2018-03-06T15:51:14.342Z opcuaIIoT:listener Handle Session Error Error:  ServiceResult is BadIdentityTokenInvalid (0x80200000)
2018-03-06T15:51:14.348Z opcuaIIoT:connector Session Start Error Error:  ServiceResult is BadIdentityTokenInvalid (0x80200000)
2018-03-06T15:51:14.349Z opcuaIIoT:listener Handle Session Error Error:  ServiceResult is BadIdentityTokenInvalid (0x80200000)`

biancode, 2018-03-06:
Please, try to build a very new flow - does it happen with a whole new flow and connector too?
Check the configuration to delete unused connectors in the background!

biancode, 2018-03-06:
happens with none too

giminni, 2018-03-08:
Sorry problem persist using NONE or using Basic256 after deleted every thing.  
I cannot understand why I need certificates if the mode is NONE and user anonymous, why is trying to use the default none. Or can you tell me how to eliminate the default keys if I am trying to use NONE. Please advise. ATM I am running out of ideas.

biancode, 2018-03-08:
In none Mode there is no key in use 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

biancode, 2018-03-08:
I updatet the prosys server to 2.3 and it works well - 2.0 of prosys Server doesn't work

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

giminni, 2018-03-08:
I am using a real OPC/UA server from Loytec Gateway 952
https://www.loytec.com/de/produkte/gateways/lgate-95x

BTW: I found that you use the certificates even in NONE mode, I guess for NONE there might not be any 
certificates right?

This behavior worked with 1.3.3

> let nodeOPCUAClientPath = coreConnector.core.getNodeOPCUAClientPath()
    coreConnector.detailDebugLog('config: ' + node.publicCertificateFile)
    if (node.publicCertificateFile === null || node.publicCertificateFile === '') {
      node.publicCertificateFile = path.join(nodeOPCUAClientPath, '/certificates/client_selfsigned_cert_1024.pem')
      coreConnector.detailDebugLog('default key: ' + node.publicCertificateFile)
    }
    coreConnector.detailDebugLog('config: ' + node.privateKeyFile)
    if (node.privateKeyFile === null || node.privateKeyFile === '') {
      node.privateKeyFile = path.join(nodeOPCUAClientPath, '/certificates/PKI/own/private/private_key.pem')
      coreConnector.detailDebugLog('default key: ' + node.privateKeyFile)
    }
    node.opcuaClientOptions = {
      securityPolicy: node.securityPolicy || 'None',
      securityMode: node.messageSecurityMode || 'NONE',
      defaultSecureTokenLifetime: node.defaultSecureTokenLifetime,
      keepSessionAlive: true,
      certificateFile: node.publicCertificateFile,
      privateKeyFile: node.privateKeyFile,
      endpoint_must_exist: node.endpointMusExist
    }

biancode, 2018-03-09:
```
      publicCertificateFile: {value: ''},
      privateKeyFile: {value: ''},
```

The keys are null - the UI shows just placeholders ;)

```
            <div class="form-row">
                <label for="node-config-input-publicCertificateFile"><i class="fa fa-file"></i>
                <span data-i18n="opcua-iiot-contrib.label.publicCertificateFile"></span></label>
                <input type="text" id="node-config-input-publicCertificateFile" placeholder="./certificates/client_selfsigned_cert_1024.pem" style="min-width:480px">
            </div>
            <div class="form-row">
                <label for="node-config-input-privateKeyFile"><i class="fa fa-file"></i>
                <span data-i18n="opcua-iiot-contrib.label.privateKeyFile"></span></label>
                <input type="text" id="node-config-input-privateKeyFile" placeholder="./certificates/PKI/own/private/private_key.pem" style="min-width:480px">
            </div>
```

biancode, 2018-03-09:
please test v2.1.1

giminni, 2018-03-10:
Sorry Klaus,
even if certificates are null, same error
first connect fails, second connect try to start a session ending with the same error:
**BadIdentityTokenInvalid (0x80200000)**

> 10 Mar 00:57:26 - [info] Starting flows
2018-03-10T00:57:26.935Z opcuaIIoT:core /data/red/node_modules/node-opcua-client
2018-03-10T00:57:26.936Z opcuaIIoT:connector:details config:
2018-03-10T00:57:26.937Z opcuaIIoT:connector:details config:
2018-03-10T00:57:26.937Z opcuaIIoT:connector:details securedCommunication: false
10 Mar 00:57:26 - [info] Started flows
2018-03-10T00:57:28.940Z opcuaIIoT:connector Connecting On opc.tcp://10.0.1.123:4840
2018-03-10T00:57:28.940Z opcuaIIoT:connector:details Options {"securityPolicy":"None","securityMode":"NONE","defaultSecureTokenLifetime":"60000","keepSessionAlive":true,"certificateFile":null,"privateKeyFile":null,"endpoint_must_exist":false}
2018-03-10T00:57:28.988Z opcuaIIoT:connector Connected On opc.tcp://10.0.1.123:4840 Options {"securityPolicy":"None","securityMode":"NONE","defaultSecureTokenLifetime":"60000","keepSessionAlive":true,"certificateFile":null,"privateKeyFile":null,"endpoint_must_exist":false}
2018-03-10T00:57:29.001Z opcuaIIoT:connector endpoint: opc.tcp://10.0.1.123:4840
2018-03-10T00:57:29.001Z opcuaIIoT:connector urn:10.0.1.123
2018-03-10T00:57:29.002Z opcuaIIoT:connector:details NONE
2018-03-10T00:57:29.002Z opcuaIIoT:connector:details securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#None
2018-03-10T00:57:29.002Z opcuaIIoT:connector:details endpoint: opc.tcp://10.0.1.123:4840
2018-03-10T00:57:29.003Z opcuaIIoT:connector:details Application URI: urn:10.0.1.123
2018-03-10T00:57:29.003Z opcuaIIoT:connector:details Product URI: Loytec OPC UA Server
2018-03-10T00:57:29.003Z opcuaIIoT:connector:details Application Name: Loytec OPC UA Server
2018-03-10T00:57:29.004Z opcuaIIoT:connector:details Type: SERVER
2018-03-10T00:57:29.004Z opcuaIIoT:connector:details NONE
2018-03-10T00:57:29.005Z opcuaIIoT:connector:details securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#None
2018-03-10T00:57:29.005Z opcuaIIoT:connector:details discoveryUrls: opc.tcp://10.0.1.123:4840
2018-03-10T00:57:29.005Z opcuaIIoT:connector:details serverCertificate: None
2018-03-10T00:57:29.006Z opcuaIIoT:connector:details Identify Token for : Security Mode=NONE Policy=http://opcfoundation.org/UA/SecurityPolicy#None
2018-03-10T00:57:29.006Z opcuaIIoT:connector:details policyId: NoIdentityTokens
2018-03-10T00:57:29.006Z opcuaIIoT:connector:details tokenType: ANONYMOUS
2018-03-10T00:57:29.007Z opcuaIIoT:connector:details issuedTokenType: null
2018-03-10T00:57:29.007Z opcuaIIoT:connector:details issuerEndpointUrl: null
2018-03-10T00:57:29.007Z opcuaIIoT:connector:details securityPolicyUri: null
2018-03-10T00:57:29.009Z opcuaIIoT:connector Setup Certified Options For opc.tcp://10.0.1.123:4840
2018-03-10T00:57:29.009Z opcuaIIoT:connector:details Options {"securityPolicy":"None","securityMode":"NONE","defaultSecureTokenLifetime":"60000","keepSessionAlive":true,"certificateFile":null,"privateKeyFile":null,"endpoint_must_exist":false}
2018-03-10T00:57:29.019Z opcuaIIoT:connector !!!!!!!!!!!!!!!!!!!!!!!!  CONNECTION CLOSED !!!!!!!!!!!!!!!!!!!
2018-03-10T00:57:29.022Z opcuaIIoT:connector Disconnected From opc.tcp://10.0.1.123:4840
2018-03-10T00:57:29.023Z opcuaIIoT:connector:details Options {"securityPolicy":"None","securityMode":"NONE","defaultSecureTokenLifetime":"60000","certificateFile":null,"privateKeyFile":null,"endpoint_must_exist":false}
2018-03-10T00:57:29.070Z opcuaIIoT:connector Secured Connected On opc.tcp://10.0.1.123:4840
2018-03-10T00:57:29.071Z opcuaIIoT:connector:details Options {"securityPolicy":"None","securityMode":"NONE","defaultSecureTokenLifetime":"60000","certificateFile":"/data/red/node_modules/node-opcua-client/certificates/client_selfsigned_cert_1024.pem","privateKeyFile":"/data/red/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem","endpoint_must_exist":false}
2018-03-10T00:57:29.072Z opcuaIIoT:client:read starting OPC UA session with delay of 1000
2018-03-10T00:57:30.073Z opcuaIIoT:client:read Read Start OPC UA Session
2018-03-10T00:57:30.074Z opcuaIIoT:connector Request For New Session From Read Node
2018-03-10T00:57:30.122Z opcuaIIoT:connector Session Start Error Error:  ServiceResult is BadIdentityTokenInvalid (0x80200000)
2018-03-10T00:57:30.123Z opcuaIIoT:client:read Handle Session Error Error:  ServiceResult is BadIdentityTokenInvalid (0x80200000)

biancode, 2018-03-10:
a) Did you test with the [opcua-commander](https://github.com/node-opcua/opcua-commander)? Does it work?

b) Did you try to set up a whole new flow or is it a modified example? Try with whole new flow, please!

c) How did you update via npm? (update/install) Did you try to uninstall and install again?

d) Please, search for issues in [node-opcua repo](https://github.com/node-opcua/node-opcua)!

e) Is it possible to connect online from my Node-RED?

biancode, 2018-03-10:
[image]

Ckeck for that your keys are trusted in the server. Does it work with keys?

biancode, 2018-03-10:
some hint: also some servers need to restart to load trusted certificates

giminni, 2018-03-11:
Want to test first without certificates in anonymous mode, before going to add one. The debug log  shoes a second connection with certificates,then the session ends with badIdentityInvalidToken.
Again no certificates, no user. Tested with ProSys Client it works with None.

biancode, 2018-03-11:
test with v2.2.x

biancode, 2018-03-12:
For the v2.2.x test - build and wire a whole new flow, please!

biancode, 2018-03-12:
do a complete uninstall before, check that Node-RED is missing the nodes and than do a fresh install, please

giminni, 2018-03-12:
Hi Klaus,

fixed!

The connection works now for securityPolicy:NONE and securityMode:None

closed 2018-03-12 by giminni