BiancoRoyal/node-red-contrib-iiot-opcua #25: "Error: Session Not Valid On Write"

JoaoMCPereira, 2018-03-09:
I'm new to node-red and to OPC-UA. After installing node-red and this module I run a few examples and always got the same error message saying "Error: Session Not Valid On Write" 
I also copy the examples 'write.json', 'read.json' and 'server-demo.json' but same error message is showing up.
What I'm missing? I need to install something else besides 'node-red-node-data-generator'?

Thanks for the help

w4tsn, 2018-03-09:
My Issue #26 might be related since it throws the error `Error: Session Not Valid On Read`. As of my understanding the Connector-Node could not connect to the OPC-UA server, so the read/write nodes can't access the server endpoint.

What I had to do in order to get the examples to work was:

1. enable anonymous access in the server node
2. allow non-existing endpoint in the connector-node `LOCAL SERVER`

biancode, 2018-03-10:
[image]

or just disable the option endpoint must exist or set the right endpoint uri

biancode, 2018-03-11:
do not inject read, listen or write before server is ready - use the start delay with some seconds

closed 2018-03-11 by biancode