BiancoRoyal/node-red-contrib-iiot-opcua #26: Failure when using OPCUA-Server open62541

w4tsn, 2018-03-09:
When trying to use the `examples/read.json` with a locally running open62541 OPC-UA server I encounter several problems.

I started with the example server and client and just had to change the following to make it work:

1. on the server enable anonymous access
2. on the client allow non-existing endpoint

After that worked and a connection was established I changed the URL from `opc.tcp://localhost:55388/` to `opc.tcp://localhost:16664/`, which results in the following errors:

1. Read-Node: Error: Session Not Valid On Read
2. Read-Node: Error:  ServiceResult is BadIdentityTokenInvalid (0x80200000)
3. Connector-Node: Error stopping node: TimeoutError: timed out after 15000ms

The Server-Log says:

```
[03/09/2018 23:26:21.106] info/network	Connection 12 | New connection over TCP from 127.0.0.1:33367
[03/09/2018 23:26:21.118] info/channel	Connection 12 | SecureChannel 9 | OpenSecureChannel: Opened SecureChannel
[03/09/2018 23:26:21.133] info/session	Try to use Session with token 00000000-0000-0000-0000-000000000000 but is not found
[03/09/2018 23:26:22.223] info/session	Connection 0 | SecureChannel 0 | Session da6f9489-ceae-7380-2f12-2162924924c2 | ActivateSession: SecureChannel 9 wants to activate, but the UserIdentify token is invalid
[03/09/2018 23:26:41.013] info/network	Connection 13 | New connection over TCP from 127.0.0.1:33384
[03/09/2018 23:26:41.025] info/channel	Connection 13 | SecureChannel 10 | OpenSecureChannel: Opened SecureChannel
[03/09/2018 23:26:41.040] info/session	Try to use Session with token 00000000-0000-0000-0000-000000000000 but is not found
[03/09/2018 23:26:42.092] info/session	Connection 0 | SecureChannel 0 | Session f112a31c-8f1a-ba90-debd-7bd714a14a64 | ActivateSession: SecureChannel 10 wants to activate, but the UserIdentify token is invalid
```

## Versions

OPC related node packages:

```
"node-opcua": "^0.2.2",
"node-red-contrib-iiot-opcua": "^2.1.1"
```

open62541 in version 0.2

Node 8.9.4
NPM 5.6.0
Node-RED 0.17.5

## Related Issues

This issue from node-opcua might be related: https://github.com/node-opcua/node-opcua/issues/452

biancode, 2018-03-11:
please test again with v2.2.0

w4tsn, 2018-03-11:
## Tested with v2.2.1
With checked fields for `Endpoint Must Exist` and `Auto Select Endpoint`.

Unchecking `Endpoint Must Exist` results in the error `Connector-Node: Error stopping node: TimeoutError: timed out after 15000ms`

Now the read node states `connected` though it throws one "Error: Session Not Ready To Read" before that. Invoking a read for the node `current.time` results in the same error message, though the read node states connected.

## Error log of server is now

Note that after the read node states connected the server reports `Session has timed out` instead of "user token invalid".

```
03/11/2018 21:05:45.655] info/network	TCP network layer listening on opc.tcp://othermo-g0002.gateways.othermo.de:16664
[03/11/2018 21:06:00.105] info/network	Connection 4 | New connection over TCP from 127.0.0.1:35856
[03/11/2018 21:06:00.325] info/channel	Connection 4 | SecureChannel 1 | OpenSecureChannel: Opened SecureChannel
[03/11/2018 21:06:00.651] info/session	Try to use Session with token 00000000-0000-0000-0000-000000000000 but is not found
[03/11/2018 21:06:12.477] info/session	Try to use Session with token 00000000-0000-0000-0000-000000000000 but is not found
[03/11/2018 21:06:12.516] info/channel	Connection 4 | SecureChannel 1 | CloseSecureChannel
[03/11/2018 21:07:12.570] info/network	Connection 5 | New connection over TCP from 127.0.0.1:35874
[03/11/2018 21:07:12.583] info/channel	Connection 5 | SecureChannel 2 | OpenSecureChannel: Opened SecureChannel
[03/11/2018 21:07:12.593] info/session	Try to use Session with token 00000000-0000-0000-0000-000000000000 but is not found
[03/11/2018 21:07:12.630] info/session	Connection 0 | SecureChannel 0 | Session ddac81cb-e374-4dc7-7117-9139ad7a57e5 | ActivateSession: SecureChannel 2 wants to activate, but the UserIdentify token is invalid
[03/11/2018 21:07:53.892] info/session	Connection 0 | SecureChannel 0 | Session ddac81cb-e374-4dc7-7117-9139ad7a57e5 | Calling service 473 on a non-activated session
[03/11/2018 21:07:54.290] info/channel	Connection 5 | SecureChannel 2 | CloseSecureChannel
[03/11/2018 21:08:10.959] info/network	Connection 6 | New connection over TCP from 127.0.0.1:35881
[03/11/2018 21:08:10.982] info/channel	Connection 6 | SecureChannel 3 | OpenSecureChannel: Opened SecureChannel
[03/11/2018 21:08:10.997] info/session	Try to use Session with token 00000000-0000-0000-0000-000000000000 but is not found
[03/11/2018 21:08:11.018] info/session	Try to use Session with token 00000000-0000-0000-0000-000000000000 but is not found
[03/11/2018 21:08:11.035] info/channel	Connection 6 | SecureChannel 3 | CloseSecureChannel
[03/11/2018 21:09:11.059] info/network	Connection 7 | New connection over TCP from 127.0.0.1:35888
[03/11/2018 21:09:11.075] info/channel	Connection 7 | SecureChannel 4 | OpenSecureChannel: Opened SecureChannel
[03/11/2018 21:09:11.085] info/session	Try to use Session with token 00000000-0000-0000-0000-000000000000 but is not found
[03/11/2018 21:09:11.117] info/session	Connection 0 | SecureChannel 0 | Session ad65aa16-effd-9925-5e85-a8daabca5c35 | ActivateSession: SecureChannel 4 wants to activate, but the UserIdentify token is invalid
[03/11/2018 21:09:55.010] info/network	Connection 8 | New connection over TCP from 127.0.0.1:35893
[03/11/2018 21:09:55.023] info/channel	Connection 8 | SecureChannel 5 | OpenSecureChannel: Opened SecureChannel
[03/11/2018 21:09:55.038] info/session	Try to use Session with token 00000000-0000-0000-0000-000000000000 but is not found
[03/11/2018 21:09:55.046] info/session	Try to use Session with token 00000000-0000-0000-0000-000000000000 but is not found
[03/11/2018 21:09:55.053] info/channel	Connection 8 | SecureChannel 5 | CloseSecureChannel
[03/11/2018 21:10:15.639] info/session	Connection 0 | SecureChannel 0 | Session ad65aa16-effd-9925-5e85-a8daabca5c35 | Session has timed out
[03/11/2018 21:10:55.071] info/network	Connection 9 | New connection over TCP from 127.0.0.1:35900
[03/11/2018 21:10:55.085] info/channel	Connection 9 | SecureChannel 6 | OpenSecureChannel: Opened SecureChannel
[03/11/2018 21:10:55.101] info/session	Try to use Session with token 00000000-0000-0000-0000-000000000000 but is not found
[03/11/2018 21:10:55.143] info/session	Connection 0 | SecureChannel 0 | Session fc55010b-b12b-d5b7-0ad0-3d73b7dbfd1f | ActivateSession: SecureChannel 6 wants to activate, but the UserIdentify token is invalid
[03/11/2018 21:11:00.268] info/session	Connection 0 | SecureChannel 0 | Session fc55010b-b12b-d5b7-0ad0-3d73b7dbfd1f | Calling service 473 on a non-activated session
[03/11/2018 21:11:00.273] info/channel	Connection 9 | SecureChannel 6 | CloseSecureChannel
[03/11/2018 21:11:17.319] info/network	Connection 10 | New connection over TCP from 127.0.0.1:35903
[03/11/2018 21:11:17.332] info/channel	Connection 10 | SecureChannel 7 | OpenSecureChannel: Opened SecureChannel
[03/11/2018 21:11:17.348] info/session	Try to use Session with token 00000000-0000-0000-0000-000000000000 but is not found
[03/11/2018 21:11:17.356] info/session	Try to use Session with token 00000000-0000-0000-0000-000000000000 but is not found
[03/11/2018 21:11:17.363] info/channel	Connection 10 | SecureChannel 7 | CloseSecureChannel
[03/11/2018 21:11:45.698] info/network	Connection 11 | New connection over TCP from 127.0.0.1:35908
[03/11/2018 21:11:45.710] info/channel	Connection 11 | SecureChannel 8 | OpenSecureChannel: Opened SecureChannel
[... cut ...]
[03/11/2018 21:19:30.706] info/network	Connection 17 | New connection over TCP from 127.0.0.1:35961
[03/11/2018 21:19:30.718] info/channel	Connection 17 | SecureChannel 14 | OpenSecureChannel: Opened SecureChannel
[03/11/2018 21:19:30.732] info/session	Try to use Session with token 00000000-0000-0000-0000-000000000000 but is not found
[03/11/2018 21:19:30.757] info/session	Connection 0 | SecureChannel 0 | Session 13ea9a9f-566b-2132-d41d-d1cdf46fd6ad | ActivateSession: SecureChannel 14 wants to activate, but the UserIdentify token is invalid
```

w4tsn, 2018-03-12:
Like suggested in #29 I setup a complete new flow with a new connector and read/inject nodes. Still the same issue. The read node states connected or even active (one time, could not reproduce).

Does someone have any idea where to look next? Do you think it is a iiot-opcua error, a node-opcua client error, a open62541 server error or something in combination between the two/three? I'm a bit puzzled right now with this :S

biancode, 2018-03-12:
I'll test it with open62541. Best for that is if you could give me a setup tutorial to get the same environment for tests.

biancode, 2018-03-12:
@erossignon did you get some questions or experinces in that topic?

biancode, 2018-03-12:
@w4tsn if you test it with the [opcua-commander](https://github.com/node-opcua/opcua-commander) and it works, then it is my problem here. The connector is like the commander.

w4tsn, 2018-03-12:
## Test-Env
@biancode my co-worker 

1. downloaded the release 0.2 package for raspberrypi, 
2. unzipped it 
3. and started the plain example server without modifications

I replicated this setup on my linux 64 machine with the corresponding release package and it worked out of the box.

## Further tests
I tested to connect to open62541 with python3 opcu-client and opcua-commander.

With opcua-client I tested a locally running server on linux 64 as well as the ssh-piped server from the remote raspberrypi, which both worked. On the remote machine I installed opcua-commander and tried to connect to the local server but failed. The program exited right after start with the same server log messages as stated above like `timeout`.

w4tsn, 2018-03-13:
So, if this is an issue with the opcua-commander, I'll open an issue for connection to open62541 there. What do you think @biancode?

biancode, 2018-03-13:
Yes, please. It could be placed as a node-opcua issue too.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

biancode, 2018-03-18:
open62541 problem

closed 2018-03-18 by biancode

w4tsn, 2018-03-18:
Seems to be fixed in open62541. Should work in 0.3-rc2.