BiancoRoyal/node-red-contrib-iiot-opcua #28: Bug: Browse -> Read

araisch, 2018-03-12:
Hi Klaus,
checked your new update today.
Since Update 2.2.1 the Read Node doesn't work with Browse-Node anymore.
[image]
Output of Browser:

araisch, 2018-03-12:
The Listener doesn't work either:
Listener Input:
[image]
Listener Output:
On first Inject: Nothing
On second Inject:
[image]

biancode, 2018-03-12:
I am sorry, but there are breaking changes. Please, uninstall, install again and wire a whole new flow.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

biancode, 2018-03-12:
Set the correct options in browser.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

araisch, 2018-03-12:
Another funny thing:
If you browse two different folders in the same flow:
[image]
It adds up the result of both flows. 
Output of both messages:
[image]
so like you see: upper folder contains 409 nodes
So the "ns=2;s=ProzReg" contains 358 items, as you see in the posting above. It's the lower Browse-Result. But it gives back 767 items (409+358). Not really a bad bug, but could cause unwanted behavior..

This is not related to the update, was like that before. 
Ok I will uninstall and try again.

biancode, 2018-03-12:
What is also new is - now it has only one session per connector, because I got that hint from Etienne to change so.

biancode, 2018-03-12:
Please test first with the [whole new example flows](https://github.com/biancode/node-red-contrib-iiot-opcua/tree/master/examples)

araisch, 2018-03-12:
Strange that it doesn't update the used nodes..
But everything above works fine with fresh installation and flows.
Great Work, thank you 👍

biancode, 2018-03-12:
Yes, Node-RED does not update the outputs and properties and build strange mixed versions of nodes.

araisch, 2018-03-13:
## Browser Message
Browsers still sum up the results in payload, but now it doesn't influence the flow, because Listener/Read uses the other arrays ("nodeToRead" and "addressItemsToRead"):
[image]
Both Browsers have the same Node-ID (a folder with 5 Nodes), one fitted for Listener and one for Read.
Output of the upper Browser:
[image]
Output of the lower Browser:
[image]

## New Messages from Read/Listener
In the flow above are the Listener and the Read Node set to "Read just Values". The messages for the **same** Variable (Boolean-Type) contain:
Browser-Message: addressItemsToRead[0].datatypeName = "ns=0;i=63" (=VariableType: BaseDataVariableType)
Read-Message: payload[0].dataType = "ns=0;i=1" (DataType: Boolean, that's correct)
Listener-Message: addressSpaceItems[0].datatypeName = "" (empty)
Listener-Message: payload.value.dataType = "Boolean" (more or less correct, gives the dataTypeName)

Shouldn't be the datatypeName = "Boolean" and the dataType = "ns=0;i=1" in all cases?

Listener-Message: addressSpaceItems[0].name = "" (empty)
While BrowseName, DisplayName, Identifier of the Node aren't empty, what should be outputted here?

## Listener with NOT "Read just Values"
In the flow above are the Listener and the Read Node **not** set to "Read just Values".
It gives me this:

```
Exception raised inside the event handler called by ClientMonitoredItem.on('change') TypeError: Converting circular structure to JSON
    at JSON.stringify (<anonymous>)
    at Object.c.sendDataFromMonitoredItem (/home/gateway/.node-red/node_modules/node-red-contrib-iiot-opcua/opcuaIIoT/maps/opcua-iiot-listener.js:167:14)
    at ClientMonitoredItem.<anonymous> (/home/gateway/.node-red/node_modules/node-red-contrib-iiot-opcua/opcuaIIoT/maps/opcua-iiot-listener.js:139:10)
    at ClientMonitoredItem.emit (events.js:127:13)
    at ClientMonitoredItemBase._notify_value_change (/home/gateway/.node-red/node_modules/node-opcua-client/src/client_monitored_item_base.js:45:13)
    at /home/gateway/.node-red/node_modules/node-opcua-client/src/client_subscription.js:192:32
    at Array.forEach (<anonymous>)
    at ClientSubscription.__on_publish_response_DataChangeNotification (/home/gateway/.node-red/node_modules/node-opcua-client/src/client_subscription.js:184:20)
    at /home/gateway/.node-red/node_modules/node-opcua-client/src/client_subscription.js:297:26
    at Array.forEach (<anonymous>)
Please verify the application using this node-opcua client
```

biancode, 2018-03-13:
Did you check the examples? Why you are using two browsers? Try to use one browser and switch the options to ckecked for listen and read output at the same time.

biancode, 2018-03-13:
[image]

biancode, 2018-03-13:
New Messages from Read/Listener - that is from node-opcua output

biancode, 2018-03-13:
Listener with NOT "Read just Values" see issue #32

araisch, 2018-03-13:
It's not a flow used for a real purpose, I just build it to explain you all the stuff.
I need one browser for each folder and because of the limitation of 1000 nodes for one Listener.
Like I said, it's working, but still it should not add up the arrays of different Browser-Outputs in different paths of the flow.

For the dataType/dataTypeName mess: Ok, but you agree that it's not as intended?

biancode, 2018-03-16:
@araisch what about the browse and read in v2.3.1? Otherwise explain more what is expected here, please!

biancode, 2018-03-17:
v2.3.2

closed 2018-03-17 by biancode