BiancoRoyal/node-red-contrib-iiot-opcua #29: v2.2.x - breaking change information

biancode, 2018-03-12:
Dear users,

I'm sorry to do that breaking change with v2.2.x for all node outputs, but with that we are now closer to the changes of the node-opcua API and node-opcua outputs.

Please, enter your questions and problems here!
Thank you!
Klaus

biancode, 2018-03-12:
Please, test first with the [whole new example flows](https://github.com/biancode/node-red-contrib-iiot-opcua/tree/master/examples)

You'll find them under Node-RED > menu > Import > Examples > iiot opcua > ...

biancode, 2018-03-12:
**best practice for v2.2.x**

* uninstall
* check for Node-RED is misssing the package
* install again

You have to wire some nodes with a fresh drag&drop into the flow somtimes.

* server
* flex server
* read
* write
* listener
* result filter

gar-syn, 2018-03-14:
Klaus,

I'm having an issue with the listener node with the latest update. The addressSpaceItems name & datatypeName are not coming through. If you deselect send just values in the listener no message comes through at all. As you said above I have uninstalled the package, reinstalled and replaced the node. I also notice now that there is only a single output on the listener, is this as expected?

[image]

`{"payload":{"value":{"dataType":"Double","arrayType":"Scalar","value":0.0852},"statusCode":{"value":0,"description":"No Error","name":"Good"},"sourceTimestamp":"2018-03-14T08:32:43.554Z","sourcePicoseconds":0,"serverTimestamp":"2018-03-14T13:06:38.055Z","serverPicoseconds":0},"addressSpaceItems":[{"name":"","nodeId":"ns=8;i=1244","datatypeName":""}],"nodetype":"listen","injectType":"subscribe","_msgid":"ea2d71aa.1e2bc"}`

biancode, 2018-03-14:
@gar-syn 
a) I'll check the coming through of addressSpaceItems properties.

b) Yes, the listener has only one output since v2.2.x. 

c) The interval in payload is and was the sample interval to monitor option - payload.options is new since v2.2.x

Thank you for your report! 
Klaus

gar-syn, 2018-03-19:
@biancode 

Just to let you know it's still the same in 2.3.2, the addressSpaceItems name & datatypeName are not coming through. I think that's more down to the below :-

`addressSpaceItems: [{name: '', nodeId: monitoredItem.itemToMonitor.nodeId.toString(), datatypeName: ''}]`

biancode, 2018-03-19:
@gar-syn could you post your exmple flow, please?

gar-syn, 2018-03-28:
@biancode Not quite sure I understand your response. Are you saying that more work is required in implementing that functional within the node? Or am I doing something wrong?

biancode, 2018-03-29:
@gar-syn more work for implementation

closed 2018-04-07 by biancode