BiancoRoyal/node-red-contrib-iiot-opcua #30: Error: Session not ready To Write, Browse, Read, ...

giminni, 2018-03-12:
Please note:
 I was unable to inject MANUALLY after startup a specific data point
using write, browse, read command.

I have installed...
node-red: 0.18.4
node: 6.10.3
npm: 3.10.10
node-red-contrib-iiot-opcua: 2.2.1 
SecurityPolicy: NONE
SecurityMode: None
User: ANONYMOUS

My logs shows always:

> 2018-03-12T13:16:21.812Z opcuaIIoT:connector:details Options {"securityPolicy":"None","securityMode":"NONE","defaultSecureTokenLifetime":60000,"certificateFile":null,"privateKeyFile":null,"endpoint_must_exist":false}
2018-03-12T13:16:21.857Z opcuaIIoT:connector connected to opc.tcp://10.0.1.123:4840/
2018-03-12T13:16:21.859Z opcuaIIoT:connector Secured Connected On opc.tcp://10.0.1.123:4840/
2018-03-12T13:16:21.860Z opcuaIIoT:connector:details Options {"securityPolicy":"None","securityMode":"NONE","defaultSecureTokenLifetime":60000,"certificateFile":"/data/red/node_modules/node-opcua-client/certificates/client_selfsigned_cert_1024.pem","privateKeyFile":"/data/red/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem","endpoint_must_exist":false}
2018-03-12T13:16:21.861Z opcuaIIoT:connector Request For New Session From
12 Mar 13:47:32 - [error] [OPCUA-IIoT-Write:e090f486.f46f08] Error: Session Not Ready To Write
    at .<anonymous> (/data/red/node_modules/node-red-contrib-iiot-opcua/opcuaIIoT/maps/opcua-iiot-write.js:126:20)
    at emitOne (events.js:96:13)
    at emit (events.js:188:7)
    at Node.receive (/data/red/node_modules/node-red/red/runtime/nodes/Node.js:215:14)
    at e.Node.send (/data/red/node_modules/node-red/red/runtime/nodes/Node.js:137:22)
    at e.<anonymous> (/data/red/node_modules/node-red-contrib-iiot-opcua/opcuaIIoT/maps/opcua-iiot-inject.js:124:13)
    at emitOne (events.js:96:13)
    at emit (events.js:188:7)
    at e.Node.receive (/data/red/node_modules/node-red/red/runtime/nodes/Node.js:215:14)
    at /data/red/node_modules/node-red-contrib-iiot-opcua/opcuaIIoT/maps/opcua-iiot-inject.js:176:8
    at Layer.handle [as handle_request] (/data/red/node_modules/express/lib/router/layer.js:95:5)
    at next (/data/red/node_modules/express/lib/router/route.js:137:13)
    at /data/red/node_modules/node-red/red/api/auth/index.js:57:28
    at complete (/data/red/node_modules/passport/lib/middleware/authenticate.js:263:13)
    at /data/red/node_modules/passport/lib/middleware/authenticate.js:270:15
    at pass (/data/red/node_modules/passport/lib/authenticator.js:431:14)

araisch, 2018-03-14:
If I use several flows for development in Node-Red and just deploy "Modify Nodes" to not disturbe my productive flow I get the same error when using OPC-UA nodes.
If I deploy "Full" it's working.

`0|node-red | 14 Mar 10:02:21 - [error] [OPCUA-IIoT-Read:ns=0;i=2295] Error: Session Not Ready To Read
`
also the node doesn't switch to "active", keeps on "waiting...". 

Same issue?

giminni, 2018-03-14:
**The problem is not seeing in the debug log "Working Session On Start Request".** 

The opcuaIIoT:connector code looks like this

> node.startSession = function (timeoutSeconds) {
      coreConnector.internalDebugLog('Request For New Session From')
      if (node.opcuaSession) {
        coreConnector.internalDebugLog('Working Session On Start Request')
        return
}

**I assume there is no working session established....**

The opcuaIIoT:browse  code looks like this

> if (!node.opcuaSession) {
             node.error(new Error('Session Not Ready To Browse'), msg)
        return
}

See the debug log below

> 2018-03-14T14:56:17.219Z opcuaIIoT:connector Request For New Session From
14 Mar 14:56:18 - [error] [OPCUA-IIoT-Browser:127] Error: Session Not Ready To Browse
    at e.<anonymous> (/data/red/node_modules/node-red-contrib-iiot-opcua/opcuaIIoT/maps/opcua-iiot-browser.js:198:12)
    at emitOne (events.js:96:13)
    at emit (events.js:188:7)
    at e.Node.receive (/data/red/node_modules/node-red/red/runtime/nodes/Node.js:215:14)
    at InjectNode.Node.send (/data/red/node_modules/node-red/red/runtime/nodes/Node.js:137:22)
    at InjectNode.<anonymous> (/data/red/node_modules/node-red/nodes/core/core/20-inject.js:77:22)
    at emitOne (events.js:96:13)
    at InjectNode.emit (events.js:188:7)
    at InjectNode.Node.receive (/data/red/node_modules/node-red/red/runtime/nodes/Node.js:215:14)
    at /data/red/node_modules/node-red/nodes/core/core/20-inject.js:105:22
    at Layer.handle [as handle_request] (/data/red/node_modules/express/lib/router/layer.js:95:5)
    at next (/data/red/node_modules/express/lib/router/route.js:137:13)
    at /data/red/node_modules/node-red/red/api/auth/index.js:57:28
    at complete (/data/red/node_modules/passport/lib/middleware/authenticate.js:263:13)
    at /data/red/node_modules/passport/lib/middleware/authenticate.js:270:15
    at pass (/data/red/node_modules/passport/lib/authenticator.js:431:14)

araisch, 2018-03-15:
Ok sorry, then it's something else.
@biancode: Blame on me. Maybe you can split up in a new issue?

Goes back to work, if:
1. change the OPC-Endpoint to something else
2. deploy modified
3. change OPC-Endpoint back 
4. deploy modified again

biancode, 2018-03-15:
v2.3.0

closed 2018-03-15 by biancode

giminni, 2018-03-16:
Sorry the last version 2.3.0 had not solved this problem.

NOW I can run a wireshark session sniffing the node-red instance mirroring a specific port that is working with a real OPC/UA server.

This is my test environment:
[image]

After connection (see state above) I get with wireshark:
[image]

[image]

After trigger the injector (see above) i get with wireshark:
NOTHING!!!!!!

Apparently the session is not there. Can you assure it is working on a real OPC/UA server otherwise I must send this debug to the vendor