BiancoRoyal/node-red-contrib-iiot-opcua #31: crashes when connecting to existing non-node.js server

sixdiamants, 2018-03-12:
I'm running a .NET standard server from the OPC-UA foundation. UAExpert can connect to the endpoint. However, when I connect a OPCUA-IIoT-Browser  to this server, node-red crashes;

> 12 Mar 14:27:07 - [red] Uncaught Exception:
> 12 Mar 14:27:07 - Error: EACCES: permission denied, open '/usr/local/lib/node_modules/node-red-contrib-iiot-opcua/node_modules/node-opcua-client/certificates/PKI/server_certificate0.pem'

 And indeed, the server certificate doesn't exist on the given location. 

[nodeREDIIoTOPCUA.log](https://github.com/biancode/node-red-contrib-iiot-opcua/files/1803046/nodeREDIIoTOPCUA.log)

biancode, 2018-03-16:
please try the [opcua-commander](https://github.com/node-opcua/opcua-commander) does it work?

biancode, 2018-03-16:
and try v2.3.1