BiancoRoyal/node-red-contrib-iiot-opcua #32: Listener with NOT "Read just Values"

biancode, 2018-03-13:
@araisch

In the flow above are the Listener and the Read Node **not** set to "Read just Values".
It gives me this:

```
Exception raised inside the event handler called by ClientMonitoredItem.on('change') TypeError: Converting circular structure to JSON
    at JSON.stringify (<anonymous>)
    at Object.c.sendDataFromMonitoredItem (/home/gateway/.node-red/node_modules/node-red-contrib-iiot-opcua/opcuaIIoT/maps/opcua-iiot-listener.js:167:14)
    at ClientMonitoredItem.<anonymous> (/home/gateway/.node-red/node_modules/node-red-contrib-iiot-opcua/opcuaIIoT/maps/opcua-iiot-listener.js:139:10)
    at ClientMonitoredItem.emit (events.js:127:13)
    at ClientMonitoredItemBase._notify_value_change (/home/gateway/.node-red/node_modules/node-opcua-client/src/client_monitored_item_base.js:45:13)
    at /home/gateway/.node-red/node_modules/node-opcua-client/src/client_subscription.js:192:32
    at Array.forEach (<anonymous>)
    at ClientSubscription.__on_publish_response_DataChangeNotification (/home/gateway/.node-red/node_modules/node-opcua-client/src/client_subscription.js:184:20)
    at /home/gateway/.node-red/node_modules/node-opcua-client/src/client_subscription.js:297:26
    at Array.forEach (<anonymous>)
Please verify the application using this node-opcua client
```

biancode, 2018-03-13:
[image]

hm.. works on my side ...

a) What is the dataType of that item?
b) Did you get a response sent from the listener node output at this case?
c) event or monitoring?

biancode, 2018-03-13:
@araisch some ByteStrings did not work and will result in JSON string output

araisch, 2018-03-13:
a) 5 Variables, Boolean, Int16 and Float. Nothing special, no ByteString.
b) No output message in this case. 
c) Monitoring

Browser with "Results to Listener" only.
If Listener is "Send just Values" it works normal, with 5 Messages on Subscription and a Message on Change.

araisch, 2018-03-13:
For injection I use "listen" and JSON { "interval": 5000, "queueSize": 10 }
Tried queueSize: 10000 with no result as well, due to I don't really know what it's good for :)

biancode, 2018-03-13:
queueSize is set to subscription monitor to node-opcua

* @method monitor
 * @async
 * @param itemToMonitor                        {ReadValueId}
 * @param itemToMonitor.nodeId                 {NodeId}
 * @param itemToMonitor.attributeId            {AttributeId}
 * @param itemToMonitor.indexRange             {null|NumericRange}
 * @param itemToMonitor.dataEncoding
 * @param requestedParameters                  {MonitoringParameters}
 * @param requestedParameters.clientHandle     {IntegerId}
 * @param requestedParameters.samplingInterval {Duration}
 * @param requestedParameters.filter           {ExtensionObject|null} EventFilter/DataChangeFilter
 * @param requestedParameters.queueSize        {Counter}
 * @param requestedParameters.discardOldest    {Boolean}
 * @param timestampsToReturn                   {TimestampsToReturn} //{TimestampsToReturnId}
 * @param  [done]                              {Function} optional done callback
 * @return {ClientMonitoredItem}

biancode, 2018-03-15:
v2.3.0

closed 2018-03-15 by biancode

araisch, 2018-03-19:
Works more or less. Both output-messages are now equal. So "Send just Values" has no influence on the output now.. for me ok 👍