BiancoRoyal/node-red-contrib-iiot-opcua #33: Problems with OPCUA Discovery

fedelodo, 2018-03-13:
Hi,
I'm running the latest version of the package, I've checked the discovery square in the Security tab on the Server node but when I connect with UA Expert  on the endpoint it says "Discovery GetEndpoint on <address> (BadCommunicationError)" and  "Discovery FindServers on <address> (BadCommunicationError)" . 
Is it a known issue?
Thanks in advance.

biancode, 2018-03-13:
Hi,

please post your server flow!

on my side it works ...

[image]

biancode, 2018-03-13:
... or you mean port 4840, right?

biancode, 2018-03-13:
[image]

biancode, 2018-03-13:
comes with the next version as a new node - named discovery

[image]

[image]

biancode, 2018-03-13:
[image]

biancode, 2018-03-13:
[image]

biancode, 2018-03-15:
v2.3.0

closed 2018-03-15 by biancode