BiancoRoyal/node-red-contrib-iiot-opcua #35: [Question] How to create a more complex model with nodes?

w4tsn, 2018-03-15:
I played around with the examples provides by this module but I'm unable to get a more complex model working. Currently I have to either manually inject with ASO nodes everything in the right order, or I have to play with some delay. Is there some neat way to create the model with nodes visually that I don't see?

The alternative would be to code the model in javascript with the Flex Server node.

biancode, 2018-03-15:
Yes, the delay is the key. Objects needs folder and property need object or folder to exist and the root or parent must exist on adding.

biancode, 2018-03-15:
idea: ASO with tree model

closed 2018-03-15 by biancode

w4tsn, 2018-03-16:
Would you like to elaborate a bit more on that tree model idea?

Currently I prototype the model with several function nodes in order to create folders, objects and variables that package those related together. With that I get a reusable node for each component, which can be inserted on server start-up without thinking about time constraints and insertion order. Within the function nodes I create messages for all objects as of the ASO-Nodes output format. When done I send all the messages with `node.send(asoMsg)` in the correct order.

I'm thinking of implementing some 'FlexASONode', where I'm able to implement related groups of ASO which are independent from each other. Like e.g. for different hardware components in a server spanning multiple machines.