BiancoRoyal/node-red-contrib-iiot-opcua #36: v2.3.0 Listener doesn't stopp / Read goes to error

araisch, 2018-03-16:
On "deploy" a Listener in a special state of connection (Waiting/active) doesn't stop:
`[OPC-IIOT-Listener] Error stopping node: TimeoutError: timed out after 15000 ms`
The (Listening) Listeners are terminated correctly
[image]
But somehow strange:
If i just "deploy modified" the flow with the waiting listener, it stops correctly.

The Read-Node goes to error if an OPC-UA-Server is rebooted. No message is given, even no error message. It doesn't try to activate the connection again, even on injection. Just deploy new solves the problem. While Browser and Listener **do** activate again.

biancode, 2018-03-16:
v2.3.1

araisch, 2018-03-19:
Didn't encounter this again. Because now on "Deploy modified" the nodes just stick on "Waiting..."
[image]
Both running. Changed "Send just values" at Listener and Read to make them deploy new. Hit "deploy modified":
[image]
Not coming back. Hit "Deploy full" -> Back to active.
Changing just the Read-Node and deploy modified:
[image]

biancode, 2018-03-19:
I have to check the state of the config node to handle that case. The state is just a wrong information and should work well on waiting, right?

biancode, 2018-03-19:
Just a full deploy resets the client (connector) otherwise the monitoring is the same after 'modified deploy', but that is the concept of Node-RED in that case.

araisch, 2018-03-19:
you mean the configuration of the config node? Don't know how to show the state..
Ok. So if I change something having a connection I have to "deploy full"?! Good to know, thanks.
Anyway, this:
> The Read-Node goes to error if an OPC-UA-Server is rebooted. No message is given, even no error message. It doesn't try to activate the connection again, even on injection. Just deploy new solves the problem. 

is solved partially. The Read-Node shows now "active" after OPC-UA-Server reboots, but it doesn't give a message on inject. Just "deploy full" makes the Node going back to work. Is that a similar problem?

biancode, 2018-03-19:
improved with v2.3.3 - you get addressSpaceItems to reinject and monitoredIterms from session to read subscription information

araisch, 2018-03-20:
Nice Improvement! Works perfectly so far on reboot 🥇 
Just two little thoughts from my side:
- Listener should spread the monitoredItems Array just with NOT "Send just values" to keep the lightweight message we taked about earlier.
- Array of monitoredItems counts up. So if you subscribe, desubscribe and subscribe again (I do this every hour to refresh for new variables in the browsed folder and write values at least once an hour) the array numbers don't start with 0. 
Example: Subscribe 41 Variables, Desubscribe and Subscribe again. 
Result:
[image]
Makes harder to work with the array in functions..

araisch, 2018-03-22:
Posted to early. With short reboot it's working. On longer absence of the Server (over night) it doesn't go back to work :(

biancode, 2018-03-22:
@araisch please, send me some logs and more information what happens over night!

araisch, 2018-03-22:
Unfortunatelly Node-Red doesn't print timestamps in all logs even with --timestamp
Log:
```
0|node-red |  Timeout .... waiting for response for  PublishRequest { /*RequestHeader*/
0|node-red |  authenticationToken           /* NodeId                           */: ns=0;i=2
0|node-red |  timeStamp                     /* UtcTime                          */: 2018-03-21T16:04:01.468Z
0|node-red |  requestHandle                 /* IntegerId                        */: 64085               0xfa55
0|node-red |  returnDiagnostics             /* UInt32                           */: 0               0x0
0|node-red |  auditEntryId                  /* UAString                         */:
0|node-red |  timeoutHint                   /* UInt32                           */: 100000               0x186a0
0|node-red |  additionalHeader              /* ExtensionObject                  */: null
0|node-red | };
0|node-red | !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
0|node-red | !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
0|node-red | !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
0|node-red | !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
0|node-red | !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
0|node-red | !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
0|node-red | !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
```
and more lines like that
`0|node-red | !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!`

The entry 16:04 was the end of a small downtime of roughly 30 min. There are values until the Server was switched of 16:38, so no further entries beside a lot of !!!
Error log:
```
0|node-red | (node:16372) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'sessionId' of null
0|node-red |     at o.nodes.registerType.credentials.user.type.t.resetBadSession (/home/gateway/.node-red/node_modules/node-red-contrib-iiot-opcua/opcuaIIoT/maps/opcua-iiot-connector.js:230:83)
0|node-red |     at Object.l.handleReadError (/home/gateway/.node-red/node_modules/node-red-contrib-iiot-opcua/opcuaIIoT/maps/opcua-iiot-read.js:56:22)
0|node-red |     at /home/gateway/.node-red/node_modules/node-red-contrib-iiot-opcua/opcuaIIoT/maps/opcua-iiot-read.js:72:14
0|node-red |     at <anonymous>
0|node-red |     at process._tickCallback (internal/process/next_tick.js:118:7)
0|node-red | (node:16372) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 548)
```
Whole Log is spammed with this. Will have a look this evening what happens in the beginning :)

biancode, 2018-03-27:
fixed with v2.4.0

biancode, 2018-03-27:
next fix v2.4.1 - I can not reproduce it, but there was an access on sessionId, which could be null

araisch, 2018-03-29:
Strange thing. When I restart the Server the Log shows this now:
```
<< More repetitions of the following block between Device switched on and this line are snipped >>
2018-03-29-09:15:58 0|node-red |  Timeout .... waiting for response for  PublishRequest { /*RequestHeader*/
2018-03-29-09:15:58 0|node-red |  authenticationToken           /* NodeId                           */: ns=0;i=5
2018-03-29-09:15:58 0|node-red |  timeStamp                     /* UtcTime                          */: 2018-03-29T07:14:18.578Z
2018-03-29-09:15:58 0|node-red |  requestHandle                 /* IntegerId                        */: 1011               0x3f3
2018-03-29-09:15:58 0|node-red |  returnDiagnostics             /* UInt32                           */: 0               0x0
2018-03-29-09:15:58 0|node-red |  auditEntryId                  /* UAString                         */: 
2018-03-29-09:15:58 0|node-red |  timeoutHint                   /* UInt32                           */: 100000               0x186a0
2018-03-29-09:15:58 0|node-red |  additionalHeader              /* ExtensionObject                  */: null
2018-03-29-09:15:58 0|node-red | };
2018-03-29-09:16:10 0|node-red |  Timeout .... waiting for response for  ReadRequest { /*RequestHeader*/
2018-03-29-09:16:10 0|node-red |  authenticationToken           /* NodeId                           */: ns=0;i=5
2018-03-29-09:16:10 0|node-red |  timeStamp                     /* UtcTime                          */: 2018-03-29T07:15:10.646Z
2018-03-29-09:16:10 0|node-red |  requestHandle                 /* IntegerId                        */: 1012               0x3f4
2018-03-29-09:16:10 0|node-red |  returnDiagnostics             /* UInt32                           */: 0               0x0
2018-03-29-09:16:10 0|node-red |  auditEntryId                  /* UAString                         */: 
2018-03-29-09:16:10 0|node-red |  timeoutHint                   /* UInt32                           */: 0               0x0
2018-03-29-09:16:10 0|node-red |  additionalHeader              /* ExtensionObject                  */: null
2018-03-29-09:16:10 0|node-red | };
2018-03-29-09:16:23 0|node-red |  warning : ClientSessionKeepAliveManager#ping_server  Transaction has timed out ( timeout = 60000 ms)
2018-03-29-09:16:37 0|node-red |  Timeout .... waiting for response for  ReadRequest { /*RequestHeader*/
2018-03-29-09:16:37 0|node-red |  authenticationToken           /* NodeId                           */: ns=0;i=5
2018-03-29-09:16:37 0|node-red |  timeStamp                     /* UtcTime                          */: 2018-03-29T07:15:37.317Z
2018-03-29-09:16:37 0|node-red |  requestHandle                 /* IntegerId                        */: 1014               0x3f6
2018-03-29-09:16:37 0|node-red |  returnDiagnostics             /* UInt32                           */: 0               0x0
2018-03-29-09:16:37 0|node-red |  auditEntryId                  /* UAString                         */: 
2018-03-29-09:16:37 0|node-red |  timeoutHint                   /* UInt32                           */: 0               0x0
2018-03-29-09:16:37 0|node-red |  additionalHeader              /* ExtensionObject                  */: null
2018-03-29-09:16:37 0|node-red | };
2018-03-29-09:16:37 0|node-red |  warning : ClientSessionKeepAliveManager#ping_server  Transaction has timed out ( timeout = 60000 ms)
2018-03-29-09:16:50 0|node-red |  Timeout .... waiting for response for  ReadRequest { /*RequestHeader*/
2018-03-29-09:16:50 0|node-red |  authenticationToken           /* NodeId                           */: ns=0;i=5
2018-03-29-09:16:50 0|node-red |  timeStamp                     /* UtcTime                          */: 2018-03-29T07:15:50.651Z
2018-03-29-09:16:50 0|node-red |  requestHandle                 /* IntegerId                        */: 1015               0x3f7
2018-03-29-09:16:50 0|node-red |  returnDiagnostics             /* UInt32                           */: 0               0x0
2018-03-29-09:16:50 0|node-red |  auditEntryId                  /* UAString                         */: 
2018-03-29-09:16:50 0|node-red |  timeoutHint                   /* UInt32                           */: 0               0x0
2018-03-29-09:16:50 0|node-red |  additionalHeader              /* ExtensionObject                  */: null
2018-03-29-09:16:50 0|node-red | };
2018-03-29-09:16:50 0|node-red |  warning : ClientSessionKeepAliveManager#ping_server  Transaction has timed out ( timeout = 60000 ms)
2018-03-29-09:17:35 0|node-red | !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! 
2018-03-29-09:17:55 0|node-red | !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! 
2018-03-29-09:18:15 0|node-red | !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! 
2018-03-29-09:18:35 0|node-red | !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! 
```
"Keep Session alive" is switched off.
As far as I understand:
The spamming is caused by the 2 Listeners they are still in status "Listening (xxx)", they are quiet until the device is switched on. Then they spam a little bit, change then to 1 min cycles. The Listeners change from "Listening (xxx)" to "Active". That would be ok for me, so i could inject as soon as the Server is ready. For this reason I try to read a single variable from the server.
The !!!! are caused by this read request I start as soon as the device is answering to ping to obtain the server status. No Error-Log-Entries anymore since you did the update. But it's not working either.

If I deploy new everything is working..
Logging level is "trace", are there further methods to check what happens?

biancode, 2018-03-29:
#44

closed 2018-03-29 by biancode