BiancoRoyal/node-red-contrib-iiot-opcua #37: Problem writing to a variable defined in FlexServer

w4tsn, 2018-03-16:
I've setup a flex-server with a variable of type boolean under the root node. I'm trying to add a certain numerical value of 38.5 into this variable through a write node. 

**The writing fails with the response:**

```
{
    "value": 2151350272,
    "description": "The access level does not allow writing to the Node.",
    "name": "BadNotWritable"
}
```

**AccessLevel is** `CurrentRead, CurrentWrite`. There is no security or user auth enabled. Should everything go with anonymous access.

**The flex-code is as follows:**

```
let test = addressSpace.addVariable({
    browseName: 'Test',
    displayName: 'Test',
    description: 'A Test',
    propertyOf: addressSpace.rootFolder,
    dataType: 'Double',
    eventSourceOf: addressSpace.rootFolder
})
```

**The connector-parameters are:**

[image]

biancode, 2018-03-16:
Hi Alexander,

it needs to add a [value get and set function](https://github.com/node-opcua/node-opcua-sampleserver/blob/master/server.js)

biancode, 2018-03-16:
reopen if you need more ;)

closed 2018-03-16 by biancode

tmk99, 2019-05-20:
Would it be possible to upload an example of the get and set functions? I have the same problem but I am unable to program a get / set function.