BiancoRoyal/node-red-contrib-iiot-opcua #38: Node-RED crash on deploy of FlexServer with faulty configuration

w4tsn, 2018-03-16:
## The problem

When there is a faulty configuration script of the FlexServer like the following, the deployment kills the whole node-red environment.

```
let test = addressSpace.addVariable({
    browseName: 'Test',
    displayName: 'Test',
    description: 'Some test variable',
    propertyOf: addressSpace.rootFolder.objects, // faulty, cause variables can't be property to this
    dataType: 'Double',
    eventSourceOf: addressSpace.rootFolder
})
```

Another example is:

```
// there is no addProperty function within the API
let test = addressSpace.addProperty({
    browseName: 'Test',
    displayName: 'Test',
    description: 'Some test variable',
    propertyOf: addressSpace.rootFolder,
    dataType: 'Double',
    eventSourceOf: addressSpace.rootFolder
})
```

## Versions

Node-RED: 0.17.5
Node-OPCUA: 8.9.4
node-red-contrib-iiot-opcua: 2.2.1
node-opcua: 0.2.2

biancode, 2018-03-16:
Yes, I tried to catch it, but for now it does not work - Node-RED ever crashing.

biancode, 2018-03-17:
v2.3.2

closed 2018-03-17 by biancode