BiancoRoyal/node-red-contrib-iiot-opcua #39: ERROR: Session Not Ready To Browse

giminni, 2018-03-16:
Sorry the last version 2.3.0 had not solved this problem.

NOW I can run a wireshark session sniffing the node-red instance mirroring a specific port that is working with a real OPC/UA server.

This is my test (injector -> browse node -> debug node) environment:
[image]

After connection (see state above) I get with wireshark:
[image]

[image]

After trigger the injector (see above) i get with wireshark:
NOTHING!!!!!!

Apparently the session is not there. 

Can you assure it is working on a real OPC/UA server otherwise I must send this debug to the vendor

biancode, 2018-03-16:
hm interesting ... if Prosys Simulator is a real OPC UA server for you, then yes it works well with a real OPC UA server **IF** the certificate is set to trusted otherwise I got the same trouble as you described:

Certificate in server not trusted:
[image]

Certificat in server trusted:
[image]

I'll try more with the Unified Automation servers too.

biancode, 2018-03-16:
Your problem is to reproduce with the Prosys server if ANONYMOUS is not allowed or user is wrong.
Is the Show Errors option checked in your flow in the Connector node settings?

Error: Cannot find ANONYMOUS user token policy in end point description

[image]

biancode, 2018-03-16:
[image]

biancode, 2018-03-16:
[image]

biancode, 2018-03-16:
[image]

biancode, 2018-03-16:
works with user too

[image]

biancode, 2018-03-16:
ANONYMOUS works for me too if it is activated in server

[image]

biancode, 2018-03-16:
[image]

biancode, 2018-03-16:
[image]

also in just ANONYMOUS mode

biancode, 2018-03-16:
please try v2.3.1

closed 2018-03-16 by biancode