BiancoRoyal/node-red-contrib-iiot-opcua #40: [Question] How to configure Client to Write?

JoaoMCPereira, 2018-03-16:
I'm currently running a FLEX Server in a Raspberry Pi, and I want to Write to the Server through IOT2020 also with Node-RED.

This is my current configuration on Client side:

[image]

And in Connector node I got this configuration: 

[image]

But I'm getting the following error "Error: Session Not Valid On Write", I'm sure this is the correct Endpoint because this is the one I'm using in UaExpert to check everything.
What I'm doing wrong?

biancode, 2018-03-16:
Please, set the Show Errors to checked and watch debug tab. What is the message there?

JoaoMCPereira, 2018-03-16:
This is the error I get

[image]

There is need any specific block, in Node-RED, to make the Client work? Like Server block or any other else?

biancode, 2018-03-16:
Please, test the examples under Node-RED menu > Import > examples > iiot opcua. Does it work?

biancode, 2018-03-16:
please test v2.3.1 and/or send your flows of both to me

JoaoMCPereira, 2018-03-17:
Can you tell me what is the best way to test de v2.3.1? 
Run the 'npm install -g node-red-contrib-iiot-opcua' will upgrade it?

biancode, 2018-03-17:
for me `npm update -g node-red-contrib-iiot-opcua` works too if the node-opcua did not change in minor, but if you got too much trouble or node-opcua changed in minor a clean uninstall and install is the best practice

JoaoMCPereira, 2018-03-17:
@biancode I dont have any node-opcua module instaled in the Server or neither in the Client, I don't know that was needed. I should install this module first? Or just clean installation of node-red-contrib-iiot-opcua is needed? 
Sorry for my silly question, but I did not understand the relation between this to modules. I was thinkng that iiot-opcua was enough to run a fully OPC-UA communication.

JoaoMCPereira, 2018-03-17:
@biancode I reinstalled node-red-contrib-iiot-opcua module with the v2.3.1 and now everything works like a charm, Thank you for the help.

closed 2018-03-17 by biancode