BiancoRoyal/node-red-contrib-iiot-opcua #41: ASONodes  or FlexASONodes for FlexServer

w4tsn, 2018-03-18:
I'd like to create a server with the OPCUA-SDK like in the FlexServer ASO definition tab, but modular like with the ASO-Nodes.

The use-case is an OPC-UA server for a system composed of multiple systems where the sub-systems are well-defined and can be expressed nicely with the SDK-script. The overall systems differ in the sub-systems that are installed. The sub-systems are independent of each other and do not require a specific injection order. Common used type-definitions will be shipped with the FlexServer node.

For this I require a way to have the SDK-script like in the FlexServer distributed in several inject nodes.

If you are able to point out some basics around this topic like if it is even possible to inject such SDK-script definition into a running server and how to do that I'll most likely implement this feature myself.

biancode, 2018-03-19:
please accept my request on Xing ;)

w4tsn, 2018-03-19:
I did :+1:

w4tsn, 2018-03-19:
Part of the use-case is the following issue:

I've implemented a normal server and I'm simulating the ASO-node with a function node, where I package dependent objects, variables etc. together. I then send the messages in the correct order per node so they don't interfere with each other.

The problem however is, since the server is booted without a model and the model is injected just after a short time after the server is ready it happens that a client with subscriptions to objects and variables connects to the rebooted server before the definition is re-injected. It then is unable to find the objects and automatically deletes it's subscription.

I've no idea how to avoid that without implementing the sole FlexServer approach. Even with a FlexASONode it could still be a tricky issue.

biancode, 2018-10-14:
Hi, you could contribute for such a node. Feel free to send you pull requests!

closed 2018-10-14 by biancode