BiancoRoyal/node-red-contrib-iiot-opcua #43: v2.3.3: Cannot read property 'uniformResourceIdentifier' of undefined

giminni, 2018-03-20:
latest lib doesn't works with securePolicy:Basic256 and secureMode:SIGNANDENCRYPT
Last state is connected.

Is this a server problem with ANONYMOUS?

Here what I made so far:
`#!/bin/sh
CERTS=$HOME/certs
KEYS=$HOME/keys

openssl req -x509 -newkey rsa:1024 -keyout $KEYS/opcua-key.pem -out $CERTS/opcua-cert.pem -days 3650 -nodes \
   -subj "/C=T/ST=U/L=V/O=W/OU=X/CN=Y/emailAddress=Z"
openssl x509 -outform der -in $CERTS/opcua-cert.pem -out $CERTS/opcua-cert.der`

Here my debug log:

`2018-03-20T06:42:33.648Z opcuaIIoT:connector Request For New Session From
2018-03-20T06:42:33.649Z opcuaIIoT:connector Create New Session Without User Identity
2018-03-20T06:42:33.852Z opcuaIIoT:connector TypeError: Cannot read property 'uniformResourceIdentifier' of undefined
    at OPCUAClient._getApplicationUri (/data/red/node_modules/node-opcua-client/src/opcua_client.js:112:68)
    at OPCUAClient.__createSession_step2 (/data/red/node_modules/node-opcua-client/src/opcua_client.js:203:31)
    at OPCUAClient._createSession (/data/red/node_modules/node-opcua-client/src/opcua_client.js:170:10)
    at OPCUAClient.createSession (/data/red/node_modules/node-opcua-client/src/opcua_client.js:623:10)
    at OPCUAClient.eval [as createSession] (eval at <anonymous> (/data/red/node_modules/thenify/index.js:33:15), <anonymous>:6:48)
    at /data/red/node_modules/node-red-contrib-iiot-opcua/opcuaIIoT/maps/core/core/opcua-iiot-core-connector.js:110:13
    at Object.de.biancoroyal.opcua.iiot.core.connector.createSession (/data/red/node_modules/node-red-contrib-iiot-opcua/opcuaIIoT/maps/core/core/opcua-iiot-core-connector.js:97:11)
    at t.startSession (/data/red/node_modules/node-red-contrib-iiot-opcua/opcuaIIoT/maps/opcua-iiot-connector.js:174:21)
    at /data/red/node_modules/node-red-contrib-iiot-opcua/opcuaIIoT/maps/opcua-iiot-connector.js:138:20
    at process._tickCallback (internal/process/next_tick.js:109:7)`

biancode, 2018-03-20:
That is out of me free support scope.

closed 2018-03-20 by biancode

giminni, 2018-03-20:
Thx for pointing me into the right direction I opened a bug to the OPC-UA server vendor.