BiancoRoyal/node-red-contrib-iiot-opcua #45: BadSessionIdInvalid(0x80250000)

davids1802, 2018-03-26:
I'm currently have a problem with NODE-RED, my flow works properly but, after a while appears that error on write OPCUA IIoT node:

[image]

I have read some post but i can't resolve this issue, I think to know what I have to do: I need to restart the session Id when Write Node status code be this one. But I don't know where I can develop this code, maybe in index.js (node-modules folder)?

Any help will be appreciated

David,

biancode, 2018-03-26:
Hi, I need some version information about nodejs, Node-RED and the package please!

davids1802, 2018-03-26:
Hi, thanks for your quickly answer, here is it:

[image]
 
[image]

If you need something more, let me know it, thank you so much!

biancode, 2018-03-26:
OK, then I need some logs, please. The 'BadSession' should kill the old session and should create a new session. How fast do you write (msec, sec)?

biancode, 2018-03-26:
Do you wait a bit before you start to write at the beginning?

davids1802, 2018-03-26:
Now is working fine like you can see, the Node-Red part that fail is this:

[image]

Every 30 sec I write a "heartbeat" (is the flow that comes from down)
The other is like 30 sec too... maybe the problem is that at some point they're crossed in the Write Node Action because they come from two different flows?
Anyway, if they  match and session goes down, why in the next flow, doesn't raise the session? 
Can I detect when occurs the BadSession and create a new one?

I copy my flows here:

> [{"id":"73c9977f.870128","type":"http request","z":"85a15ec8.2bc6e","name":"Weigh request","method":"GET","ret":"obj","url":"http://HC15:8087/api/integration/weighing/search?form.status=Pending&form.code=T1%25","tls":"","x":320,"y":140,"wires":[["4cfcd5fe.587fb8","bdb58cbe.1f3c3"]]},{"id":"d12890f8.aeafd","type":"inject","z":"85a15ec8.2bc6e","name":"timestamp","topic":"","payload":"","payloadType":"date","repeat":"40","crontab":"","once":false,"onceDelay":"","x":122.00000953674316,"y":141.0000123977661,"wires":[["73c9977f.870128"]]},{"id":"69ba7a21.2e6604","type":"function","z":"85a15ec8.2bc6e","name":"EXEC STORED","func":"\npld =       \"EXEC AETech_Flow1_InsertarRecepcion \";\npld = pld + \"'\"+ msg.payload +\"'\";\n\nmsg.payload = pld;\nreturn msg;","outputs":"1","noerr":0,"x":1362.9998397827148,"y":89.99993991851806,"wires":[["76799c99.3794d4","87ccb413.cb8858"]],"inputLabels":["JSON"],"outputLabels":["JSON filtrat Pes > 15000"]},{"id":"87ccb413.cb8858","type":"debug","z":"85a15ec8.2bc6e","name":"","active":true,"console":"false","complete":"true","x":1640.0001831054687,"y":124.0000410079956,"wires":[]},{"id":"bdb58cbe.1f3c3","type":"function","z":"85a15ec8.2bc6e","name":"Iteració per filtrar dins el JSON","func":"var array = [];\nfor (var i=0; i < msg.payload.content.length;i++) {\n    for (var j=0; j < msg.payload.content[i].Lines.length;j++) {\n        console.log('i:'+i);\n        if ((msg.payload.content[i].Lines[j].FirstWeighing > 22000) && \n            (msg.payload.content[i].Manual === false)) {\n            array.push({payload: msg.payload.content[i]});\n            //return msg.payload.content[i];\n        }\n    }\n}\n//console.log(array);\nmsg.payload = [ array ];\n//return [ array ];\nreturn msg.payload;","outputs":"1","noerr":0,"x":930,"y":120,"wires":[["27cd5421.e9410c","53616dd0.d9d274"]],"inputLabels":["JSON"],"outputLabels":["JSON filtrat Pes > 15000"]},{"id":"27cd5421.e9410c","type":"debug","z":"85a15ec8.2bc6e","name":"","active":false,"console":"false","complete":"payload","x":1209.000156402588,"y":182.000018119812,"wires":[]},{"id":"53616dd0.d9d274","type":"json","z":"85a15ec8.2bc6e","name":"","pretty":false,"x":1188.0000457763672,"y":139.00000858306885,"wires":[["94cc316f.61d08","69ba7a21.2e6604"]]},{"id":"94cc316f.61d08","type":"debug","z":"85a15ec8.2bc6e","name":"","active":false,"console":"false","complete":"true","x":1333.0001640319824,"y":140.0000352859497,"wires":[]},{"id":"b7c47703.93e418","type":"comment","z":"85a15ec8.2bc6e","name":"FLOW 1","info":"Cada 30 segons  preguntarem, filtrant per State = Pending i Code: T1%.(Significa que ha entrat a Mayorazgo)\nA dintre fem un filtre per FirstWeighing > 18000, si passen, seràn recepcions i executarem el stored.","x":115.3334846496582,"y":101.99989223480224,"wires":[]},{"id":"4b4e5c39.510e34","type":"comment","z":"85a15ec8.2bc6e","name":"FET","info":"Part del NodeRed","x":106.35300827026367,"y":66.00379848480224,"wires":[]},{"id":"b440bff7.11022","type":"debug","z":"85a15ec8.2bc6e","name":"","active":false,"console":"false","complete":"true","x":890,"y":180,"wires":[]},{"id":"f9510e58.0ddcd","type":"debug","z":"85a15ec8.2bc6e","name":"","active":false,"console":"false","complete":"true","x":3210,"y":380,"wires":[]},{"id":"cefd5dff.48f0c","type":"debug","z":"85a15ec8.2bc6e","name":"","active":true,"console":"false","complete":"true","x":3387.9999389648437,"y":329.9999942779541,"wires":[]},{"id":"e0ef203e.25f53","type":"debug","z":"85a15ec8.2bc6e","name":"","active":false,"console":"false","complete":"true","x":2952.999988555908,"y":35.9999942779541,"wires":[]},{"id":"d6335df3.7b772","type":"debug","z":"85a15ec8.2bc6e","name":"","active":false,"console":"false","complete":"true","x":2953.999987602234,"y":84.99999523162842,"wires":[]},{"id":"dc6ad1ca.f1f9e","type":"switch","z":"85a15ec8.2bc6e","name":"","property":"payload[0].Error","propertyType":"msg","rules":[{"t":"eq","v":"1","vt":"str"},{"t":"eq","v":"2","vt":"str"},{"t":"eq","v":"3","vt":"str"}],"checkall":"true","repair":false,"outputs":3,"x":2310,"y":60,"wires":[["48707146.9949e"],["10b29163.70b82f"],["bf3f7708.1dee98"]],"outputLabels":["1","2","3"]},{"id":"60cc04ca.34a5ec","type":"debug","z":"85a15ec8.2bc6e","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","x":1830,"y":120,"wires":[]},{"id":"3473ab9f.a2c5b4","type":"comment","z":"85a15ec8.2bc6e","name":"FET","info":"A node red","x":92.68644714355469,"y":257.9997606277466,"wires":[]},{"id":"2c41aa8d.508a06","type":"comment","z":"85a15ec8.2bc6e","name":"FLOW 5","info":"Cada 5 minuts incrementarem/canviarem aleatoriament el valor\nd'un element del PLC per així saber que tots els flows flueixen \ncorrectament i no hi ha cap caiguda de servidor o de WebService o de Flow.","x":103.6669692993164,"y":291.995885848999,"wires":[]},{"id":"86e5b95e.75f138","type":"debug","z":"85a15ec8.2bc6e","name":"","active":false,"console":"false","complete":"count","x":938.3531646728516,"y":296.9999313354492, [... cut ...] :"","name":"Harinalia","server":"192.168.9.1","encyption":true,"database":"Harinalia"},{"id":"f55a3722.1d7d18","type":"OPCUA-IIoT-Connector","z":"","discoveryUrl":"","endpoint":"opc.tcp://192.168.9.1:49320/","keepSessionAlive":true,"loginEnabled":false,"securityPolicy":"None","securityMode":"NONE","name":"Harinalia","showErrors":false,"publicCertificateFile":"","privateKeyFile":"","defaultSecureTokenLifetime":"60000","endpointMustExist":false,"autoSelectRightEndpoint":false}]

Thank you so much!

biancode, 2018-03-26:
please test v2.4.0

davids1802, 2018-03-27:
Good morning, 
Thank you so much, I update to v2.4.0 and now I'm waiting if I get some error, in the next days I'll send my feeback here.

Have a nice day and thanks again.

David

biancode, 2018-03-28:
reopen if not fixed

davids1802, 2018-03-28:
I'm here another time, and it isn't work : 

[image]

As you can see the she session isn't reopen. 
We take a look and the session reopen properly in the next connect when it fails, but at 00:01 (maybe the problem is the new day?) it fail again and never reconnect since I restart node red or redeploy.

Log:
[image]

By the other hand, the OPCUA IIoT Node component doesn't work properly since I updated it.
I pass screenshoot because the inject component with the same configuration is working but the node component isn't when it was working on the old version, maybe i don't have a good configuration? take a look:

Inject and Node debug:

[image]

You can see the Write-Node configuration in the old message from this post.

I paste you a part of flow with this last problem:

> [{"id":"f2e10c21.f5d3a","type":"inject","z":"85a15ec8.2bc6e","name":"timestamp","topic":"","payload":"","payloadType":"date","repeat":"","crontab":"","once":false,"onceDelay":"","x":2180,"y":720,"wires":[["2d640885.81d6d8","3e284b63.e69bbc"]]},{"id":"b19d5c24.c162a","type":"inject","z":"85a15ec8.2bc6e","name":"timestamp","topic":"","payload":"","payloadType":"date","repeat":"","crontab":"","once":false,"onceDelay":"","x":2180,"y":780,"wires":[["97b93793.528d68"]]},{"id":"5e9dd080.c6102","type":"inject","z":"85a15ec8.2bc6e","name":"timestamp","topic":"","payload":"","payloadType":"date","repeat":"","crontab":"","once":false,"onceDelay":"","x":2180,"y":840,"wires":[["acfd0ca5.cca5a"]]},{"id":"acfd0ca5.cca5a","type":"OPCUA-IIoT-Node","z":"85a15ec8.2bc6e","nodeId":"ns=2;s=Channel1.PLC01.Global.GESTRUCK_EST","datatype":"Int16","value":"3","usingListener":false,"name":"GESTRUCK_EST Error 3","x":2450,"y":840,"wires":[["836f6f11.ef4b78","389b9759.08bb08"]]},{"id":"97b93793.528d68","type":"OPCUA-IIoT-Node","z":"85a15ec8.2bc6e","nodeId":"ns=2;s=Channel1.PLC01.Global.GESTRUCK_EST","datatype":"Int16","value":"2","usingListener":false,"name":"GESTRUCK_EST Error 2","topic":"","x":2450,"y":780,"wires":[["ed8e7ff4.66836","389b9759.08bb08"]]},{"id":"3e284b63.e69bbc","type":"OPCUA-IIoT-Node","z":"85a15ec8.2bc6e","nodeId":"ns=2;s=Channel1.PLC01.Global.GESTRUCK_EST","datatype":"Int16","value":"1","usingListener":false,"name":"GESTRUCK_EST Error 1","topic":"","x":2450,"y":720,"wires":[["4b937e23.3a8b9c","389b9759.08bb08"]]},{"id":"2d640885.81d6d8","type":"debug","z":"85a15ec8.2bc6e","name":"","active":true,"console":"false","complete":"true","x":2330,"y":660,"wires":[]},{"id":"4b937e23.3a8b9c","type":"debug","z":"85a15ec8.2bc6e","name":"","active":true,"console":"false","complete":"true","x":2790,"y":700,"wires":[]},{"id":"ed8e7ff4.66836","type":"debug","z":"85a15ec8.2bc6e","name":"","active":true,"console":"false","complete":"true","x":2790,"y":760,"wires":[]},{"id":"836f6f11.ef4b78","type":"debug","z":"85a15ec8.2bc6e","name":"","active":true,"console":"false","complete":"true","x":2794.31982421875,"y":837.8499755859375,"wires":[]},{"id":"2b1a053f.795e9a","type":"OPCUA-IIoT-Inject","z":"85a15ec8.2bc6e","injectType":"write","payload":"1","payloadType":"num","topic":"","repeat":"","crontab":"","once":false,"startDelay":"","name":"NODE_RED_GESTRUCK Error 1","addressSpaceItems":[{"name":"GESTRUCK_EST Error 1","nodeId":"ns=2;s=Channel1.PLC01.Global.GESTRUCK_EST","datatypeName":"Int16"}],"x":2400,"y":520,"wires":[["389b9759.08bb08","937bd26b.e66aa"]]},{"id":"937bd26b.e66aa","type":"debug","z":"85a15ec8.2bc6e","name":"","active":true,"console":"false","complete":"true","x":2630,"y":400,"wires":[]},{"id":"389b9759.08bb08","type":"OPCUA-IIoT-Write","z":"85a15ec8.2bc6e","connector":"f55a3722.1d7d18","name":"WRITE ERROR","justValue":false,"showStatusActivities":true,"showErrors":true,"x":2986.000099182129,"y":330.0000591278076,"wires":[["f9510e58.0ddcd","31ec2f77.18e28"]]},{"id":"31ec2f77.18e28","type":"OPCUA-IIoT-Response","z":"85a15ec8.2bc6e","name":"","showStatusActivities":false,"showErrors":false,"x":3217.9999389648438,"y":329.9999942779541,"wires":[["cefd5dff.48f0c"]]},{"id":"f9510e58.0ddcd","type":"debug","z":"85a15ec8.2bc6e","name":"","active":false,"console":"false","complete":"true","x":3210,"y":380,"wires":[]},{"id":"cefd5dff.48f0c","type":"debug","z":"85a15ec8.2bc6e","name":"","active":true,"console":"false","complete":"true","x":3387.9999389648438,"y":329.9999942779541,"wires":[]},{"id":"f55a3722.1d7d18","type":"OPCUA-IIoT-Connector","z":"","discoveryUrl":"","endpoint":"opc.tcp://192.168.9.1:49320/","keepSessionAlive":true,"loginEnabled":false,"securityPolicy":"None","securityMode":"NONE","name":"Harinalia","showErrors":false,"publicCertificateFile":"","privateKeyFile":"","defaultSecureTokenLifetime":"60000","endpointMustExist":false,"autoSelectRightEndpoint":false}]

Thank you so much

biancode, 2018-03-29:
please test v2.5.1, otherwise it needs a flexible control of the connector (#44)

closed 2018-03-29 by biancode