BiancoRoyal/node-red-contrib-iiot-opcua #48: After upgrade to 2.5.2 when adding a new connector error `TypeError: Connector invalid` occurs

w4tsn, 2018-04-04:
**I'm submitting a ...**  (check one with "x")

### Problem
 - (X) bug report => please search github for a similar issue or PR before submitting
 - ( ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
After upgrading to the newest version (2.5.2) and creating a new connector the runtime responds with `TypeError: Connector not Valid`, while the read and write nodes using this connector remain in a `connecting...` state.
 
 **Expected behavior**
The new created connector should normally connect.
 
 **Minimal reproduction of the problem with instructions**

1. Setup node-red
2. Install node-red-contrib-iiot-opcua@2.5.2
3. Create a server node (e.g. on port 55380)
4. Create a read-node
5. Create a new connector (eg. on port 55380)

It should fail to connect with the above error.

I replicated this behaviour on my live-system as well as a fresh docker node image.

## Notes 

After playing around with the error I got a work-around by removing the pre-set parameters within the strategy tab. This work-around works on both normal and docker setup.

## Your Environment
**Please tell us about your environment:**

  * (X) I am using just the Node-RED package and got node-opcua installed with it
  * ( ) I have installed node-opcua global from source (using git clone)
  * ( ) I have installed node-opcua as a global package (using npm i -g)

  * Device:
    
    - ( ) macOS: v 
    - (X) Linux: Linux 3.4.113-sun8i #18 SMP PREEMPT armv7l GNU/Linux
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name: 
         - Version:
   
   * Description of the connecting OPC UA system:
   
     - Name: node-red-contrib-iiot-opcua read/write nodes (connector)
     - Version: 2.5.2
     - Vendor:
     - Link: https://
     
* **node-red-contrib-iiot-opcu version:** `` 2.5.2
* **Node:** (min. 8.x)
   `node --version` = 8.9.4

biancode, 2018-04-05:
[image]

that default should work - right?

biancode, 2018-04-05:
[image]

that is default from the lib

biancode, 2018-04-05:
How does it work for you? What are the parameters to get it work again?

w4tsn, 2018-04-06:
Oddly it works when removing the defaults in the strategy tab. I noticed that the old working connectors have no values set in the tab, the new ones have. Removing these pre-set values makes the new ones work.

Maybe the max delay of 10000 is to low?

biancode, 2018-04-07:
fixed with v2.6.0 - reopen if problem is not solved, please

closed 2018-04-07 by biancode