BiancoRoyal/node-red-contrib-iiot-opcua #50: reconnection after timeout

reijosirila, 2018-04-14:
### Problem
 - (X ) bug report => please search github for a similar issue or PR before submitting
 - ( ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
If you subscribe monitoring, and stop OPC server, then start OPC server back, it doesn't re-subscribe items and node-red crasches; 
 
 **Expected behavior**
Subcribe to all items after reconnection; 

 **Minimal reproduction of the problem with instructions**
   1. start two different node-red engines (so you can stop server)
   2. subscription flow: 
`[{"id":"6079e9d1.cf1128","type":"OPCUA-IIoT-Listener","z":"c698100c.39382","connector":"f3a4064b.588be8","action":"subscribe","queueSize":"50","name":"OPC` `Monitor","justValue":true,"showStatusActivities":true,"showErrors":true,"x":410,"y":200,"wires":[["aa92cfb3.f6f78"]]},{"id":"5f122c24.b32004","type":"OPCUA-IIoT-Inject","z":"c698100c.39382","injectType":"listen","payload":"1000","payloadType":"num","topic":"","repeat":"","crontab":"","once":false,"startDelay":"0","name":"","addressSpaceItems":[{"name":"","nodeId":"ns=4;s=.a","datatypeName":""},{"name":"","nodeId":"ns=4;s=.b","datatypeName":""}],"x":190,"y":200,"wires":[["6079e9d1.cf1128"]]},{"id":"aa92cfb3.f6f78","type":"debug","z":"c698100c.39382","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","x":590,"y":180,"wires":[]},{"id":"f3a4064b.588be8","type":"OPCUA-IIoT-Connector","z":"","discoveryUrl":"","endpoint":"opc.tcp://127.0.0.1:55380","keepSessionAlive":true,"loginEnabled":false,"securityPolicy":"None","securityMode":"NONE","name":"DEVELOPMENT","showErrors":true,"publicCertificateFile":"","privateKeyFile":"","defaultSecureTokenLifetime":"","endpointMustExist":false,"autoSelectRightEndpoint":true,"strategyMaxRetry":"9999","strategyInitialDelay":"1000","strategyMaxDelay":"30000","strategyRandomisationFactor":"","requestedSessionTimeout":"30000"}]`
   3. Server flow: 
`[{"id":"fe83deaf.1c2d","type":"debug","z":"b75ebf50.46e1","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":736,"y":231,"wires":[]},{"id":"f74155e7.e80048","type":"OPCUA-IIoT-Flex-Server","z":"b75ebf50.46e1","port":"55380","endpoint":"","acceptExternalCommands":true,"maxAllowedSessionNumber":"","maxConnectionsPerEndpoint":"","maxAllowedSubscriptionNumber":"","alternateHostname":"","name":"","showStatusActivities":true,"showErrors":true,"allowAnonymous":true,"isAuditing":false,"serverDiscovery":true,"users":[],"xmlsets":[],"publicCertificateFile":"","privateCertificateFile":"","maxNodesPerRead":1000,"maxNodesPerBrowse":2000,"addressSpaceScript":"function constructAlarmAddressSpace(server, addressSpace, eventObjects, done) {\n    // server = the created node-opcua server\n    // addressSpace = script placeholder\n    // eventObjects = to hold event variables in memory from this script\n\n    // internal global sandbox objects are\n    // node = node of the flex server,\n    // coreServer = core iiot server object for debug and access to nodeOPCUA,\n    // and scriptObjects to hold variables and functions\n\n    // globals are to find on node.context().global and should be initialized here\n    node.context().global.set(\"TestOPCUAVarValue\", {})\n\n    coreServer.internalDebugLog('init dynamic address space')\n    node.warn('construct new address space for OPC UA')\n\n\n    // from here - see the node-opcua docs how to build address sapces\n    let tanks = addressSpace.addObject({\n        browseName: 'MyTest',\n        description: 'The Object representing some boxes',\n        organizedBy: addressSpace.rootFolder.objects,\n        notifierOf: addressSpace.rootFolder.objects.server\n    })\n\n\n    let myVariables = addressSpace.addObject({\n        browseName: 'MyVariables',\n        description: 'The Object representing some variables',\n        organizedBy: addressSpace.rootFolder.objects,\n        notifierOf: addressSpace.rootFolder.objects.server\n    })\n\n    if(coreServer.core) {\n        addressSpace.addVariable({\n            componentOf: myVariables,\n            nodeId: 'ns=4;s=.a',\n            browseName: 'bySecurityStatus',\n            displayName: 'bySecurityStatus',\n            dataType: 'Double',\n            value: {\n                get: function () {\n                    return new coreServer.core.nodeOPCUA.Variant({\n                        dataType: 'Double',\n                        value: scriptObjects.a\n                    })\n                },\n                set: function (variant) {\n                    scriptObjects.a = parseFloat(variant.value)\n                    return coreServer.core.nodeOPCUA.StatusCodes.Good\n                }\n            }\n\n        });\n        addressSpace.addVariable({\n            componentOf: myVariables,\n            nodeId: 'ns=4;s=.b',\n            browseName: 'bySecurityStatus',\n            displayName: 'bySecurityStatus',\n            dataType: 'Double',\n            value: {\n                get: function () {\n                    return new coreServer.core.nodeOPCUA.Variant({\n                        dataType: 'Double',\n                        value: scriptObjects.b\n                    })\n                },\n                set: function (variant) {\n                    scriptObjects.b = parseFloat(variant.value)\n                    return coreServer.core.nodeOPCUA.StatusCodes.Good\n                }\n            }\n\n        })\n\n    } else {\n        coreServer.internalDebugLog('coreServer.core needed for coreServer.core.nodeOPCUA')\n    }\n\n    done()\n}\n","x":535,"y":108,"wires":[["d34a1e70.9051f"]]},{"id":"d34a1e70.9051f","type":"debug","z":"b75ebf50.46e1","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":755,"y":108,"wires":[]},{"id":"75a4cfc1.dd71d","type":"OPCUA-IIoT-Inject","z":"b75ebf50.46e1","injectType":"write","payload":"","payloadType":"date","topic":"","repeat":"8","crontab":"","once":false,"startDelay":"","name":"change simulated box security status","addressSpaceItems":[{"name":"bySecurityStatus","nodeId":"ns=4;s=.a","datatypeName":"Double"}],"x":155,"y":228,"wires":[["9d7969ee.46bad8"]]},{"id":"9d7969ee.46bad8","type":"function","z":"b75ebf50.46e1","name":"random state","func":"var states = [0, 1, 11];\nvar random = states[Math.floor(Math.random()*states.length)];\n\nmsg.valuesToWrite = [random]; \nreturn msg;","outputs":1,"noerr":0,"x":415,"y":228,"wires":[["1d451586.55e5ca"]]},{"id":"1d451586.55e5ca","type":"OPCUA-IIoT-Write","z":"b75ebf50.46e1","connector":"1add6727.060519","name":"","justValue":true,"showStatusActivities":true,"showErrors":true,"x":575,"y":228,"wires":[["fe83deaf.1c2d"]]},{"id":"98d4be44.0f8b4","type":"OPCUA-IIoT-Server-Command","z":"b75ebf50.46e1","commandtype":"restart","nodeId":"","name":"","x":285,"y":108,"wires":[["f74155e7.e80048"]]},{"id":"8b082c6d.feb82","type":"OPCUA-IIoT-Inject","z":"b75ebf50.46e1","injectType":"inject","payload":"","payloadType":"date","topic":"","repeat":"","crontab":"","once":false,"startDelay":"","name":"","addressSpaceItems":[],"x":105,"y":108,"wires":[["98d4be44.0f8b4"]]},{"id":"1add6727.060519","type":"OPCUA-IIoT-Connector","z":"","discoveryUrl":"","endpoint":"opc.tcp://127.0.0.1:55380","keepSessionAlive":true,"loginEnabled":false,"securityPolicy":"None","securityMode":"NONE","name":"DEVELOPMENT","showErrors":true,"publicCertificateFile":"","privateKeyFile":"","defaultSecureTokenLifetime":"","endpointMustExist":false,"autoSelectRightEndpoint":true,"strategyMaxRetry":"0","strategyInitialDelay":"2000","strategyMaxDelay":"3000","strategyRandomisationFactor":"","requestedSessionTimeout":"3000"}]`
  
Start both flows, different nod-red engines; After subscription, stop opc server node-red engine; 
Wait a while and then start it again; it will fail to subscribe back all the nodes: 
` [error] [OPCUA-IIoT-Listener:OPC Monitor] TypeError: Cannot read property 'deleteMonitoredItems' of null`
Most of the cashes whole node-red crashes: 
`
./node-red/node-red-contrib-iiot-opcua/node_modules/node-opcua-client/src/client_monitored_item_base.js:169
    assert(subscription.session);
    ^
AssertionError [ERR_ASSERTION]: subscription.session
`

## Your Environment
Just latest node-red-contrib-iiot-opcua@2.6.1, node-red@v0.18.4, nodejs@v9.5.0

  * (X ) I am using just the Node-RED package and got node-opcua installed with it
  * ( ) I have installed node-opcua global from source (using git clone)
  * ( ) I have installed node-opcua as a global package (using npm i -g)
        
  * Device :
    
    - ( X) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   

* **Node:** (min. 8.x)
   `node --version` =v9.5.0

biancode, 2018-04-15:
https://github.com/node-opcua/node-opcua/issues/474

biancode, 2018-04-15:
please try v2.6.2 it is back on v0.2.2 of node-opcua and works well on my side

reijosirila, 2018-04-15:
Well, it's not working; For some reason it terminates one monitor: 

`
  opcuaIIoT:connector Session Created On opc.tcp://10.9.8.14:55380/UA/NodeREDFlexIIoTServer For on connect +276ms
  opcuaIIoT:listener:subscribe Monitored Item Unsubscribe ns=4;s=.starrStorages[35].bySecurityStatus +1m
  opcuaIIoT:listener Terminated For 1 +1m
  opcuaIIoT:listener Terminate Monitored Item ns=4;s=.starrStorages[35].bySecurityStatus +0ms
  opcuaIIoT:listener listening (1) +1ms

`

And with second break it crashes: 

` Server State =  Running
  opcuaIIoT:connector !!!!!!!!!!!!!!!!!!!!!!!!  CLIENT STARTING RECONNECTION !!!!!!!!!!!!!!!!!!! +56s
  opcuaIIoT:connector CONNECTION STARTING RECONNECTION: opc.tcp://10.9.8.14:55380/UA/NodeREDFlexIIoTServer +0ms

/Users/reijo/dev/kookon/node-red/node_modules/node-opcua-client/src/client_base.js:290
    assert(!self.isReconnecting);
    ^
AssertionError [ERR_ASSERTION]: !self.isReconnecting
`

But as I can see, on a master branch of node-opcua there is commit: https://github.com/node-opcua/node-opcua/commit/d5d9e51bc219de1ab242640714fbc40dc946d72b
Those yesterdays commits are related with my really old reconnection issue: https://github.com/node-opcua/node-opcua/issues/237  :) 👍

Nafling, 2018-08-19:
Hiya reij, did you ever solve this? I am getting a "Server State = Running" timeout, without a reconnect.

I am using a test OPC server from Prosys, but it should not be the issue.

I cannot seem to find any other thread that is talking about this issue.

biancode, 2018-10-10:
new in v3.0.0 - improved node-opcua since v0.4.6 - FSM to reconnect via states - very new reconnecting automatically via node-opcua

biancode, 2018-10-10:
open a new issue if it does not work from v3.0.0 (not released for now)

closed 2018-10-10 by biancode