BiancoRoyal/node-red-contrib-iiot-opcua #52: Writing Byte-Values with Write-Node

mnMichaelNeher, 2018-06-12:
**I'm submitting a ...**  (check one with "x")

 ## Feature
 - (X ) feature request
 
 **What is the motivation / use case for changing the behavior?**
 
 
I want to write Data with the Write-Node to a CodeSys-PLC. I have a lot of values which are formatted as "USINT". This datatyp is representet as "BYTE" in OPC UA. In the current node-red-contrib-iiot-opcua Version the datatyp "BYTE" is not implemented.
Is it possible to add this feature to this package, because i think it is not a lot of work to implement it, but it is useful for a lot of users.
 

## Your Environment
**Please tell us about your environment:**

  * ( ) I am using just the Node-RED package and got node-opcua installed with it
  * ( ) I have installed node-opcua global from source (using git clone)
  * ( X) I have installed node-opcua as a global package (using npm i -g)
        
  * Device :
    
    - ( ) macOS: v 
    - (X ) Linux: v 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
     
    
* **node-red-contrib-iiot-opcua version:** `2.6.2`

* **Node:** (min. 8.x)
   `node --version` = 0.18.4

StephanLeschke, 2018-08-23:
As long as specific data types are not supported they should not be available in the data type drop down lists.

biancode, 2018-10-14:
All values without any special implementation sending the data directly. That means it should work if you set the value correct to BYTE. If it does not work for you, then write down your use case in detail to reproduce it in a test case, please! Reopen if you have more details, please!

mnMichaelNeher, 2018-10-17:
Ok - tried it again with the newest Version of this module - i got the same result, it doesn't work - just the Error-Message now only says error.

My Use-Case:

I have a CodeSys-PLC with an OPC-UA-Server. I can read and write all configured Variables apart from that of Type "USINT" (it's a 8-bit-Variable with the Range from 0 to 255).

In the OPC-UA-Client UA-Expert i can also read an write these Variables, which are represented with the Datatyp "Byte". (See Screenshot)
[image]

My Node-Red-Flow looks like this:

[{"id":"dfc621ab.d2262","type":"debug","z":"a3d1b4ec.b83f78","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","x":783.2000122070312,"y":735.2000122070312,"wires":[]},{"id":"331e0bab.b9edbc","type":"OPCUA-IIoT-Response","z":"a3d1b4ec.b83f78","name":"","compressStructure":true,"showStatusActivities":false,"showErrors":false,"activateFilters":false,"filters":[],"x":623.5,"y":742,"wires":[["dfc621ab.d2262"]]},{"id":"90edfcb2.00f35","type":"debug","z":"a3d1b4ec.b83f78","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","x":608.2000122070312,"y":699.2000122070312,"wires":[]},{"id":"585f5bb4.782db4","type":"OPCUA-IIoT-Write","z":"a3d1b4ec.b83f78","connector":"ed458005.5c597","name":"","justValue":false,"showStatusActivities":false,"showErrors":true,"x":429.00001525878906,"y":744.2000122070312,"wires":[["90edfcb2.00f35","331e0bab.b9edbc"]]},{"id":"d287966.9d8df68","type":"OPCUA-IIoT-Inject","z":"a3d1b4ec.b83f78","injectType":"write","payload":"","payloadType":"date","topic":"","repeat":"","crontab":"","once":true,"startDelay":"","name":"inject","addressSpaceItems":[{"name":"TestReadWrite","nodeId":"ns=4;s=|var|Nerve MFN100 Edge Gateway.Application.Global.gastMappings[100].nblockId","datatypeName":"Byte"}],"x":113.00001525878906,"y":746.2000122070312,"wires":[["315ed838.6f36c"]]},{"id":"315ed838.6f36c","type":"function","z":"a3d1b4ec.b83f78","name":"","func":"msg.valuesToWrite = [2];\nreturn msg;","outputs":1,"noerr":0,"x":272.49998474121094,"y":746.6000213623047,"wires":[["585f5bb4.782db4","90edfcb2.00f35"]]},{"id":"ed458005.5c597","type":"OPCUA-IIoT-Connector","z":"","discoveryUrl":"","endpoint":"opc.tcp://20.20.20.1:4840","keepSessionAlive":true,"loginEnabled":false,"securityPolicy":"None","securityMode":"NONE","name":"","showErrors":false,"publicCertificateFile":"","privateKeyFile":"","defaultSecureTokenLifetime":"","endpointMustExist":false,"autoSelectRightEndpoint":false,"strategyMaxRetry":"","strategyInitialDelay":"","strategyMaxDelay":"","strategyRandomisationFactor":"","requestedSessionTimeout":"","connectionStartDelay":"","reconnectDelay":""}]

If i inject the Input the following message goes into the Write-Node:

[image]

But as Output of the Write-Node i just get the Message: "Error: Error"

[image]

Where is here the Problem?? With the other Datatypes my Testflow is really working fine? Where is the difference in this Datatype??

I'm looking forward to your reply.

biancode, 2018-10-17:
fixed from coming soon v3.1.0

[image]

closed 2018-10-17 by biancode