BiancoRoyal/node-red-contrib-iiot-opcua #59: How to set DataChangeFilter in listener

xshan8333, 2018-10-15:
### Problem
 - (X) bug report => please search github for a similar issue or PR before submitting
 - ( ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
I get Square1 data from Prosys simulation server. but get changed data only!
 
 **Expected behavior**
 
  I want DataChangeFilter=StatusValueTimestamp. How can i set it?

 **Minimal reproduction of the problem with instructions**
 
   1.inject is  { "interval": 1000,   "queueSize": 1}
   2. Prosys sim server log is:
  ItemsToCreate=MonitoredItemCreateRequest[1]
    [0]=MonitoredItemCreateRequest
    ItemToMonitor=ReadValueId
      NodeId=ns=5;s=Square1
      AttributeId=13
      IndexRange=null
      DataEncoding=null
    MonitoringMode=MonitoringMode
      name=Reporting
      ordinal=2
    RequestedParameters=MonitoringParameters
      ClientHandle=2
      SamplingInterval=1000.0
      Filter=null
      QueueSize=1
      DiscardOldest=true
   3. 
   

## Your Environment
**Please tell us about your environment:**

  * (X) I am using just the Node-RED package and got node-opcua installed with it
  * ( ) I have installed node-opcua global from source (using git clone)
  * (X) I have installed node-opcua as a global package (using npm i -g)
        
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - (X) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting OPC UA system:
   
     - Name   : prosys opc ua simulation server
     - Version: v3.0.0-157
     - Vendor : prosys
     
* **node-red-contrib-iiot-opcua version:** ``
  node-red-contrib-iiot-opcua@2.7.1

* **Node:** (min. 8.x)
   node = v8.12.0
   node-red = V0.19.4

biancode, 2018-10-15:
Hi, 

- was it working in the versions before? (you set bug report, not unexpected behaviour)
- is it node-opcua v0.4.6 in the global install?
- could you build some example to get it to work with node-opcua 0.4.6?

cya
Klaus

biancode, 2018-10-15:
API docs

>  * options.trigger       {DataChangeTrigger} {Status|StatusValue|StatusValueTimestamp}
> * options.deadbandType  {DeadbandType}      {None|Absolute|Percent}
> * options.deadbandValue {Double}

xshan8333, 2018-10-15:
Hi, biancode:
I just study OPC UA with node-red.  be prepare for access some PLC device.
I use node-red-contrib-iiot-opcua  for first time now, so i need some help.
sorry set bug report, but i donot known how to set  "DataChangeTrigger ..." options in node-red.
next is my flow in node-red. maybe give me some advice?
thanks.

[
    {
        "id": "a6f7c370.9501a8",
        "type": "tab",
        "label": "opc listen",
        "disabled": false,
        "info": ""
    },
    {
        "id": "b53de21a.c7a54",
        "type": "OPCUA-IIoT-Listener",
        "z": "a6f7c370.9501a8",
        "connector": "70d1df5f.2610f",
        "action": "subscribe",
        "queueSize": "1",
        "name": "",
        "justValue": true,
        "showStatusActivities": false,
        "showErrors": false,
        "x": 360,
        "y": 120,
        "wires": [
            [
                "e1aa9291.176f7",
                "768cc25c.ecc684"
            ]
        ]
    },
    {
        "id": "b94f91b8.790088",
        "type": "OPCUA-IIoT-Inject",
        "z": "a6f7c370.9501a8",
        "injectType": "listen",
        "payload": "{\"interval\":1000,\"queueSize\":1}",
        "payloadType": "json",
        "topic": "",
        "repeat": "",
        "crontab": "",
        "once": true,
        "startDelay": "",
        "name": "listen Square1",
        "addressSpaceItems": [
            {
                "name": "",
                "nodeId": "ns=5;s=Square1",
                "datatypeName": ""
            }
        ],
        "x": 160,
        "y": 120,
        "wires": [
            [
                "b53de21a.c7a54"
            ]
        ]
    },
    {
        "id": "768cc25c.ecc684",
        "type": "debug",
        "z": "a6f7c370.9501a8",
        "name": "",
        "active": true,
        "tosidebar": true,
        "console": false,
        "complete": "true",
        "x": 850,
        "y": 120,
        "wires": []
    },
    {
        "id": "e1aa9291.176f7",
        "type": "OPCUA-IIoT-Response",
        "z": "a6f7c370.9501a8",
        "name": "",
        "showStatusActivities": false,
        "showErrors": true,
        "x": 520,
        "y": 180,
        "wires": [
            [
                "29963671.fed68a"
            ]
        ]
    },
    {
        "id": "29963671.fed68a",
        "type": "function",
        "z": "a6f7c370.9501a8",
        "name": "",
        "func": "var msg1 = {};\n\nmsg1.topic = msg.addressSpaceItems[0].nodeId;\nmsg1.payload = msg.payload.value.value;\nmsg1.timestamp = msg.payload.sourceTimestamp;\n\n//{topic:\"name\", payload:value, timestamp:1520527095000}\nreturn msg1;",
        "outputs": 1,
        "noerr": 0,
        "x": 670,
        "y": 180,
        "wires": [
            [
                "768cc25c.ecc684"
            ]
        ]
    },
    {
        "id": "70d1df5f.2610f",
        "type": "OPCUA-IIoT-Connector",
        "z": "",
        "discoveryUrl": "",
        "endpoint": "opc.tcp://10.190.41.9:53530/OPCUA/SimulationServer",
        "keepSessionAlive": true,
        "loginEnabled": true,
        "securityPolicy": "None",
        "securityMode": "NONE",
        "name": "PROSYS SERVER",
        "showErrors": true,
        "publicCertificateFile": "",
        "privateKeyFile": "",
        "defaultSecureTokenLifetime": "60000",
        "endpointMustExist": false,
        "autoSelectRightEndpoint": false,
        "strategyMaxRetry": "",
        "strategyInitialDelay": "",
        "strategyMaxDelay": "",
        "strategyRandomisationFactor": "",
        "requestedSessionTimeout": ""
    }
]

biancode, 2018-10-15:
it works as expected via listener parameters

`{
    "interval": 200,
    "queueSize": 10,
    "options": {
        "requestedPublishingInterval": 1000,
        "requestedLifetimeCount": 1000,
        "requestedMaxKeepAliveCount": 12,
        "maxNotificationsPerPublish": 100,
        "publishingEnabled": true,
        "priority": 6,
        "trigger": "StatusValueTimestamp"
    }
}`

biancode, 2018-10-15:
That is to set from your first inject to the listener. It does not change with next option injects without a deploy before.

closed 2018-10-15 by biancode