BiancoRoyal/node-red-contrib-iiot-opcua #67: Flex server - namespace not accepted anymore

Toppert93, 2018-10-19:
Hello,

When installing the new 3.1 version, the flex server won't take anything else than ns=1 when adding variables to the address space. Is that true or just an issue?

I had every variable on the ns=2, I guess I will need to change this when updating.

Cordially,

Steve

biancode, 2018-10-19:
Hell Steve,

node-opcua has a new namespace concept with v0.4. You will find some examples in the code of node-opcua, how to work with that to set ns=2 as your namespace in the flex server. It is a breaking change here and I decided to use the ns=1 as the default does.

If I find some time I could attach an example here and to the Node-RED examples with the Flex Server.

Happy wiring!
Klaus

Toppert93, 2018-10-19:
Thanks a lot Klaus for your time and your answer ;)

Steve

biancode, 2018-10-20:
`const ns2 = addressSpace.registerNamespace("N2");`
`ns2.add...`

closed 2018-10-20 by biancode