BiancoRoyal/node-red-contrib-iiot-opcua #69: node-red shutdown after connection loss and reconnect attempt

maccap, 2018-11-02:
## Problem
 - (x) bug report => please search github for a similar issue or PR before submitting
 - ( ) unexpected behavior => please visit the Wiki before

### Information
 **Current behavior**
node-red is working as opc-client and doing a simple read tasks of several variables in a 2 second interval. Everything works fine until I disconnect the Ethernet-cable to the plc system (opc server). I get a timeout after 60.000ms regardless which time I set in the connector node. When I reconnect the cable it takes some time and then suddenly my whole node-red collaps. This does not occur everytime, but 2-4 times out of 10. If I start node-red without connection to the opc server and then connect the cable to the server it always crashes...

This is my error-log:
2 Nov 11:02:13 - [error] [OPCUA-IIoT-Connector:MEBA] Error: Transaction has timed out ( timeout = 60000 ms)

    at Timeout._onTimeout (/data/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:1020:27)

    at ontimeout (timers.js:498:11)

    at tryOnTimeout (timers.js:323:5)

    at Timer.listOnTimeout (timers.js:290:5)

 Timeout .... waiting for response for  CloseSessionRequest { /*RequestHeader*/

 authenticationToken           /* NodeId                           */: ns=0;i=56

 timeStamp                     /* UtcTime                          */: 2018-11-02T11:01:13.981Z

 requestHandle                 /* IntegerId                        */: 76               0x4c

 returnDiagnostics             /* UInt32                           */: 0               0x0

 auditEntryId                  /* UAString                         */: 

 timeoutHint                   /* UInt32                           */: 0               0x0

 additionalHeader              /* ExtensionObject                  */: null

};

ERROR !!! , please check here !!!! callback may be called twice !!  Error: Callback was already called.

    at /data/node_modules/async/dist/async.js:966:32

    at /data/node_modules/async/dist/async.js:1137:13

    at /data/node_modules/node-opcua-client/src/client_base.js:868:13

    at /data/node_modules/node-opcua-client/src/opcua_client.js:728:9

    at /data/node_modules/node-opcua-client/src/opcua_client.js:693:13

    at Object.<anonymous> (/data/node_modules/node-opcua-client/src/client_session.js:1168:20)

    at Object.modified_callback [as callback] (/data/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:1005:28)

    at ClientSecureChannelLayer.cancelPendingTransactions (/data/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:1298:25)

    at ClientSecureChannelLayer.close (/data/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:1319:10)

    at OPCUAClientBase.disconnect (/data/node_modules/node-opcua-client/src/client_base.js:964:21)

ERROR !!! , please check here !!!! callback may be called twice !!  Error: Callback was already called.

    at /data/node_modules/async/dist/async.js:966:32

    at /data/node_modules/async/dist/async.js:1137:13

    at /data/node_modules/node-opcua-client/src/client_base.js:868:13

    at /data/node_modules/node-opcua-client/src/opcua_client.js:728:9

    at /data/node_modules/node-opcua-client/src/opcua_client.js:693:13
[... 68 lines cut ...]
  * ( ) I have installed node-opcua global from source (using git clone)
  * ( ) I have installed node-opcua as a global package (using npm i -g)
        
  * Device :
    
    - ( ) macOS: v 
    - (x) Linux: v 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting OPC UA system:
   
     - Name: CODESYS_OPC_UA_Server
     - Version: 3.9.0.7
     - Vendor : Lenze
     - Link   : https://
     
    
* **node-red-contrib-iiot-opcua version:** ``

* **Node:** (min. 8.x)
   `node --version` = 8.12.10

biancode, 2018-11-02:
@erossignon do you have some idea about that?

biancode, 2018-11-02:
It never happens with a software OPC UA server. I'll test it with PLC200 if I find some time.

maccap, 2018-11-02:
In order to save time I can provide you openvpn access to my PLC device

biancode, 2018-11-02:
> In order to save time I can provide you openvpn access to my PLC device

Great - sure send me an e-mail., please

maccap, 2018-11-02:
PLC IP address 192.168.5.99

biancode, 2018-11-02:
test protocol: VPN disconnected 
"Error: Transaction has timed out ( timeout = 60000 ms)" - raises never any timeout - wait was 4 min

biancode, 2018-11-02:
tiemout is fixed in node-opcua channel wiht min 30sec. 

```
request.requestHeader.timeoutHint || defaultTransactionTimeout (60sec.)
timeout = Math.max(minTransactionTimeout, timeout);
```

biancode, 2018-11-02:
same test with VPN brings now "Error: Connection Break" and all is working well

biancode, 2018-11-04:
Please, test v3.4.0!

maccap, 2018-11-05:
Hello Klaus,
Great work. Now nodered running stable after reconnecting the connection cable.
I have tracked down two different behaviors regarding the fault state.
Max Delay is set to 30s in connector mode

case a) after 60s when cable is disconnected the following message is shown:
"Error: Transaction has timed out ( timeout = 60000 ms)"
"Error: Client Not SESSIONACTIVE On Read"
When I reconnect the cable it takes 60s again to get my connection reestablished and read correct data. In this case the max delay time is not working properly.

case b) If I start nodered with disconnected cable I get this output:
"Error: Client Not SESSIONACTIVE On Read"
and 10s later this:
"Error: Backoff aborted."
When I connect the cable it takes about 2- 4 s and the connection is working properly again

The main issue is fixed with your new version.
Thank you very much.

biancode, 2018-11-05:
case a) please @erossignon could you answer this? - I mean it is fixed in node-opcua by the client channel transaction timeout (_performMessageTransaction / defaultTransactionTimeout) for now - the session is to control and would use your settings

closed 2018-11-05 by biancode