BiancoRoyal/node-red-contrib-iiot-opcua #70: Result filter works in version 2.7.1 but not in 3.4.1

jewendt, 2018-11-06:
Setup a simple inject, read, filter request with one tag however the filter returns null with version 3.4.1.  Reinstalled 2.7.1 and it works perfectly.  I am running 19.4 version of node red utilizing Node version 10.13.0 on Windows 10.

biancode, 2018-11-06:
Some example to reproduce that?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

jewendt, 2018-11-06:
Sure. This flow works fine in 2.7.1.  I thought I was losing my mind.  About as basic as you can get unless I am missing something. 

[{"id":"7745920a.b8d67c","type":"tab","label":"Flow 5","disabled":false,"info":""},{"id":"a0845b5.42933a8","type":"OPCUA-IIoT-Inject","z":"7745920a.b8d67c","injectType":"inject","payload":"","payloadType":"date","topic":"","repeat":"","crontab":"","once":false,"startDelay":"","name":"","addressSpaceItems":[{"name":"","nodeId":"ns=2;s=Channel5.Device1.Global.LINE_10_BARCODE.DATA","datatypeName":""}],"x":124.10000610351562,"y":99.80000305175781,"wires":[["1c6124b.348acdb"]]},{"id":"1c6124b.348acdb","type":"OPCUA-IIoT-Read","z":"7745920a.b8d67c","attributeId":0,"maxAge":1,"depth":1,"connector":"1a2d29f3.31d5c6","name":"","justValue":true,"showStatusActivities":false,"showErrors":false,"parseStrings":false,"x":193.10001373291016,"y":25.800004959106445,"wires":[["c708e125.042e1","627656c9.dc7758"]]},{"id":"c708e125.042e1","type":"OPCUA-IIoT-Result-Filter","z":"7745920a.b8d67c","nodeId":"ns=2;s=Channel5.Device1.Global.LINE_10_BARCODE.DATA","datatype":"","fixedValue":false,"fixPoint":2,"withPrecision":false,"precision":2,"entry":1,"justValue":true,"withValueCheck":false,"minvalue":"","maxvalue":"","defaultvalue":"","topic":"","name":"","showErrors":false,"x":564.1000900268555,"y":51.80000591278076,"wires":[["b7311ce3.fbd3e"]]},{"id":"627656c9.dc7758","type":"debug","z":"7745920a.b8d67c","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":548.1000366210938,"y":208.39999389648438,"wires":[]},{"id":"b7311ce3.fbd3e","type":"debug","z":"7745920a.b8d67c","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":577.1000366210938,"y":122.40000915527344,"wires":[]},{"id":"1a2d29f3.31d5c6","type":"OPCUA-IIoT-Connector","z":"","discoveryUrl":"","endpoint":"opc.tcp://Cgshopopsdev5.andersencorp.com:49320","keepSessionAlive":true,"loginEnabled":false,"securityPolicy":"None","securityMode":"NONE","name":"LOCAL SERVER","showErrors":false,"publicCertificateFile":"","privateKeyFile":"","defaultSecureTokenLifetime":"","endpointMustExist":true,"autoSelectRightEndpoint":true,"strategyMaxRetry":"","strategyInitialDelay":"","strategyMaxDelay":"","strategyRandomisationFactor":"","requestedSessionTimeout":""}]

jewendt, 2018-11-06:
I see you tagged it as a bug. Were you able to recreate the problem or do you need more details?

biancode, 2018-11-06:
Yes, it is a problem with the Read node output. It was out of my scope because I do just use them with the Listener node. I write some tests and release a fix for that soon.

jewendt, 2018-11-06:
Thanks.  I was surprised that this wasn't caught before now. so I was thinking I was using the nodes wrong.   Just for my education, can you explain the difference in intent between utilizing the listening versus read node?

biancode, 2018-11-07:
> Thanks. I was surprised that this wasn't caught before now. so I was thinking I was using the nodes wrong. Just for my education, can you explain the difference in intent between utilizing the listening versus read node?

[Watch this](https://www.youtube.com/watch?v=JCNm_7yDKcQ&list=PLdcYFhtEB7oke6US0tPIOLgp4oPe4cami)

jewendt, 2018-11-10:
#biancode I tested this again and the filter now isn't returning anything at all.  Before it was returning null now, it is not doing anything.   Should I open up another issue?  Here is my test case

[{"id":"8eef632e.f4387","type":"tab","label":"Flow 9","disabled":false,"info":""},{"id":"6eb3a8e0.2efa88","type":"OPCUA-IIoT-Read","z":"8eef632e.f4387","attributeId":0,"maxAge":1,"depth":1,"connector":"9313e652.2ea988","name":"","justValue":true,"showStatusActivities":true,"showErrors":true,"parseStrings":false,"historyDays":1,"x":425.0000305175781,"y":76,"wires":[["b76cc4e2.97b278","d7b2a39b.c14a8"]]},{"id":"b12b549a.250b28","type":"OPCUA-IIoT-Inject","z":"8eef632e.f4387","injectType":"read","payload":"","payloadType":"date","topic":"","repeat":"","crontab":"","once":false,"startDelay":"","name":"","addressSpaceItems":[{"name":"","nodeId":" ns=2;s=Channel5.Device1.Global.LINE_10_BARCODE.DATA","datatypeName":""},{"name":"","nodeId":" ns=2;s=Channel5.Device1.Global.LINE_10_BARCODE_COUNT.DATA","datatypeName":""}],"x":203,"y":86,"wires":[["6eb3a8e0.2efa88"]]},{"id":"b76cc4e2.97b278","type":"OPCUA-IIoT-Result-Filter","z":"8eef632e.f4387","nodeId":"ns=2;s=Channel5.Device1.Global.LINE_10_BARCODE_COUNT.DATA","datatype":"ns=0;i=12","fixedValue":false,"fixPoint":2,"withPrecision":false,"precision":2,"entry":"2","justValue":true,"withValueCheck":false,"minvalue":"","maxvalue":"","defaultvalue":"","topic":"","name":"","showErrors":true,"x":418.00006103515625,"y":211.8000030517578,"wires":[["73c8c4ad.07b97c"]]},{"id":"d7b2a39b.c14a8","type":"debug","z":"8eef632e.f4387","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":625.0000305175781,"y":70.5999984741211,"wires":[]},{"id":"73c8c4ad.07b97c","type":"debug","z":"8eef632e.f4387","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":815.0001220703125,"y":116.60000419616699,"wires":[]},{"id":"9313e652.2ea988","type":"OPCUA-IIoT-Connector","z":"","discoveryUrl":"","endpoint":"opc.tcp://Cgshopopsdev5.andersencorp.com:49320","keepSessionAlive":true,"loginEnabled":false,"securityPolicy":"None","securityMode":"NONE","name":"LOCAL SERVER","showErrors":false,"publicCertificateFile":"","privateKeyFile":"","defaultSecureTokenLifetime":"","endpointMustExist":false,"autoSelectRightEndpoint":false,"strategyMaxRetry":"","strategyInitialDelay":"","strategyMaxDelay":"","strategyRandomisationFactor":"","requestedSessionTimeout":"","maxBadSessionRequests":10}]

biancode, 2018-11-11:
Please, try to remove the data type on reading all instead of value!

biancode, 2018-11-11:
If you set a data type use the selection, please!

biancode, 2018-11-11:
If you just want to filter the name, please use the filter inside the response node!

jewendt, 2018-11-11:
Ok, yes I have it working.  But help me understand if there is any easy way to just read a simple value??? I am trying to develop a course on this and am just looking for the simplest way.  On those lines what is the difference between the IIOT-OPCUA and just OPCUA nodes?  Is one replacing the other?  Looks like you are involved it both.

[{"id":"de7e6820.990bc8","type":"tab","label":"Flow 1","disabled":false,"info":""},{"id":"12bf2fd8.7fb88","type":"OPCUA-IIoT-Inject","z":"de7e6820.990bc8","injectType":"inject","payload":"","payloadType":"date","topic":"","repeat":"","crontab":"","once":false,"startDelay":"","name":"","addressSpaceItems":[{"name":"","nodeId":"ns=2;s=Channel5.Device1.Global.LINE_10_BARCODE_COUNT.DATA","datatypeName":""},{"name":"","nodeId":"ns=2;s=Channel5.Device1.Global.LINE_10_BARCODE.DATA","datatypeName":""}],"x":167.10000610351562,"y":120.19999694824219,"wires":[["f14fcc59.75518"]]},{"id":"f14fcc59.75518","type":"OPCUA-IIoT-Read","z":"de7e6820.990bc8","attributeId":0,"maxAge":1,"depth":1,"connector":"ef402e5a.c0c36","name":"","justValue":true,"showStatusActivities":false,"showErrors":false,"parseStrings":false,"historyDays":1,"x":344.1000061035156,"y":113.19999694824219,"wires":[["26cc436e.82cecc","e6fa77c0.d8c2d8","a30bbf.e24d544"]]},{"id":"26cc436e.82cecc","type":"OPCUA-IIoT-Result-Filter","z":"de7e6820.990bc8","nodeId":"ns=2;s=Channel5.Device1.Global.LINE_10_BARCODE_COUNT.DATA","datatype":"","fixedValue":false,"fixPoint":2,"withPrecision":false,"precision":2,"entry":"1","justValue":true,"withValueCheck":false,"minvalue":"","maxvalue":"","defaultvalue":"","topic":"","name":"","showErrors":true,"x":703.1000671386719,"y":39.80000305175781,"wires":[["d7ef6386.d857e"]]},{"id":"e6fa77c0.d8c2d8","type":"OPCUA-IIoT-Result-Filter","z":"de7e6820.990bc8","nodeId":"ns=2;s=Channel5.Device1.Global.LINE_10_BARCODE.DATA","datatype":"","fixedValue":false,"fixPoint":2,"withPrecision":false,"precision":2,"entry":"2","justValue":true,"withValueCheck":false,"minvalue":"","maxvalue":"","defaultvalue":"","topic":"","name":"","showErrors":false,"x":677.1000823974609,"y":183.00000667572021,"wires":[["1045c626.29f4ea"]]},{"id":"a30bbf.e24d544","type":"OPCUA-IIoT-Response","z":"de7e6820.990bc8","name":"","compressStructure":true,"showStatusActivities":false,"showErrors":false,"activateUnsetFilter":false,"activateFilters":true,"negateFilter":false,"filters":[{"name":"nodeId","value":"s=Channel5.Device1.Global.LINE_10_BARCODE_COUNT.DATA"}],"x":535.1000633239746,"y":352.2000102996826,"wires":[[]]},{"id":"1045c626.29f4ea","type":"debug","z":"de7e6820.990bc8","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":677.1000366210938,"y":262.8000183105469,"wires":[]},{"id":"d7ef6386.d857e","type":"debug","z":"de7e6820.990bc8","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":677.1000366210938,"y":95.80000305175781,"wires":[]},{"id":"75f55273.3fac6c","type":"OpcUa-Client","z":"de7e6820.990bc8","endpoint":"580209d0.8e26c8","action":"read","time":10,"timeUnit":"s","name":"Simple Value Read","x":657.5,"y":474,"wires":[["3fdc50bb.9a26d"]]},{"id":"706aea5a.f28094","type":"OpcUa-Item","z":"de7e6820.990bc8","item":"ns=2;s=Channel5.Device1.Global.LINE_10_BARCODE.DATA","datatype":"String","value":"","name":"Static Double","x":449.5,"y":476.40000915527344,"wires":[["75f55273.3fac6c"]]},{"id":"3fdc50bb.9a26d","type":"debug","z":"de7e6820.990bc8","name":"","active":true,"console":"false","complete":"false","x":887.4999847412109,"y":472.8000030517578,"wires":[]},{"id":"acae793f.ca63f8","type":"inject","z":"de7e6820.990bc8","name":"Test Double","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"x":251.49998474121094,"y":476.6000061035156,"wires":[["706aea5a.f28094"]]},{"id":"180c52a6.b6264d","type":"comment","z":"de7e6820.990bc8","name":"READ","info":"","x":68,"y":478.40000915527344,"wires":[]},{"id":"ef402e5a.c0c36","type":"OPCUA-IIoT-Connector","z":"","discoveryUrl":"","endpoint":"opc.tcp://Cgshopopsdev5.andersencorp.com:49320","keepSessionAlive":true,"loginEnabled":false,"securityPolicy":"None","securityMode":"NONE","name":"LOCAL SERVER","showErrors":true,"individualCerts":false,"publicCertificateFile":"","privateKeyFile":"","defaultSecureTokenLifetime":"","endpointMustExist":false,"autoSelectRightEndpoint":false,"strategyMaxRetry":"","strategyInitialDelay":"","strategyMaxDelay":"","strategyRandomisationFactor":"","requestedSessionTimeout":"","connectionStartDelay":"","reconnectDelay":"","maxBadSessionRequests":"10"},{"id":"580209d0.8e26c8","type":"OpcUa-Endpoint","z":"","endpoint":"opc.tcp://Cgshopopsdev5.andersencorp.com:49320","secpol":"None","secmode":"NONE","login":false}]

biancode, 2018-11-11:
I do a refactoring of the whole project at the moment and it will get more tests in the future.

jewendt, 2018-11-11:
Ok, so to try and be "future proof" I should be using the IIOT-OPCUA one?  BTW...thanks for all your hard work on this!

biancode, 2018-11-11:
yes, please, use the IIoT OPC UA package and give me your feedback! It should become stable again in the next few days.

biancode, 2018-11-11:
fixed with v3.6.+

closed 2018-11-11 by biancode