BiancoRoyal/node-red-contrib-iiot-opcua #71: OPCUA nodes breaking other nodes

jewendt, 2018-11-06:
I recently discovered that after installing the OPCUA nodes some other nodes stopped working.  I worked with one of the developers to "fix" their nodes but when I asked them what they found they had this response. 

"The other package seems to hook the global function "Array.from()" and there
version was not working correctly. I've update our library to no more use this
global function and everything is working fine now."

Not sure if this is a big deal or if it affects other nodes but I definitely struggled to try and figure out what was happening.

biancode, 2018-11-06:
hm, there is no Array.from() on my side. It is not clear, what exactly you want to report here. Please, could you describe that more in detail?

jewendt, 2018-11-06:
I will try and get some more details.  I was hoping maybe this was enough information to for you to understand what is happening but basically the OPC nodes were breaking other nodes.  This took quite a bit of time to figure out what two nodes were conflicting and I was trying to avoid having to have other users struggle with this if it was an easy fix.

biancode, 2018-11-06:
The whole package is in a named scope like de.biancoroyal.opcua.iiot.* so I'm not sure where it could collide with other nodes.

biancode, 2018-11-06:
please, reopen if you got more details

jewendt, 2018-11-08:
Here is the code details that needed to be changed on the other node.  Sounds like there might be dependency in the OPC stuff that is using array from. Again, I traced it this issue down the the OPC stuff breaking this other node.  
[https://github.com/yoctopuce/yoctolib_es2017/commit/529bc8cd80d326d78e61c1e779846d8fdb2567d5#diff-220547470ce4afde4575674415777835]

biancode, 2018-11-08:
> I was discovered the this was the conflicting detail:
> await Promise.all(Array.from(hubs, hub => hub.hubUpdateDeviceList()));

biancode, 2018-11-08:
> I was discovered the this was the conflicting detail:> await Promise.all(Array.from(hubs, hub => hub.hubUpdateDeviceList()));

biancode, 2018-11-08:
@erossignon some comment from your side about that problem, please? I have searched node-opcua, but did not find any global Array.from calls

biancode, 2018-11-08:
Please, could you tell me, which node packages are broken and how to reproduce that problem?

erossignon, 2018-11-08:
Array.from is not supported by all javascript version (https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Array/from) and is sometime replaced with a polyfill
It may be also that one of the package dependencies introduces a faulty polyfill causing other packages to fail.
What nodejs version is in used here ?

biancode, 2018-11-08:
@jewendt could answer, please ...

jewendt, 2018-11-09:
node -v v8.12.0- Broke as soon as I did the installation. Did not even have to add Nodes to the flow. 
This node broke kept returning null until I worked the the developer as logged above.

[{"id":"89286633.d36348","type":"YSensor","z":"d90be689.21c558","yoctohub":"f68600d2.6338c","name":"","useTimeNot":false,"hwid":"YRNGFND1-DC645.rangeFinder1","x":187.10000610351562,"y":191.60000610351562,"wires":[[]]},{"id":"f68600d2.6338c","type":"yoctohub","z":"","hostname":"127.0.0.1","port":"4444"}]

Steve-Mcl, 2018-11-09:
node-red v19.5, NodeJS V8

Installing and enabling the Listener node seems to destroy/delete 'global' context.

Steps to reproduce.

1. With node-red-contrib-iiot-opcua disabled or uninstalled, add the below flow.  See how flow & global context are set correctly when the injects are operated.
`[{"id":"d9ab9415.507d08","type":"inject","z":"b5d5f872.968d18","name":"","topic":"","payload":"","payloadType":"date","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":116,"y":136,"wires":[["d1617850.7dc248"]]},{"id":"d1617850.7dc248","type":"change","z":"b5d5f872.968d18","name":"Store payload in FLOW and GLOBAL","rules":[{"t":"set","p":"myvar","pt":"global","to":"payload","tot":"msg"},{"t":"set","p":"myvar","pt":"flow","to":"payload","tot":"msg"}],"action":"","property":"","from":"","to":"","reg":false,"x":371,"y":136,"wires":[["d717bb39.af47c8"]]},{"id":"858ca526.afbda8","type":"function","z":"b5d5f872.968d18","name":"store JSON in flow and GLOBAL","func":"global.set(\"url\",msg.payload.url);\nflow.set(\"url\",msg.payload.url);\nmsg.globs = global;\nmsg.globkeys = global.keys();\nreturn msg;","outputs":1,"noerr":0,"x":352,"y":42,"wires":[["88c8d227.d7fb9"]]},{"id":"2a6ed784.678728","type":"inject","z":"b5d5f872.968d18","name":"goog","topic":"","payload":"{\"url\":\"www.google.com\"}","payloadType":"json","repeat":"","crontab":"","once":false,"onceDelay":"","x":102,"y":42,"wires":[["858ca526.afbda8"]]},{"id":"9b195bd1.b511c8","type":"inject","z":"b5d5f872.968d18","name":"yahoo","topic":"","payload":"{\"url\":\"www.yahoo.com\"}","payloadType":"json","repeat":"","crontab":"","once":false,"onceDelay":"","x":102,"y":82,"wires":[["858ca526.afbda8"]]},{"id":"d717bb39.af47c8","type":"debug","z":"b5d5f872.968d18","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","x":622,"y":133,"wires":[]},{"id":"88c8d227.d7fb9","type":"debug","z":"b5d5f872.968d18","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","x":576,"y":40,"wires":[]}]`

2. Install or enable node-red-contrib-iiot-opcua
 > note how on the Context Data tab that global is now empty (flow is still set)

3. Remove or disable node-red-contrib-iiot-opcua
 > Still not working

4. STOP / START node-red
 > global is now working again.

To help narrow this down, I have noticed as soon as I enable the "listener" node, the global context is gone.

biancode, 2018-11-09:
here what is changed - https://github.com/biancode/node-red-contrib-iiot-opcua/commit/72b89485ce8a7fcaa2fd229784a7fd39b75e8bdb

biancode, 2018-11-09:
Please, test it with v3.5.0 - the context still working on my side - a big mistake was that I forgot to implement the removal of all listeners in the new event-based handling. The memory is much less with the latest version, too.

biancode, 2018-11-09:
[image]

biancode, 2018-11-10:
* memory is very stable or decreases after some time significantly
* external OPC UA server closed and reopen and node-red client reconnects
* deploy inside node-red works well after 7 hours runtime
* external client connects well with and without user to node-red OPC UA server

biancode, 2018-11-11:
constant memory usage after 24h

biancode, 2018-11-11:
fixed with v3.6.+

anthrogan, 2018-11-12:
Hi @biancode 

global.get/set still appears broken after update. Flow/Context stores are fine
[image]
I have global.set underneath each flow.set, flow works fine however global does not appear to work at all.

[image]

Current workaround is to store something in a Json key store or something similar.

biancode, 2018-11-12:
maybe the NODE-RED core team could help...

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

anthrogan, 2018-11-12:
I will explore further and submit a discussion with the core team 😃

biancode, 2018-11-12:
some links:

https://nodered.org/docs/user-guide/context

https://nodered.org/docs/creating-nodes/context

https://nodered.org/docs/writing-functions#storing-data

biancode, 2018-11-12:
[image]

biancode, 2018-11-12:
I mean it is maybe just a problem with the context plugin?

Steve-Mcl, 2018-11-16:
Hi biancode, I appreciate your efforts to fix this. 

I'd just like to say that while it may be a problem with the context object, what still remains true is...

* Installing node-red-contrib-iiot-opcua breaks global 

* Removing it fixes global. 

Please follow the instruction I posted originally - step for step - you will see at step 1, both flow and global have values. After installing node-red-contrib-iiot-opcua, flow still has values but global no longer works. 

Hope that helps in some way?

biancode, 2018-11-16:
@Steve-Mcl which version do you use of Node-RED? Did you see the issues on 0.19.4: Maintenance Release with global and flow context?

biancode, 2018-11-17:
The problem comes defently from the Listener node.

biancode, 2018-11-17:
I got it .. require('collections/map') is the problem

biancode, 2018-11-17:
require('collections/map') has to be removed, because it has collisions with node-red

biancode, 2018-11-17:
@Steve-Mcl  and @jewendt - please - test the v3.7.2 without the collections/map

closed 2018-11-17 by biancode

jewendt, 2018-11-17:
Yes. I didn't test yet but looking at the comments it looks like a number of people were having a problem with this package due to array.from.  Nice find!