BiancoRoyal/node-red-contrib-iiot-opcua #76: Result filter boolean issues

NichtEuler, 2018-12-05:
**I'm submitting a ...** 

## Problem
 - ( x ) unexpected behaviour => please visit the Wiki before

### Information
 **Current behaviour**
 
Result filter always returns true when "Send Just Values" Box checked.
If "Send Just Values" unchecked:  - true: value in msg.payload
                                                   - false: value in msg.payload.value.value

 **Expected behaviour**
 
 Boolean values should be in same location.
 **Minimal reproduction of the problem with instructions**
 
   1. A minimum flow with this problem [flow1.txt](https://github.com/biancode/node-red-contrib-iiot-opcua/files/2648172/flow1.txt)
   2. UAExpert of node when false
[image]

   3. UAExpert of node when true
[image]

   

#### Your Environment
**Please tell us about your environment:**

  * ( ) I am using just the Node-RED package and got node-opcua installed with it
  * ( ) I have installed node-opcua global from source (using git clone)
  * ( ) I have installed node-opcua as a global package (using npm i -g)
  * (x) I have installed https://hub.docker.com/r/nodered/node-red-docker/
        
  * Device :

    - ( x ) Windows: 10 

   
   * Description of the connecting OPC UA system:
   
     - Name   :  UA-.NETStandard
     - Link   : https://github.com/OPCFoundation/UA-.NETStandard
     
    
* **node-red-contrib-iiot-opcua version:** ``

* **Node:** (min. 8.x)
   `node --version` = v6.14.4 (Maybe this is the issue?)

biancode, 2018-12-05:
> {"value":{"dataType":"Boolean","arrayType":"Scalar","value":false}} value origin type object convert to Boolean

is now object and not boolean

biancode, 2018-12-05:
opcuaIIoT:core:details convertDataValue: {"value":true} value origin type boolean convert to Boolean +5s
  opcuaIIoT:core:details convertDataValue is: true +0ms
  opcuaIIoT:core:details convertDataValue: {"value":{"dataType":"Boolean","arrayType":"Scalar","value":false}} value origin type object convert to Boolean +10s
  opcuaIIoT:core:details convertDataValue is: [object Object] +0ms

biancode, 2018-12-05:
hint to @erossignon -> there are two different results from node-opcua 0.4.6 on boolean

biancode, 2018-12-06:
I'm sorry, but the payload will be an object for now if it is false.

closed 2018-12-06 by biancode