BiancoRoyal/node-red-contrib-iiot-opcua #79: Default self-signed security certificates appear to be missing

Wexford, 2018-12-10:
**I'm submitting a ...**  (check one with "x")
   
## Feature
 - ( ) feature request
 
 **What is the motivation or Use Case for your feature request?**
 
 
 

## Support
 - support request => **Please do not submit support requests here**, instead see use [the link to our support page](https://bianco-royal.cloud/supporter/#/content/backer)
 - ( ) simple question => please visit the Wiki and Info tab by clicking on the node inside Node-RED before
 - FAQ: Can I send you individual support requests via E-Mail? 
   - Yes, sure! :-D
   
   
   

## Problem
 - (x) bug report => please search GitHub for a similar issue or PR before submitting
 - ( ) unexpected behaviour => please visit the Wiki before

### Information
 **Current behaviour**
 

Cannot connect to third-party local OPC UA Server. Error is:

Error:  cannot locate certificate file /opt/lib/node_modules/node-red-contrib-iiot-opcua/node_modules/node-opcua-client/certificates/client_selfsigned_cert_1024.pem
    at OPCUAClientBase (/opt/lib/node_modules/node-red-contrib-iiot-opcua/node_modules/node-opcua-client/src/client_base.js:74:15)
    at new OPCUAClient (/opt/lib/node_modules/node-red-contrib-iiot-opcua/node_modules/node-opcua-client/src/opcua_client.js:74:21)
    at /opt/lib/node_modules/node-red-contrib-iiot-opcua/opcuaIIoT/maps/opcua-iiot-connector.js:761:19
    at Layer.handle [as handle_request] (/opt/lib/node_modules/node-red/node_modules/express/lib/router/layer.js:95:5)
    at next (/opt/lib/node_modules/node-red/node_modules/express/lib/router/route.js:137:13)
    at /opt/lib/node_modules/node-red/red/api/auth/index.js:63:13
    at Layer.handle [as handle_request] (/opt/lib/node_modules/node-red/node_modules/express/lib/router/layer.js:95:5)
    at next (/opt/lib/node_modules/node-red/node_modules/express/lib/router/route.js:137:13)
    at Route.dispatch (/opt/lib/node_modules/node-red/node_modules/express/lib/router/route.js:112:3)
    at Layer.handle [as handle_request] (/opt/lib/node_modules/node-red/node_modules/express/lib/router/layer.js:95:5)

 
 **Expected behaviour**
 

Expect that the default self-signed certificates are created automatically in the default location (this appeared to be working in earlier versions).
 
 **Minimal reproduction of the problem with instructions**
 
   1. Start with a clean system.
   2. Install node-red and node-red-contrib-iiot-opcua
   3. Create a simple flow that reads a single OPC UA variable from a local, third-party OPC UA Server.
   

#### Your Environment
**Please tell us about your environment:**

Node.js  version: v6.11.2
Node-RED version: v0.19.5
node-red-contrib-iiot-opcua version: v3.7.4

  * (x ) I am using just the Node-RED package and got node-opcua installed with it
  * ( ) I have installed node-opcua global from source (using git clone)
  * ( ) I have installed node-opcua as a global package (using npm i -g)
        
  * Device :
    
    - ( ) macOS: v 
    - (x ) Linux: v 4.14.63-rt40-pxc arm LE (custom OS for PLCnext controller from Phoenix Contact)
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting OPC UA system:
   
     - Name   : embedded OPC UA Server from Phoenix Contact
     - Version:
     - Vendor :
     - Link   : https://
     
    
* **node-red-contrib-iiot-opcua version:** ``

* **Node:** (min. 8.x)
   `node --version` = v6.11.2

biancode, 2018-12-10:
Dear Martin,

> /opt/lib/node_modules/node-red-contrib-iiot-opcua/node_modules/node-opcua-client/certificates/

please, could you check that this folder exists?
If not, then there is a problem on the installation of node-opcua.
Somtimes you have to tkae care of that the openssl is installed as well as other basic compile libraries.
see [node-opcua](https://github.com/node-opcua/node-opcua)

> /opt/lib/node_modules/node-red-contrib-iiot-opcua/

You could try to just install the node-opcua in that folder again too.

If it doesn't cause any problems, then I would recommend Node.js v10 LTS instead of v6.

Happy installing!
Klaus

Wexford, 2018-12-10:
Hi Klaus,

Thank you for the quick response.

That ‘certificates’ folder does not exist.

I noticed when filling out the issue that I’m running a very old version of
Node, so I’ll upgrade that tomorrow morning and try re-installing your
package. I’ll update the issue when I find the solution.

In case you’re interested, I’m preparing a how-to video for using the
iiot-opcua package on our PLCnext controller. I’ll mention your website in
the show notes, so hopefully that will direct more traffic your way.

Regards,
Martin Boers
Phoenix Contact Electronics GmbH

biancode, 2018-12-10:
here is some [YouTube](https://www.youtube.com/playlist?list=PLdcYFhtEB7oke6US0tPIOLgp4oPe4cami) content too

Wexford, 2018-12-11:
Solved by installing node version v10.14.1 and re-installing node-red and node-red-contrib-iiot-opcua.

Thank you Klaus for your help!

closed 2018-12-11 by Wexford