BiancoRoyal/node-red-contrib-iiot-opcua #83: method caller node not working with Boolean input parameter

imSomnath, 2019-01-24:
**I'm submitting an issue** 

## Problem
 - ( ) bug report => Not able to pass boolean input parameter to method using method caller node.
Error = " Invalid variant arrayType: Scalar dataType: Boolean value:true"

### Information
 **Current behaviour**
 
I am trying to setup a node-red flow where i am using "IIOT method caller" node for connection. My configuration seems OK but when i am passing Boolean input parameter which is acceptable parameter for my method then it is not working and giving error. But same method is called successfully through UaExpert client.
 
 **Expected behaviour**
 
 I should call it successfully without any error.

  * ( ) I have installed node-opcua as a global package (using npm i -g)
        
  * Device :
    
    - ( ) Linux: v debian 
     
    
* **node-red-contrib-iiot-opcua version:** ``
 latest

* **Node:** (min. 8.x)
   `node --version` = latest

biancode, 2019-03-15:
moved to https://github.com/BiancoRoyal/node-red-contrib-iiot-opcua/issues/2

closed 2019-03-15 by biancode

biancode, 2019-03-15:
the value has to be a real boolean it is set directly to the value of input argument - please send a flow with that test case or write a test and send a pull request, please

ambrosini-smeup, 2019-12-03:
Hi,
I'm experiencing the same issue as above, but I can't understand wich is the fix from this thread.

If I add the boolean parameter in the Method Caller Node I get the " Invalid variant arrayType: Scalar dataType: Boolean value:true" error.

Here it is my flow

`[{"id":"2f5960f.75300a","type":"tab","label":"Flow 1","disabled":false,"info":""},{"id":"8e88dc77.383da","type":"function","z":"2f5960f.75300a","name":"parametri","func":"msg.payload = {\n    objectId: 'ns=2;s=Devices',\n    methodId: 'ns=2;s=setOrder',\n    inputArguments: [\n        {name: 'machineID', dataType:'String', value:'MAC_01'},\n        {name: 'refNo', dataType:'UInt32', value:'1'},\n        {name: 'oderNo', dataType:'String', value:'OTST'},\n        {name: 'opNo', dataType:'String', value:'10'},\n        {name: 'amount', dataType:'Int64', value:'1000'},\n        {name: 'article', dataType:'String', value:'art 123'},\n        {name: 'status', dataType:'UInt32', value:'1'},\n        {name: 'limit', dataType:'Int64', value:'11000'},\n        {name: 'produced', dataType:'Int64', value:'0'},\n        {name: 'bad', dataType:'Int64', value:'0'},\n        {name: 'good', dataType:'Int64', value:'0'},\n        {name: 'idle', dataType:'Int64', value:'0'},\n        {name: 'pps', dataType:'Int64', value:'1'},\n        {name: 'presort', dataType:'Int64', value:'0'},\n        {name: 'qualityCounterInterval', dataType:'UInt32', value:'1000'},\n        {name: 'qualityCounterCurrent', dataType:'UInt32', value:'0'},\n       // {name: 'qualityCounterStop', dataType:'boolean', value:'0'},\n        {name: 'qualityCounterWarning', dataType:'Double', value:'0.8'}\n    ],\n    methodType: 'basic'\n}\nreturn msg;","outputs":1,"noerr":0,"x":260,"y":100,"wires":[["e38134df.672208"]]},{"id":"2acff1e.a139f0e","type":"inject","z":"2f5960f.75300a","name":"","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":130,"y":100,"wires":[["8e88dc77.383da"]]},{"id":"e38134df.672208","type":"OPCUA-IIoT-Method-Caller","z":"2f5960f.75300a","connector":"6f81924a.7159fc","objectId":"","methodId":"","methodType":"basic","value":"","justValue":true,"name":"SetOrder","showStatusActivities":true,"showErrors":true,"inputArguments":[],"x":420,"y":100,"wires":[["ef63241c.addbb8"]]},{"id":"ef63241c.addbb8","type":"OPCUA-IIoT-Response","z":"2f5960f.75300a","name":"","compressStructure":true,"showStatusActivities":false,"showErrors":false,"activateUnsetFilter":false,"activateFilters":false,"negateFilter":false,"filters":[],"x":630,"y":100,"wires":[["42541e9e.29bc"]]},{"id":"42541e9e.29bc","type":"debug","z":"2f5960f.75300a","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","x":830,"y":100,"wires":[]},{"id":"6f81924a.7159fc","type":"OPCUA-IIoT-Connector","z":"","discoveryUrl":"","endpoint":"opc.tcp://10.110.0.60:4840","keepSessionAlive":true,"loginEnabled":false,"securityPolicy":"None","securityMode":"NONE","name":"Vibo","showErrors":false,"individualCerts":false,"publicCertificateFile":"","privateKeyFile":"","defaultSecureTokenLifetime":"","endpointMustExist":false,"autoSelectRightEndpoint":false,"strategyMaxRetry":"","strategyInitialDelay":"","strategyMaxDelay":"","strategyRandomisationFactor":"","requestedSessionTimeout":"","connectionStartDelay":"","reconnectDelay":"","maxBadSessionRequests":"10"}]`