BiancoRoyal/node-red-contrib-iiot-opcua #84: Memroy usage goes up with time

rskhatri, 2019-02-06:
I'm submitting an issue

## Problem
 - ( ) bug report => We are using this node for armv7 based platform with Linux. Problem we are facing is it is taking lot of memory with time and at the end device runs out of memory.

### Information
 **Current behaviour**
 We are trying to create flow where 18 tags/messages would update in dynamic user space. It works fine and after few hours device goes out of memory.

 **Expected behaviour**
 It should not use more memory on run time. 

#### Your Environment
  * ( ) I am using just the Node-RED package and got node-opcua installed with it
        
  * Device :
    - ( ) Linux: v 4.10
 
* **node-red-contrib-iiot-opcua version:** ``
3.8.0

* **Node:** (min. 8.x)
   `node --version` = 8.11.4

biancode, 2019-02-06:
Hi, could you write about the Node-RED version too, please?

biancode, 2019-02-09:
reopen with the information

closed 2019-02-09 by biancode