BiancoRoyal/node-red-contrib-iiot-opcua #87: Response and Listerner limited to 1000 variables

thecalle, 2019-02-26:
I'm submitting an  unexpected behaviour.
   
**Current behaviour**: IIoT OPC UA Browse2Listener example scenario. Connect to OPC-UA server exposing 4000 variables, Response gives 1000 variables, Listener subscribes to 1000 variables.
 
**Expected behaviour**: Connect to server exposing 4000 variables, Response gives 4000 variables, Listener subscribes to 4000 variables.

**Environment:**
I am using just the Node-RED package and got node-opcua installed with it.

**Device** : Raspbian: GNU/Linux 9

thecalle, 2019-03-01:
Browsing also (using magnifing button) is limited to 1000!

biancode, 2019-03-01:
Hi, check your server settings! We use 10k nodes in other projects.

thecalle, 2019-03-01:
Hi Klaus. I connect to the same server using UaExpert Windows client and I get about 4K varibles under Tags. Node-red browser stops exactly at 999.

biancode, 2019-03-01:
v3.8.0 of the package? Please, Francesco, could you take a screenshot of the sever limits from UAExpert and pot it here or send it by E-Mail!

thecalle, 2019-03-01:
Hi Klaus.

Package is v.3.8.0, here the screenshot feom UaExpert. I'm using our
Browser2Listen example to test.

Thanks in advance.

Francesco

[image: image.png]

biancode, 2019-03-01:
hm - no image to see here

thecalle, 2019-03-01:
Check now.

F.

biancode, 2019-03-01:
sorry, no image just 
[image: image.png]

thecalle, 2019-03-04:
Hi Klaus.

I think images are dropped by GITHub. Last mail has a correct .JPG attach

Francesco

biancode, 2019-03-04:
Please, just send it by E-Mail. I did not get any image. Just [image: image.png] also in the E-Mails

biancode, 2019-03-04:
[image]

thecalle, 2019-03-06:
Hi Klaus.

I checked the parameter you suggested. The MaxNodesPerBrowse is set to
65536. All the parameters in Operation Limits are set to the same value.

Francesco.

anthrogan, 2019-03-11:
I have experienced similar with only returning 1000 tags, and ultimately bringing the opc server down (WinCC... Again! I can't confirm max nodes per browse either with wincc as the server does not expose this)

biancode, 2019-03-15:
How can I reproduce this use case?

biancode, 2019-03-17:
browse is restricted now from node-opcua and it will provide the browse next in a next step end of the month march 2019

erossignon, 2019-04-07:
@biancode  browse will limit the number of returned node to 1000 and will set a continuous point in the response if more than 1000 nodes exists. To get all the node , one shall call browseNext repeatedaly  until continous point becomes null.

biancode, 2019-04-07:
@erossignon you said it is not possible last time. Is it possible now and from which version of node-opcua, please?

erossignon, 2019-04-07:
on the client side, OPCUACLient#browseNext can be used since a long time and shall be available in v0.7.0.
version 2.0.0 will add further enhancements and browseNext support in the server and in the client crawler.

biancode, 2019-12-14:
[image]
browseNext works

biancode, 2019-12-14:
[6kNodesFlowBrowse.json.zip](https://github.com/BiancoRoyal/node-red-contrib-iiot-opcua/files/3964364/6kNodesFlowBrowse.json.zip)

needs v3.11.3 or above

biancode, 2019-12-15:
fixed with v3.12.0 - **but** the listener itself is also limited by the server how many items to be monitored at the same server

closed 2019-12-15 by biancode