BiancoRoyal/node-red-contrib-iiot-opcua #92: Method

abraham200200, 2019-04-08:
How can i define a function as a Method of opc ua server with iiot nodes

biancode, 2019-04-17:
within the flex server

closed 2019-04-17 by biancode

Jack-Linda, 2019-09-03:
can i find a example of define a function as a Method within flex server

biancode, 2019-09-03:
@Jack-Linda check out the Import inside Node-RED: Import -> Examples -> iiot opcua

Jack-Linda, 2019-10-08:
> @Jack-Linda check out the Import inside Node-RED: Import -> Examples -> iiot opcua

I am very sorry I can't find a examples of define a function as a Method of opc ua server by flex server.can I define it in address Space of flex server?how to do it? I'm dying to get your help.

biancode, 2019-10-08:
@Jack-Linda http://node-opcua.github.io/ you can use the whole node-opcua API - take a bit time to learn more about the API, please!