BiancoRoyal/node-red-contrib-iiot-opcua #93: Node-red-contrib-iiot-opcua | empty field

Sadafaga, 2019-04-11:
**Current behaviour**

I have an OPC UA server running and connected to UA Expert. I have tried to inject data from the general "inject node" to a "OPCUA-IIoT-Server-ASO" and after that to the OPC UA Server, everything is ok except the field "value" that has as value " " instead the string that I am injecting. How can I fix it?

 I hope that this image can help to understand my problem. You can see remarked in red the type of "opc ua node" that I am using and remarked in blue the ouput with the field value with a " " instead of "hello" that is what I am injecting.
 
[image]

 **Expected behaviour**

I expect that in the field "value" it appears the string "hello" and not " ".
    
 
  * Device :   
   
    - (x ) Raspbian: Raspbian GNU/Linux 9 (stretch)   
    
* **Node-RED version:** 
0.20.0

* **Node:** (min. 8.x)
10.5.3

abraham200200, 2019-04-16:
to write a value you need the write node('OPCUA IIoT Write')

Sadafaga, 2019-04-16:
my intention is to take the data I receive in mqqt to the server opcua but I recive this " " or the following error message "Error: ASO NodeId Is Not Valid"

Sadafaga, 2019-04-16:
I'm getting the following error "nodeId ns = 1; s = XXXX already registered" and I would like that every time I inject a message this overwrites the previous one and therefore does not appear this error message. How can I fix it? Thanks

closed 2019-04-17 by biancode