BiancoRoyal/node-red-contrib-iiot-opcua #97: Error: ServiceResult is BadTooManyOperations (0x80100000) request was ReadRequest

giminni, 2019-04-26:
## Problem
 - ( ) bug report => please search GitHub for a similar issue or PR before submitting
 - (x) unexpected behaviour => please visit the Wiki before

### Information
 **Current behaviour**
 After upgrading from 2.7.1 to latest version I checked on the read node all for all attribute Id and got the following error message **Error: ServiceResult is BadTooManyOperations (0x80100000) request was ReadRequest**

 **Expected behaviour**
I expect a list of all attributes in one nodeId
 
 **Minimal reproduction of the problem with instructions**

   1. injector node
   2. browse node with root nodeId
   3. read node with set ALL on attributeId
   4. response node
   

#### Your Environment

     -(x) nodeRED: v0.20.5
     -(x) node: v8.x

   * Device :
    
    - (x) macOS: v10.14.4
    - ( ) Linux: v 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:

giminni, 2019-04-26:
This error occurred while getting 900th node IDs with all attributes out of an OPCUA device.

Is there a chance to add a numeric entryfield where a user can enter a numeric number to get only some specific Ids?

```
export enum AttributeIds {
    NodeId= 1,
    NodeClass= 2,
    BrowseName= 3,
    DisplayName= 4,
    Description= 5,
    WriteMask= 6,
    UserWriteMask= 7,
    IsAbstract= 8,
    Symmetric= 9,
    InverseName= 10,
    ContainsNoLoops= 11,
    EventNotifier= 12,
    Value= 13,
    DataType= 14,
    ValueRank=  15,
    ArrayDimensions= 16,
    AccessLevel= 17,
    UserAccessLevel= 18,
    MinimumSamplingInterval= 19,
    Historizing= 20,
    Executable= 21,
    UserExecutable= 22,
    INVALID= 999
}
```

giminni, 2019-04-29:
is this a restriction of a buffer? I want read approx. 2000 data points in one directory with ALL attributes, how can I achieve this?

anthrogan, 2019-05-23:
If memory serves node-opcua can only browse 1000 nodes maximum.

biancode, 2019-12-15:
fixed with v3.12.0

closed 2019-12-15 by biancode