BiancoRoyal/node-red-contrib-iiot-opcua #98: Docker build fails, probably due to deasync

echtelerp, 2019-05-16:
**I'm submitting a ...**  (check one with "x")

## Problem
 - (X) bug report => please search GitHub for a similar issue or PR before submitting
 - ( ) unexpected behaviour => please visit the Wiki before

### Information
 **Current behaviour**
When I try to build a docker container the build fails due to your package. 

I used this package.json file: 
`{
    "name": "node-red-docker",
    "version": "1.0.0",
    "description": "A visual tool for wiring the Internet of Things",
    "homepage": "http://nodered.org",
    "license": "Apache-2.0",
    "repository": {
        "type": "git",
        "url": "https://github.com/node-red/node-red-docker.git"
    },
    "main": "node_modules/node-red/red/red.js",
    "scripts": {
        "start": "node $NODE_OPTIONS node_modules/node-red/red.js -v $FLOWS"
    },
    "contributors": [
        {
            "name": "Dave Conway-Jones"
        },
        {
            "name": "Nick O'Leary"
        },
        {
            "name": "James Thomas"
        }
    ],
    "dependencies": {
        "node-red": "0.20.5",
        "node-red-node-msgpack": "*",
        "node-red-node-base64": "*",
        "node-red-node-suncalc": "*",
        "node-red-node-random": "*",
        "node-red-dashboard":"*",
        "node-red-contrib-iiot-opcua":"*",
        "node-red-contrib-influxdb":"*",
        "node-red-contrib-simple-message-queue":"*",
        "node-red-contrib-s7comm":"*",
        "node-red-contrib-blockly":"*",
        "node-red-node-ping":"*",
        "node-red-contrib-telegrambot":"*"
    },
    "engines": {
        "node": "8.*.*"
    }
}
`

If I remove the OPC-UA package, the build succeeds... 
therefore it breaks because of your package. 

I used a custom build node v8.15.0 container, based on a minimal alpine image and with the ARM architecture. (echteler/node:arm32v7)
Missing python was at first a problem. I added python, but it still breaks. 
It seems to have something to do with deasync. 

 
 **Expected behaviour**
Build should succeed :-)
 
  * ( ) I am using just the Node-RED package and got node-opcua installed with it
  * ( ) I have installed node-opcua global from source (using git clone)
  * ( ) I have installed node-opcua as a global package (using npm i -g) 
  None of the above, I Used the provided package.json file. 
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - (X) Other: Docker
         - Name:
         - Version:

* **Node-RED version:** 
Docker NodeRed 0.20.5

* **node-red-contrib-iiot-opcua version:** ``
latest on? 

* **Node:** (min. 8.x)
   `node --version` = 8.15

biancode, 2019-05-16:
@erossignon

echtelerp, 2019-08-16:
hi @biancode,

i was able to resolve the issue.
I had to add the following packages via pkg add to the node js container: 
in the docker file add the line: 
`RUN apk add --no-cache python make g++`

after that the build succeeded :-)

so I would suggest maybe adding that to the readme file. 
sort of a make sure: those packages are available in the base container when trying to use the node in a alpine arm based docker image. 

have a great weekend 
philipp

echtelerp, 2019-08-16:
i also maybe think, this was actually related to the node-opcua dependency. was quite a pain in the a** to dig into

erossignon, 2019-08-18:
@echtelerp  @biancode  deasync is not used anymore in latest versions of node-opcua (@2.1.4 and @0.7.0 ). This will remove the need of installing the python and compilation tool set

biancode, 2019-11-26:
prerelease 3.11.1-4 uses now 0.7.4

closed 2019-11-26 by biancode